ipxe/ipxe #122: IPXE fails to boot CentOS 8

zerocarbthirty, 2020-07-21:
Hello, I am converting some PXELINUX over to IPXE (for UEFI). I am booting with ipxe.uefi:

this is the pxelinux config that works flawlessly (in BIOS mode)

label c8m
  IPAPPEND 2
  KERNEL CentOS8.2x64/vmlinuz
  append initrd=CentOS8.2x64/initrd.img ksdevice=bootif ks=http://10.1.13.195/Centos8/kickstart-manual

Here is the IPXE config that I am trying to use

:centos8m
kernel CentOS8.2x64/vmlinuz initrd=initrd.img
initrd CentOS8.2x64/initrd.img ksdevice=bootif ks=http://10.1.13.195/Centos8/kickstart-manual
boot

When I try to boot it with IPXE it gives a bunch of dracut-initqueue timeout errors and then drops to an emergency shell.

Thanks in advance if anyone can assist me.

mcb30, 2020-07-21:
Try instead:
```
:centos8m
kernel CentOS8.2x64/vmlinuz initrd=initrd.img ks=http://10.1.13.195/Centos8/kickstart-manual
initrd CentOS8.2x64/initrd.img
boot
```
(i.e. move the kernel command line arguments to the `kernel` line, and remove `ksdevice=bootif`)

zerocarbthirty, 2020-07-21:
Yeah that fixed it. Was also able to get ubuntu debian, vmware esxi, and
windows working fine. Thanks!!

closed 2020-07-21 by mcb30

jamaya77, 2023-08-14:
This worked for me as well. Thank you!