ipxe/ipxe #125: Tether iPXE via Android or iOS devices does not seem to work anymore

Hammarskjold, 2020-07-22:
When using iPXE builds with ECM--NCM before, iPXE was successfully pairing when connected to a phone via USB cable when the phone was set in USB tether mode. With the latest builds the phones are not detected as network interfaces anymore. With DEBUG=usb enabled the device is found, but does not seem to show itself as a network device.

Tried devices;

iPhone5
iPhone7
Google Pixel 2
Google Pixel 4

mcb30, 2020-07-22:
With my Android device (Samsung Galaxy S9), you need the `acm` driver since it presents as an ACM (RNDIS) device.  This may be Android version-specific.

I am able to successfully use `bin-x86_64-efi/snp--ecm--ncm--acm.usb` to obtain an address from DHCP via my tethered S9 (attached as a USB device passed through to a qemu VM).

Hammarskjold, 2020-07-22:
Indeed, including the ACM driver made the Google Pixel 2 kick into action. Same with the iPhone6S, so think we can bury this one.

closed 2020-08-04 by mcb30