ipxe/ipxe #126: Scalable hosting for ca.ipxe.org files

mcb30, 2020-07-22:
### Description

As a sysadmin, I would like the cross-signed certificate files (see https://ipxe.org/cfg/crosscert) to be hosted on a highly available service, so that I do not have to worry about outages of the single machine currently hosting `ca.ipxe.org`.

The cross-signed certificate files are just static files (with filenames constructed from a hash of the certificate subject name) and can be served from any infrastructure capable of serving static files via HTTP.  The request from iPXE includes a query string encoding the raw subject name but this is only for debugging purposes and can be safely ignored.

### Acceptance criteria

* [ ] Cross-signed certificate files are hosted on a highly available service (such as AWS S3 in multiple regions)
* [ ] Cross-signing process is updated to upload files automatically to the HA hosting service
* [ ] Cross-signed certificate files can be downloaded using the existing `http://ca.ipxe.org/auto` base URI
* [ ] iPXE is able to perform an HTTPS boot using a cross-signed certificate downloaded from the HA hosting service, with no changes to iPXE code or configuration required.

mcb30, 2020-07-22:
Added #127 to handle the related scalable hosting for OCSP.

MaxPeal, 2020-07-22:
we should ask packethost https://www.packet.com/community/open-source/
as the use ipxe in many parts

MaxPeal, 2020-07-22:
i have send packethost a friendly message, to ask about to support ipxe.
@mcb30 i give you direct feedbeck after i got a info form @packethost

vielmetti, 2020-07-22:
@MaxPeal @mcb30 We at @packethost are interested in helping, I'll follow up on the email.

MaxPeal, 2020-07-22:
@mcb30 after i got all the infos and org stuff, you get a Summary

kraxel, 2020-07-24:
qemu can make root ca certificates available to guests, via fw_cfg, and ipxe could use that.

In that case managing the list of certificates would be the job of the host machine,
there would be no need for ipxe to query ca.ipxe.org or ocsp.ipxe.org.

Of course this would solve the problem for (qemu/kvm) virtual machines only,
Deployments on physical hardware still need ca.ipxe.org.