ipxe/ipxe #127: Scalable hosting for ocsp.ipxe.org

mcb30, 2020-07-22:
### Description

As a sysadmin, I would like the iPXE OCSP service to be hosted on some kind of highly available infrastructure, so that I do not have to worry about outages of the single machine currently hosting `ocsp.ipxe.org`.

Unlike the cross-signed certificate files (see #126), OCSP does not just serve static files.  Some investigation is required to determine a sensible way to host this service.

### Acceptance criteria

* [ ] OCSP service is hosted on a highly available service (such as AWS in multiple regions)
* [ ] Cross-signing process is updated to provide OCSP signing certificates to the HA hosting service
* [ ] HA OCSP service can be accessed using the existing `http://ocsp.ipxe.org` base URI
* [ ] iPXE is able to perform an HTTPS boot using a cross-signed certificate verified using the HA OCSP service, with no changes to iPXE code or configuration required.

robinsmidsrod, 2020-10-06:
@mcb30 Have you considered using e.g. Azure Functions or AWS Lamba (serverless) method to run the OCSP responder?

NiKiZe, 2026-08-19:
https://github.com/ipxe/ipxe/issues/1261#issuecomment-5348228949