ipxe/ipxe #130: aarch64 kvm qemu iPXE fails with “Connection reset (http://ipxe.org/0f0a6095)”

xin3liang, 2020-08-04:
Ironic iPXE arm64 qemu booting always fails with “. Connection reset (http://ipxe.org/0f0a6095)” and http download is very slow

Build ipxe and cp to tftproot
```
$ git clone git://git.ipxe.org/ipxe.git (commit: 70b1a641)
$ cd ipxe/src; make bin-arm64-efi/ipxe.efi
$ cp ~/ipxe/src/bin-arm64-efi/ipxe.efi /opt/stack/data/ironic/tftpboot
```
Booting log
```
boot.ipxe : 758 bytes [script]
Attempting to boot from MAC 52-54-00-09-06-3e
pxelinux.cfg/52-54-00-09-06-3e................................. Connection reset (http://ipxe.org/0f0a6095)
Attempting to boot from MAC 52-54-00-df-72-db
pxelinux.cfg/52-54-00-df-72-db................................. Connection reset (http://ipxe.org/0f0a6095)
PXE boot failed! No configuration found for any of the present NICs.
Press any key to reboot...
Could not boot image: Connection timed out (http://ipxe.org/4c22e092)
No more network devices

Press any key to continue…
```

Running kvm qemu with virtio-net-pci NIC
```stack@j12-d05-07:~/ipxe$ ps aux|grep qemu
libvirt+  406329  3.1  0.2 3810752 274228 ?      Sl   07:32   3:25 qemu-system-aarch64 -enable-kvm -name guest=node-0,debug-threads=on -S -object secret,id=masterKey0,format=raw,file=/var/lib/libvirt/qemu/domain-17-node-0/master-key.aes -blockdev {"driver":"file","filename":"/var/lib/libvirt/images/OVMF_CODE.fd","node-name":"libvirt-pflash0-storage","auto-read-only":true,"discard":"unmap"} -blockdev {"node-name":"libvirt-pflash0-format","read-only":true,"driver":"raw","file":"libvirt-pflash0-storage"} -blockdev {"driver":"file","filename":"/var/lib/libvirt/images/OVMF_VARS.fd-node-0","node-name":"libvirt-pflash1-storage","auto-read-only":true,"discard":"unmap"} -blockdev {"node-name":"libvirt-pflash1-format","read-only":false,"driver":"raw","file":"libvirt-pflash1-storage"} -machine virt-4.2,accel=kvm,usb=off,dump-guest-core=off,gic-version=3,pflash0=libvirt-pflash0-format,pflash1=libvirt-pflash1-format -cpu host -m 3072 -overcommit mem-lock=off -smp 1,sockets=1,cores=1,threads=1 -uuid cafb41a4-9612-4255-a14c-202612c69239 -display none -no-user-config -nodefaults -chardev socket,id=charmonitor,fd=30,server,nowait -mon chardev=charmonitor,id=monitor,mode=control -rtc base=utc -no-shutdown -boot menu=off,strict=on -device pcie-root-port,port=0x8,chassis=1,id=pci.1,bus=pcie.0,multifunction=on,addr=0x1 -device pcie-root-port,port=0x9,chassis=2,id=pci.2,bus=pcie.0,addr=0x1.0x1 -device pcie-root-port,port=0xa,chassis=3,id=pci.3,bus=pcie.0,addr=0x1.0x2 -device pcie-root-port,port=0xb,chassis=4,id=pci.4,bus=pcie.0,addr=0x1.0x3 -device pcie-root-port,port=0xc,chassis=5,id=pci.5,bus=pcie.0,addr=0x1.0x4 -device pcie-root-port,port=0xd,chassis=6,id=pci.6,bus=pcie.0,addr=0x1.0x5 -device pcie-root-port,port=0xe,chassis=7,id=pci.7,bus=pcie.0,addr=0x1.0x6 -device pcie-root-port,port=0xf,chassis=8,id=pci.8,bus=pcie.0,addr=0x1.0x7 -device pcie-root-port,port=0x10,chassis=9,id=pci.9,bus=pcie.0,multifunction=on,addr=0x2 -device pcie-root-port,port=0x11,chassis=10,id=pci.10,bus=pcie.0,addr=0x2.0x1 -blockdev {"driver":"file","filename":"/var/lib/libvirt/images/node-0.qcow2","node-name":"libvirt-1-storage","cache":{"direct":false,"no-flush":true},"auto-read-only":true,"discard":"unmap"} -blockdev {"node-name":"libvirt-1-format","read-only":false,"cache":{"direct":false,"no-flush":true},"driver":"qcow2","file":"libvirt-1-storage","backing":null} -device virtio-blk-pci,scsi=off,bus=pci.10,addr=0x0,drive=libvirt-1-format,id=virtio-disk0,write-cache=on -netdev tap,fd=32,id=hostnet0,vhost=on,vhostfd=33 -device virtio-net-pci,netdev=hostnet0,id=net0,mac=52:54:00:09:06:3e,bus=pci.1,addr=0x0,bootindex=1 -netdev tap,fd=34,id=hostnet1,vhost=on,vhostfd=35 -device virtio-net-pci,netdev=hostnet1,id=net1,mac=52:54:00:df:72:db,bus=pci.2,addr=0x0 -add-fd set=5,fd=37 -chardev pty,id=charserial0,logfile=/dev/fdset/5,logappend=on -serial chardev:charserial0 -device virtio-balloon-pci,id=balloon0,bus=pcie.0,addr=0x7 -sandbox on,obsolete=deny,elevateprivileges=deny,spawn=deny,resourcecontrol=deny -msg timestamp=on
stack     426200  0.0  0.0   5904  2020 pts/7    S+   09:22   0:00 grep --color=auto qemu
```

xin3liang, 2020-08-04:
Sometimes, it can got into kernel download step, the kernel size is around 8M, but it always fails with "Connection reset (http://ipxe.org/0f0a6095)"
```
No response, retrying in {i} seconds.09-06-3e
http://10.30.96.1:3928/84e0a28a-ac5a-4971-a1e1-a3d239a6a702/deploy_kernel............................................................................................................
.....................................................................................................................................................................................
............................................................... Connection reset (http://ipxe.org/0f0a6095)
```

It is sure that kernel download is ok on other machine in the same LAN.
```
linaro@debian:~/tmp$ wget --show-progress -c http://10.30.96.1:3928/84e0a28a-ac5a-4971-a1e1-a3d239a6a702/deploy_kernel
--2020-08-04 07:20:56--  http://10.30.96.1:3928/84e0a28a-ac5a-4971-a1e1-a3d239a6a702/deploy_kernel
Connecting to 10.30.96.1:3928... connected.
HTTP request sent, awaiting response... 200 OK
Length: 8450888 (8.1M)
Saving to: ‘deploy_kernel’

deploy_kernel                                 100%[==============================================================================================>]   8.06M  --.-KB/s    in 0.09s

2020-08-04 07:20:57 (92.8 MB/s) - ‘deploy_kernel’ saved [8450888/8450888]
```

mcb30, 2020-08-06:
I don't see the same issue, sorry.  In my test, I am building iPXE using
```
make CROSS=aarch64-linux-gnu- bin-arm64-efi/1af41000.efirom
```
and starting qemu as
```
qemu-system-aarch64 -cpu cortex-a57 -M virt -m 2048 -serial stdio \
    -bios ..../edk2/Build/ArmVirtQemu-AARCH64/DEBUG_GCC5/FV/QEMU_EFI.fd \
    -netdev tap,ifname=tap0,id=tap0 \
    -device virtio-net,romfile=bin-arm64-efi/1af41000.efirom,netdev=tap0
```
With this setup (also using aarch64, qemu, and a virtio NIC) I am able to repeatedly download a 512MB test file via HTTP in around 6.3s.

xin3liang, 2020-08-07:
> I don't see the same issue, sorry. In my test, I am building iPXE using
>
> ```
> make CROSS=aarch64-linux-gnu- bin-arm64-efi/1af41000.efirom
> ```
>
> and starting qemu as
>
> ```
> qemu-system-aarch64 -cpu [...]

Hi @mcb30 , thanks for replying the issue. That seems to work, could you tell me which qemu version you are using. I will try.
My qemu version is:
 /usr/bin/qemu-system-aarch64 --version
QEMU emulator version 4.2.0 (Debian 1:4.2-3ubuntu6.3)

I've also tried qemu without kvm with cpu cortex-a53, not working too.   
Does it use 1af41000.efirom too when PXE booting ipxe.efi? 
On my environment, it seems, PXE nic has no issue, but iPXE nic doesn't work well . From below tcpdump, it seems there is some problem with iPXE(10.0.0.57) packet receiving which makes http server(10.30.96.1) retransmit packet again and again then timeout.
[image]

mcb30, 2020-08-07:
> Hi @mcb30 , thanks for replying the issue. That seems to work, could you tell me which qemu version you are using.

```
$ qemu-system-aarch64 --version
QEMU emulator version 4.2.0 (qemu-4.2.0-7.fc32)
```

> On my environment, it seems, PXE nic has no issue, but iPXE nic doesn't work well . From below tcpdump, it seems there is some problem with iPXE(10.0.0.57) packet receiving which makes http server(10.30.96.1) retransmit packet again and again then timeout.

Firstly, it's worth noting from the `wireshark` screenshot that you are using iSCSI rather than HTTP.  If you are not expecting to be using iSCSI, then you have some unrelated issues that you need to resolve first.

It looks as though the packet reception problem occurs on the first full-length packet sent by your iSCSI server.  When problems occur on the first full-length packet in a connection, it is normally indicative of an MTU issue somewhere.  Check the size of the overall Ethernet frame (shown as `Frame <X>: <N> bytes on wire` in the `wireshark` user interface), and make sure that this frame is within the MTU of your network (after allowing for the overhead of any VLAN tags, etc).

Michael

mcb30, 2020-08-12:
Marking as closed; please reopen if the previous comments do not resolve the issue.

closed 2020-08-12 by mcb30

xin3liang, 2020-10-28:
...
> It looks as though the packet reception problem occurs on the first full-length packet sent by your iSCSI server. When problems occur on the first full-length packet in a connection, it is normally [...]

Good catch! I check the MTU of tap interface found that it is 1400, and qemu machine iPXE interface is 1450. After I set qemu machine interface as the same MTU size as tap interface this issue gone. 
iPXE> show mtu
net0.dhcp/mtu:int16 = 1450
iiPXE> set net0.dhcp/mtu 1400  (mtu from dhcp server, need to fix it in dhcp server)
iPXE> imgfetch http://10.30.96.1:3928/0793a94f-7cbe-4859-8028-29fd20244b33/ramdisk
http://10.30.96.1:3928/0793a94f-7cbe-4859-8028-29fd20244b33/ramdisk... ok
This time it fetch 26M file using about 2 seconds. Which is rapid.

Thanks a lot Michael :-)

Xinliang

mcb30, 2020-10-28:
> This time it fetch 26M file using about 2 seconds. Which is rapid.
> 
> Thanks a lot Michael :-)

Glad you got it working!  :slightly_smiling_face: