ipxe/ipxe #132: Wierd Windows ARP issue

Adam5Wu, 2020-08-15:
I have a Synology iSCSI target on the same subnet as a machine with Intel i210 NIC, which I build iPXE uEFI firmware for iSCSI booting a Windows Server 2016. Both iPXE and Windows configures to use DHCP on the NIC.

The problem I have is that, once Windows boots, it's ARP cache entry of the iSCSI target is stuck as "invalid", and all iSCSI traffic from Windows to iSCSI target is sent to the gateway, which dutifully forwards the traffic to the destination. But the performance suffers, cannot reach line-speed because of the extra "routing" hop.

Interestingly, the incoming iSCSI traffic from the iSCSI target to Windows are correctly sent directly from the Synology to the Windows machine, creating a strange traffic "triangle".

Packet sniffer shows that, when iPXE starts, it does send ARP request of the iSCSI target; But once Windows boot takes over, it never send out any ARP request of the iSCSI target. Moreover, once boot completes, I try to:
- Manually remove the "invalid" ARP entry, which doesn't do anything at all;
- Trace route inside Windows always show 2 hops, the first being the router;
- Adding permanent ARP entry of the iSCSI target (via powershell) did make the arp tool render the iSCSI target having corrected MAC address, but it does not change the route.

It almost feels like Windows somehow has, at a very low level, stuck the iSCSI target address as not reachable on the link layer, and force all packets addressed to the iSCSI target to be routed.

Then, I made a (self-proclaimed) "brilliant" hack which seemingly resolved the problem: "if you (Windows) insist on routing the traffic, what happens if there is no gateway to use?"
I commented out [this line](https://github.com/ipxe/ipxe/blob/e7f67d5a4c6e9f06aa7a9db1b4245f5e16f00bb2/src/drivers/block/ibft.c#L290) basically prevent iBFT from passing Windows the gateway. Then magic happened -- during Windows boot, an ARP request of the iSCSI target showed up on the wire, and Windows suddenly realized iSCSI target is link-layer reachable!

Could this be a Windows specific glitch? Or is there a specification for iSCSI boot says if iSCSI target is L2 reachable, then do not pass gateway in iBFT?

NiKiZe, 2020-08-15:
It is documented in several places that this is known issue, and the recommendation is to remove gateway in iPXE before boot.

Adam5Wu, 2020-08-15:
Ah, I wish I came here and asked earlier, instead of pulling my hair for 2 days... LOL!

So, would something along the line of the following solve my problem:
> clear net0.dhcp/gateway:ipv4

mcb30, 2020-08-17:
Yes, that should work.

As far as I know, it's Windows that is misbehaving here: the gateway address is a property of the NIC structure in the iBFT and is independent of the iSCSI target.  It is possible to have multiple iSCSI targets in the iBFT, and it is therefore possible that one may be directly reachable while one requires routing via the gateway, which would give a situation in which the Windows behaviour provably cannot be correct.

closed 2020-08-17 by mcb30