ipxe/ipxe #133: Emulex NIC not working with latest ipxe.efi

Lee-Kwang, 2020-08-17:
Hi,
Emulex NIC in HPE Gen9  is not working with latest ipxe.efi or snp.efi.

Same issue was reported in this thread:
https://lists.ipxe.org/pipermail/ipxe-devel/2019-October/006786.html

I did git bisect as suggested in the above thread.
found that it started not to work with emulex driver since commit 6324227dcaa820436da4acd88ee174d3e26f0d54.

I built snp.efi with parent commit of 6324227 ( fdcdc5203b230fef23fa721573fd30ad093679d1) 
and it works for emulex dirver  and also works for Mellanox MT27710 driver in HPE Gen9 or DELL PowerEdge R640.

mcb30, 2020-08-17:
Thanks for the report.  Given the convoluted history of this issue (as documented in the code comments), the only solution I can think of is to first attempt initialisation with cable detection and then, if that fails, fall back to initialising without cable detection.

mcb30, 2020-08-17:
@Lee-Kwang Could you try the patch in PR #134 to see if this fixes the problem?

If so, could you please confirm that I have your name and e-mail address correct in the commit message?

Thanks.

Lee-Kwang, 2020-08-17:
@mcb30 
built new snp.efi using the patch and it works for emulex nic!
Yes, my name and email address are correct.

Thanks,

closed 2020-08-17 by mcb30