ipxe/ipxe #138: Access to Current Working URI (cwuri)

omniproc, 2020-09-03:
**Description**

As a user, I would like the option of accessing the URI the current iPXE was loaded from (cwuri) from an iPXE script.

**Acceptance criteria**

- [ ] Have a IPXE script variable to access the cwuri when booting iPXE over EFI HTTP boot
- [ ] Split the URI into it's components (protocol, hostname, port, path, query) which can be accessed individually

omniproc, 2021-01-12:
Rational behind this: when we have the option to know where the initial iPXE image came from at runtime of the iPXE script we can load additional ressources from that location without the need to hardcode the server / location in the iPXE script. This provides great flexibility for any dynamic chainloading process. A single image can then be used to have the same install procedure in different geographical locations / on-/off-prem and so on (distributed) where latency and bandwidth would make it  hard to serve dynamic data from a single source.

NiKiZe, 2021-01-12:
If you do imgfetch script.ipxe, that should already be relative, no need to get or mangle the URL?

omniproc, 2021-01-12:
Well, not quite. This works only for TFP or mounted filesystems but not if the image was loaded from an URL e.g. using EFI HTTP(S) boot.

NiKiZe, 2021-01-12:
In that case, how was the initial script to iPXE provided?

omniproc, 2021-01-12:
UEFI HTTP Boot. The URL to the initial image is provided to UEFI as a boot option. The initial image can contain a generic (generic to the extend that it doesn't need to include any static URLs / data but can load anything dynamicly depending on where it was initially loaded from) embedded ipxe script that can then chainload more specific scripts from a path relative to where the image came from.

stappersg, 2021-11-25:
For what it is worth:  I just stumbled upon http://boot.ipxe.org/tinycore.ipxe which has this content:
```text
#!ipxe

set base http://tinycorelinux.net/11.x/x86/release/distribution_files

cpuid --ext 29 && set arch 64 || set arch

kernel ${base}/vmlinuz${arch} initrd=rootfs.gz initrd=modules${arch}.gz
initrd ${base}/rootfs.gz
initrd ${base}/modules${arch}.gz
boot
```
in `${base}` do I see `${cwuri}`.

( Now _time will tell_ what the effect is of touching this stalled issue ... )

omniproc, 2021-11-25:
@stappersg 

no worries. I'm still watching this issue and hoping for it to be implemented.
The example you made does use simple variable definition and access. The variable `base` is created with a URL hardcoded in the iPXE script we're looking at. Then it's used. We're already doing that and the feature is documented for iPXE in the normal iPXE scripting guide.

The problem with that approach is that you have to hardcode your URI in your iPXE script. This restricts the use of your iPXE script / images to a single server. If you want to use the same image in another location you'd have to change the iPXE script. 

This feature request is about providing a method to provide access to a dynamic variable, let's call it `cwuri`, which will always return the `uri` the current iPXE script was loaded from. E.g. if the script was loaded from `https://my-ipxe-server/base/` then using `${cwuri}/some/next/image.gz` would be resolved to `https://my-ipxe-server/base/some/next/image.gz`. Yet, since the URI is not hardcoded you could place that same image on another server, say `https://my-other-ipxe-server/base/` and `${cwuri}/some/next/image.gz` would be resolved to `https://my-other-ipxe-server/base/some/next/image.gz`.

That's all this feature request is about.

stappersg, 2021-11-25:
On Thu, Nov 25, 2021 at 06:22:00AM -0800, Omniproc wrote:
> no worries.

Nope, the response did trigger worries.

> I'm still watching this issue and hoping for it to be implemented.

Advice for while doing that: 

Implement the wanted feature and publish the modification.

Suggestion for a cover letter:

Our business is flourishing and iPXE is part of that success.
Upon opening a new branch we realized how valuable the feature is
and switched from waiting for others doing it to doing it ourself.
It took blood, sweat, tears and various other resources,
but giving back made it extra worthwhile.

omniproc, 2021-11-25:
@stappersg 

didn't know feature requests are ment only for those having the skills to implement them. I'm doing my part here trying to explain as good as I can what the feature is and what benefits it would bring to iPXE. But sure, if that's unwanted in this project then consider this issue as closed.

NiKiZe, 2021-11-25:
@omniproc thank you for the request, Not sure if the information is actually available, how is the http url sent to the client in the first place? (DHCP?)

(sorry that you got that bad kind of response)

NiKiZe, 2021-11-25:
Oh never mind you already kind of answered this 😅

omniproc, 2021-11-25:
>  thank you for the request, Not sure if the information is actually available, how is the http url sent to the client in the first place? (DHCP?)

You're very welcome. I think the feature should be (if possible) independent of the way the http URL is set. It could be DHCP but in my case we're using UEFI HTTP boot. I know that many ppl. are unaware of UEFI HTTP boot so please let me know if you need more details but basically you can submit the URL to UEFI by whatever means your UEFI firmware supports (e.g. with HP servers you could do that using their OoBM named ILO). 

>(sorry that you got that bad kind of response)

Nothing to be sorry for. You're not responsible for stuff random GitHub users respond and I know that you guys who actually contributed alot to this project are thankful for feedback (confirmed multiple times in IRC already).

omniproc, 2024-12-13:
As discussed in #1168 the commits 88c2a01e1 and 390bce951 resolve the first part of this issue. 
https://ipxe.org/cfg/cwuri
https://ipxe.org/cfg/cwduri

What's still open is to have the URI parts split so e.g. even if the image was initially booted using `http`, so the original `${cwuri}` points to `http://example.com` the autor of the iPXE script might want to change the protocol to `https` for the next `chain --replace --autofree` command.

NiKiZe, 2026-01-11:
changing http -> https could be seen as string manipulation, and when that comes up we end up saying do that serverside.
So instead of the http request doing a Location: 30x redirect, it could return a text/plain `#!ipxe\nchain --autofreee https:/....` when the user-agent is iPXE.

Is there any other usecase where manipulation is needed, and the new settings or just relative chaining isn't enough?