ipxe/ipxe #142: IPXE boot on amd based architecture with NVME

Badb0yBadb0y, 2020-09-16:
We have a quite big infrastructure and now we started to use AMD based servers instead of the intel ones with 1 nvme only for os.

We are using ipxe to kickstart the servers but for some reason with this new amd nvme setup the ipxe fails during the kickstart

Here are the issue that I'm seeing:

https://i.ibb.co/VQksZTF/1.png

https://i.ibb.co/7Gh80BT/2020-09-15-16-34-23-Greenshot.png

Here is the pxelinux and the tftpboot config as well: https://jpst.it/2hIpU

If I type manually the # for grub 2.02

It will kickstart the system, but it should be automated :) somehow it doesn't go to that step.

NiKiZe, 2020-09-16:
For the sake of documentation.
picture shows EFI firmware chaining ipxe.efi commit 4e85b27
Finding NII interface
Downloads and starts ipxe script over tftp, which calls into pxelinux.cfg
ending up in grub with `error: symbol 'grub_calloc' not found.`

Have you verified that the same grub binary executes correctly from say a USB device (making sure to bypas iPXE and netboot entirely)
This does seem like an grub issue, not iPXE, so good to narrow it down first.

mcb30, 2020-09-16:
The screenshot shows a GRUB issue, not an iPXE issue.  You could try to debug the GRUB issue.

However, from looking at your iPXE script, it's not clear to me why you are using GRUB at all.  If you just add `initrd=initrd.img` to your kernel command-line arguments then your "Booting with legacy BIOS" section will work for both BIOS and UEFI.  This would eliminate the use of GRUB and would give you a simpler setup to maintain.

Badb0yBadb0y, 2020-09-16:
> For the sake of clarity: you should be able to simplify your script down
> to just:
>
> #!ipxe
>
> kernel http://10.118.192.250:80/cobbler/images/CentOS-7-1611-x86_64/vmlinuz  ksdevice=bootif lang=  bootdev=eno5 text ip=eno5:dhcp kssendmac  ks=http://10.118.192.250/cblr/svc/op/ks/system/servername-2163 initrd=initrd.img
> initrd http://10.118.192.250:80/cobbler/images/CentOS-7-1611-x86_64/initrd.img
> boot

Thank you so much your help, now it is moving forward a bit, now stuck here:

[image: image.png]

This is in my ipxe as you've suggested:

#!ipxe

kernel http://10.118.192.250:80/cobbler/images/CentOS-7-1611-x86_64/vmlinuz
 ksdevice=bootif lang=  bootdev=eno5 text ip=eno5:dhcp kssendmac  ks=
http://10.118.192.250/cblr/svc/op/ks/system/servername -2163
initrd=initrd.img
initrd
http://10.118.192.250:80/cobbler/images/CentOS-7-1611-x86_64/initrd.img
boot
~

and this is in my tftp grub:

  # generate linux ipxe here
  default=0
  timeout=0

  # for grub 0.97
  title  servername -2163
      root (nd)
      kernel /images/CentOS-7-1611-x86_64/vmlinuz  ksdevice=bootif lang=
 bootdev=eno5 text ip=eno5:dhcp kssendmac  ks=
http://10.118.192.250/cblr/svc/op/ks/system/servername -2163
      initrd /images/CentOS-7-1611-x86_64/initrd.img

  # for grub 2.02
  menuentry "grub2-servername-2163"{
      linuxefi /images/CentOS-7-1611-x86_64/vmlinuz  ksdevice=bootif lang=
 bootdev=eno5 text ip=eno5:dhcp kssendmac  ks=
http://10.118.192.250/cblr/svc/op/ks/system/servername -2163
      install=/cblr/links/CentOS-7-1611-x86_64
      initrdefi /images/CentOS-7-1611-x86_64/initrd.img
   }

I think very small step is missing only.

mcb30, 2020-09-17:
@Badb0yBadb0y The screenshot in your most comment is missing.  Could you please repost it?

Badb0yBadb0y, 2020-09-17:
Oh, yeah, looks like from email didn’t attach, here it is @mcb30 

[image]

mcb30, 2020-09-17:
I am able to reproduce a similar failure in QEMU using CentOS-7-1611 with a simpler iPXE script:
```
#ipxe
set base http://<server>/path/to/centos/mirror/7
kernel ${base}/images/pxeboot/vmlinuz initrd=initrd.img repo=${base}
initrd ${base}/images/pxeboot/initrd.img
boot
```
The problem does not occur with CentOS 8.  I have not yet tried any newer CentOS 7 updates.

Initial debugging indicates that something is corrupting the system memory map: the OVMF firmware is producing errors of the form:
```
ConvertPages: Incompatible memory types, the pages to allocate have been allocated
```

mcb30, 2020-09-22:
After adding extra debug capabilities to `efi_wrap.c` (commit e08ad61bf), some very strange behaviour is observable.

The kernel function `efi_convert_cmdline_to_ascii()` calls `efi_low_alloc()` to allocate a small amount of memory.  The logic in `efi_low_alloc()` retrieves the system memory map via the UEFI boot services call `GetMemoryMap()`, then iterates over each sufficiently large free memory region and calls `AllocatePages()` to attempt to allocate from that region.

On the previous call to `efi_low_alloc()` (from within `make_boot_params()`), everything works as expected.  On this call to `efi_low_alloc()`, the code somehow ends up calling `AllocatePages()` once per entry in the system memory map, but with the requested address always being zero, instead of being the start address of the system memory map region.

This ought not to be possible from an examination of the code.

mcb30, 2020-09-22:
Found the root cause!  `efi_convert_cmdline_to_ascii()` calls `efi_low_alloc()` with `align=0`, and the line
```c
start = round_up(start, align);
```
in `efi_low_alloc()` will therefore always end up setting `start` to zero.

Badb0yBadb0y, 2020-09-22:
Wow :) deep RCA, but this is a next level for me, I don’t speak C :)

mcb30, 2020-09-23:
It looks as though the bug was inadvertently fixed in kernel commit https://github.com/torvalds/linux/commit/38dd9c0 since this introduced a minimum alignment (and therefore eliminated the problem caused by using a zero alignment).

As far as I can tell, this commit has not been backported to any of the CentOS 7 kernels.

A sensible fix would be to either cherry-pick https://github.com/torvalds/linux/commit/38dd9c0 or to fix the requested alignment from zero (invalid) to one (i.e. byte alignment):
```diff
diff --git a/drivers/firmware/efi/efi-stub-helper.c b/drivers/firmware/efi/efi-stub-helper.c
index 20bec5fe3691..ed5e1db8ed08 100644
--- a/drivers/firmware/efi/efi-stub-helper.c
+++ b/drivers/firmware/efi/efi-stub-helper.c
@@ -484,7 +484,7 @@ static char *efi_convert_cmdline_to_ascii(efi_system_table_t *sys_table_arg,
 
        options_size++;  /* NUL termination */
 
-       status = efi_low_alloc(sys_table_arg, options_size, 0, &cmdline_addr);
+       status = efi_low_alloc(sys_table_arg, options_size, 1, &cmdline_addr);
        if (status != EFI_SUCCESS)
                return NULL;
```

@Badb0yBadb0y Want to try taking this up with RedHat?

Badb0yBadb0y, 2020-09-24:
You mean report it to redhat to backport to centos?

NiKiZe, 2021-01-17:
Should we close this as a kernel bug with a fix? Or is there anything that can (and should) be done in iPXE to reduce this issue?

Badb0yBadb0y, 2021-01-18:
Can close yes.

closed 2021-01-18 by NiKiZe