ipxe/ipxe #144: Extend cachedhcp to handle the proxydhcp packet

alkisg, 2020-09-25:
Suppose that I boot from the PXE stack, which loads a custom build of undionly.kpxe with the following embedded ipxe script:

```shell
ifopen net0
chain ${proxydhcp/dhcp-server}/ltsp/ltsp.ipxe
```

The problem is that at that point, the proxydhcp variables aren't there; they're only available if a `dhcp net0` gets executed.

When using proxydhcp, there's little benefit in doing an additional DHCP request, as we have no control over the main DHCP server anyway, and our only variables that we'll need are ${proxydhcp/dhcp-server} and ${proxydhcp/filename}.

Would it be possible to get the ProxyDHCP packet and make it available at that point, to avoid the extra DHCP request?
Thank you again for the wonderful iPXE!

Hammarskjold, 2020-09-25:
Talked to @mcb30 many times about this. Should be doable, just haven't been prioritized, but I agree it's a good thing as we can skip all extra DHCP requests when using a proxyDHCP. But for this to work also for EFI, if I remember correctly, the EFI subsystem doesn't cache the DHCP response at all. @mcb30 should we open up a separate issue on this?

mcb30, 2020-09-25:
@Hammarskjold Yes, the UEFI cached DHCP support should probably be a separate issue.

mcb30, 2020-12-28:
> @Hammarskjold Yes, the UEFI cached DHCP support should probably be a separate issue.

Created as #188

mcb30, 2021-07-27:
Implemented for UEFI (but not yet legacy BIOS) in commit https://github.com/ipxe/ipxe/commit/e09e1142a

alkisg, 2022-09-27:
Thanks for the UEFI implementation!
Any progress with the BIOS implementation?
I think it would also help in cases where the second DHCP fails (e.g. https://ipxe.org/err/4c1060 and cisco switches), and we resort to using pxelinux.0...