ipxe/ipxe #147: Can't assemble TLS cert chain for *.herokuapp.com -> DigiCert root

mcsaucy, 2020-09-28:
Howdy,

I've got an iPXE script that attempts to hit a Heroku webapp, which redirects me to a URL that has the actual content I need. Unfortunately, this it seems like we're unable to trust Heroku's star cert. https://ipxe.org/crypto indicates that anything that "In the default configuration, iPXE will therefore automatically trust the same set of certificates as the Firefox web browser." Firefox likes these, but iPXE doesn't.

Any idea what could be at play or where I could start digging?

My build is pretty simple: `#define DOWNLOAD_PROTO_HTTPS` in `src/config/general.h`, `make bin/ipxe.usb`.

### Attempting to hit anything.herokuapp.com:
```
iPXE> kernel https://foo.herokuapp.com
https://foo.herokuapp.com...
X509 chain 0xeec04 added X509 0xf4234 "*.herokuapp.com"
X509 chain 0xeec04 added X509 0xf4ed4 "DigiCert SHA2 High Assurance Server CA"
X509 chain 0xeec04 added X509 0xf5444 "DigiCert High Assurance EV Root CA"
X509 chain 0xeec04 found no usable certificates
Permission denied (http://ipxe.org/0216eb3c)
```

### Issue seems to be trusting DigiCert:
```
iPXE> kernel https://digicert.com
https://digicert.com...
X509 chain 0xeec04 added X509 0xfdaf4 "digicert.com"
X509 chain 0xeec04 added X509 0xf7614 "DigiCert SHA2 Extended Validation Server CA"
X509 chain 0xeec04 added X509 0xf5444 "DigiCert High Assurance EV Root CA"
X509 chain 0xeec04 found no usable certificates
Permission denied (http://ipxe.org/0216eb3c)
```
### At least one other CA works:
```
iPXE> kernel https://ipxe.org
https://ipxe.org...X509 chain 0xeeab4 added X509 0xf7ff4 "ipxe.org"
X509 chain 0xeeab4 added X509 0xfd024 "Let's Encrypt Authority X3"
<...>
X509 0xf7ff4 "ipxe.org" requires an OCSP check
X509 0xf7ff4 "ipxe.org" successfully validated using issuer 0xfd024 "Let's Encrypt Authority X3"
ok
```
### Firefox seems to like Heroku
So either my Firefox setup is wonky or ipxe.org/crypto is incorrect.
[image]

mcsaucy, 2020-09-29:
Finally got the ["Mozilla list of public CA certificates"](http://mxr.mozilla.org/comm-central/source/mozilla/security/nss/lib/ckfw/builtins/certdata.txt) file to load (it was timing out yesterday; it redirects to [this](https://hg.mozilla.org/mozilla-central/raw-file/tip/security/nss/lib/ckfw/builtins/certdata.txt), which seems to more reliably load).

Anyway, it seems to include the root cert I care about. Checking in firefox, the serial in `certdata.txt` matches the root for the chain that Firefox assembles, so that seems to rule out my Firefox install being wonky.

mcsaucy, 2020-10-01:
Attaching [ipxe_sad.txt](https://github.com/ipxe/ipxe/files/5315038/ipxe_sad.txt), with verbose x509 logging, pulled from qemu.

Downside, it's got a lot of terminal escape sequences in there so it'll be messy in a text editor. 

Upside, `cat` it for pretty colors.

mcsaucy, 2020-10-04:
I  haven't been able to reproduce pulling a kernel (which seems to have a very strong resemblance to html... how about that) from https://digicert.com, so I'm gonna guess that case was either a transient issue or I got some wires crossed when pruning that output. Made some decent progress, though.

I've applied some local changes to vomit some more debugging message. If we try to hit `https://foo.herokuapp.com`, we get:
```
TLS in tls parse chain
TLS 0xef004 found certificate *.herokuapp.com
TLS 0xef004 found certificate DigiCert SHA2 High Assurance Server CA
TLS 0xef004 found certificate DigiCert High Assurance EV Root CA (<--- !!!!!!)
TLS 0xef004 we done saying hello
VALIDATOR happy birthday
VALIDATOR 0xef404 "*.herokuapp.com" validating X509 chain 0xeda74
VALIDATOR step
VALIDATOR no ocsp
VALIDATOR 0xef404 "DigiCert High Assurance EV Root CA" is self-signed; we're not
 going to try to find a cross-signed cert.
TLS 0xef004 certificate validation failed: Permission denied (http://ipxe.org/02
16eb3c)
```
(emphasis mine). Note that the server gives us the *full* chain, all the way to the root (which is self-signed). Now, there's a cross-signed version of that self-signed cert that would work just fine here, but we [don't even try](https://github.com/ipxe/ipxe/blob/c70b3e04e86cefca335e36f883829d89583a6921/src/net/validator.c#L591-L606) to grab it.

If we try to hit `https://www.digicert.com`, we get:
```
TLS in tls parse chain
TLS 0xef004 found certificate digicert.com
TLS 0xef004 found certificate DigiCert SHA2 Extended Validation Server CA
TLS 0xef004 we done saying hello
VALIDATOR happy birthday
VALIDATOR 0xef3e4 "digicert.com" validating X509 chain 0xeda84
VALIDATOR step
VALIDATOR no ocsp
VALIDATOR let's download some shit for 'DigiCert SHA2 Extended Validation Server
 CA'
VALIDATOR in validator start download
VALIDATOR 0xef3e4 "digicert.com" downloading issuer of "DigiCert SHA2 Extended V
alidation Server CA"'s cross-signature from http://ca.ipxe.org/auto/582fb9d4.der
```
Since this server doesn't provide the root in its chain, iPXE tries to complete the chain with its cross-signed roots and then things largely work.

---------------------------

Seems like we should try to find cross-signed alternatives before we give up on self-signed certs. In my testing, removing [this block](https://github.com/ipxe/ipxe/blob/c70b3e04e86cefca335e36f883829d89583a6921/src/net/validator.c#L591-L598) makes things work. I'll throw a PR out there for discussion.

mcsaucy, 2020-10-05:
Additional note from [RFC 8446 § 4.4.2](https://tools.ietf.org/html/rfc8446#section-4.4.2):

>  Because certificate validation requires that trust anchors be distributed independently, a certificate that specifies a trust anchor MAY be omitted from the chain, provided that supported peers are known to possess any omitted certificates.

So yeah, if my understanding is correct and the issue lies with how we handle root-bearing cert chains, then the present behavior does align with the RFC, as "trust anchor MAY be omitted" certainly seems to imply "trust anchor MAY NOT be omitted".

mcb30, 2024-02-15:
> So yeah, if my understanding is correct and the issue lies with how we handle root-bearing cert chains, then the present behavior does align with the RFC, as "trust anchor MAY be omitted" certainly [...]

Your understanding is correct, and this should now be fixed via #1152

closed 2024-02-15 by mcb30