ipxe/ipxe #149: SNP failure when chainloading from U-Boot

ljfranklin, 2020-09-29:
[Cross-posting](https://forum.ipxe.org/showthread.php?tid=30714) from iPXE forum for added visibility

## Issue

Using the setup below, I'm trying to chainload iPXE from U-Boot on a small ARM32 board (ODroid MC1). iPXE is able to see the SNP interface with `ifstat`, but running `ifopen net0` shows `NETDEV net0 failed to receive 0x0: The socket is not connected` and the iPXE shell freezes. Full log is shown [here](https://gist.github.com/ljfranklin/99d8c4560d2ddf9e01a49c45bee5c09c#file-ipxe-log) and iPXE config page is [here](https://raw.githubusercontent.com/ljfranklin/ipxe/odroid-snp-failure/ipxe-config.png) and [here](https://raw.githubusercontent.com/ljfranklin/ipxe/odroid-snp-failure/ipxe-net0-config.png).

Any tips for debugging this further? Thanks y'all!

## Setup

Board: [ODroid MC1](https://www.hardkernel.com/shop/odroid-mc1-solo/)
CPU: ARM32 Samsung Exynos5422
iPXE build: bin-arm32-efi/snp.efi with lots of DEBUG enabled, [script](https://gist.github.com/ljfranklin/99d8c4560d2ddf9e01a49c45bee5c09c#file-compile-ipxe-sh)
iPXE source: latest master but had to disable a couple optional features to avoid freezes, [diff](https://github.com/ljfranklin/ipxe/commit/d7e93c9c8b0aa8c7da63f10d9ec486168d9df4c7)
U-Boot build: [script](https://gist.github.com/ljfranklin/7ae0109b7850effbbb77dc0064d07074)

mcb30, 2021-01-01:
I note that you have added `-mno-unaligned-access` to your build flags.  iPXE (along with UEFI) requires that unaligned accesses are permitted.  There will be code within iPXE that assumes it is safe to take the address of a packed structure member, thereby producing a pointer that the compiler will subsequently treat as aligned.

NiKiZe, 2021-07-09:
From what I can understand from http://u-boot.10912.n7.nabble.com/PATCH-arm-enable-unaligned-access-on-ARMv7-td126385.html this issue is resolved in u-boot?

closed 2022-05-27 by NiKiZe