ipxe/ipxe #151: cannot load a larger initramfs

jyf007, 2020-10-01:
platform:qemu and deskmini H110/COM
cpu i7-7700
ram up to 32 gig

i build a ipxe http boot site at links below
http://xz-nat.uovz.com:10423/
with 8 script from menu0.ipxe to menu7.ipxe
but i always failed with a large initramfs to unpack to boot entire space in ram.

NiKiZe, 2020-10-01:
Any error message?
Running pcbios or efi?

jyf007, 2020-10-02:
qemu doesn't show anything
deskmini cannot shot a screenshot, just VFS mount error

jyf007, 2020-10-02:
both of platform is uefi

NiKiZe, 2020-10-02:
There is not enough information here to try and help you.
General VFS errors is most of the time due to incorrect scripts or init files. and no information has been provided around that at all.

The FAQ on the forum probably have this case covered.

NiKiZe, 2020-10-02:
I will close this issue since there is no evidence that this is an issue with iPXE itself.
If there is more information to reproduce this, such as ipxe script and all other needed files, then we will reopen.

closed 2020-10-02 by NiKiZe

jyf007, 2020-10-03:
i have shoted the pictures
[image]
[image]
[image]

NiKiZe, 2020-10-04:
Still No script, does the same files work if you boot locally with say grub from usb? Also does your kernel have the compression algo supported?

jyf007, 2020-10-04:
the context in menu0.ipxe file is here:
#!ipxe
kernel vmlinuz-linux-ramos
initrd rootfs-openrc-base.cpio.lz4
boot
and grub with x86_64-efi work proper

mcb30, 2020-10-04:
@jyf007 You will need to add `initrd=rootfs-openrc-base.cpio.lz4` to your kernel command line to tell it which file to open as the initramfs image.  For example:

```
#!ipxe
kernel vmlinuz-linux-ramos initrd=rootfs-openrc-base.cpio.lz4
initrd rootfs-openrc-base.cpio.lz4
boot
```

Also, you are using what looks to be a very old version of iPXE: please consider using an up-to-date version.

jyf007, 2020-10-05:
thanks,and up to date version is always building failed

jyf007, 2020-10-05:
well,i tried the up-to-date version,and failed

jyf007, 2020-10-08:
update -- failed in qemu