ipxe/ipxe #155: Fail to boot to winpe (UEFI, x64, non-secure boot)

aheusser, 2020-10-08:
Problem: I am unable to boot all the way into winpe successfully (as of c70b3e04e86cefca335e36f883829d89583a6921). The same setup using the previous commit works.

Setup:
- PXE client is ESXi v11 VM; 64-bit, UEFI, non-secure boot enabled
- chainloading ipxe.efi (using self-built `make bin-x86_64-sb/ipxe.efi`)
- using wimboot to provide a winpe WIM image

Problem:
The iPXE firmware chainloads and successfully downloads the winpe wim. But it only briefly shows the Windows logo screen ... and never shows the swirling dots below it). After a few seconds, the machine shuts down.

The exact same setup using the previous commit (fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb) works as expected (i.e., I see the swirling dots for a moment ... and then boot all the way into winpe).  

Also note that the problem is only exhibiting for me in the above scenario. 32 and 64-bit BIOS boots work, and 32-bit UEFI works as well. The only scenario that seems to break with this commit is 64-bit UEFI.

Please let me know if there is anything else I can do to provide insight into the problem.

NiKiZe, 2020-10-08:
just for reference and comparassion,
could you grab the full output of `imgstat` just before `boot` in your script?
and if possible the relevant parts of the script as well.

aheusser, 2020-10-08:
Hello. Thanks for the suggestion. Here is some of the information you indicated would be helpful.

The script snippet
`#!ipxe

# load the cfg file to obtain extra script configuration values
chain --autofree KCMTEMP_02D971FD0E81234BA78AE1F2F13C99A5.ipxe.cfg

iseq ${boot-pause} 1 && kernel wimboot pause || kernel wimboot
initrd --name winpeshl.ini        ${winpeshl.ini-filename}                         winpeshl.ini
initrd --name TSBootShellPrep.cmd TSBootShellPrep.cmd                              TSBootShellPrep.cmd
initrd --name BCD                 BCD                                              BCD
initrd --name boot.sdi            boot.sdi                                         boot.sdi
initrd --name boot.wim            packages/${boot-pkg-content-id}/${wim-filename}  boot.wim
iseq ${boot-pause} 1 && imgstat || goto wimboot-prompt

:wimboot-prompt
iseq ${boot-pause} 1 && prompt || goto wimboot-boot

:wimboot-boot
boot || goto failed
`

The imgstat and following from the working example (using fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb):

[image]
[image]

The imgstat and folliowing from the non-working example (using fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb):

[image]
[image]

mcb30, 2020-10-12:
@aheusser Thanks for the report.  Could you try building iPXE with `DEBUG=efi_driver:3`?

aheusser, 2020-10-13:
Hello @mcb30 per your request I rebuilt with `DEBUG=efi_driver:3`

With https://github.com/ipxe/ipxe/commit/fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb (working example), following is what I see on the winpe logo screen
[image]

And then with the very next commit (https://github.com/ipxe/ipxe/commit/c70b3e04e86cefca335e36f883829d89583a6921)  (broken example), following is what I see on the winpe logo screen (just before the VM gives up and turns itself off):

[image]

NOTE: I see that the HEAD of master has now moved on a bit since my original report. I also tried the same environment/setup with latest (https://github.com/ipxe/ipxe/commit/02748d0a584f22a568347d719291725e6fd63029) ... and I see the same failure.

mcb30, 2020-10-27:
@aheusser It is possible that commit https://github.com/ipxe/ipxe/commit/5b41b9a80 will have fixed the problem.  Could you please retest with current master?  Thanks.

aheusser, 2020-10-27:
@mcb30 thanks for letting me know. Absolutely, I will retest with latest master and report back.

NiKiZe, 2020-10-30:
@aheusser any chance you have been able to test?

aheusser, 2020-10-30:
@NiKiZe ... apologies, I have not had the chance yet. It is on my list. I will try to get to it today ... and will report back either way.

aheusser, 2020-11-03:
@mcb30 @NiKiZe I have re-tested and have the following results for you:

1) Reconfirmed that in my test setup fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb still functions as expected (i.e., WinPE boots successfully on 64-bit UEFI non-secure boot ESXi v11 VM)
2) Reconfirmed that in my test setup c70b3e04e86cefca335e36f883829d89583a6921 (the very next commit) still **breaks** my WinPE boot
3) Tested with 86c6c79fcdc00fef78373d3f0d35cb749d6b6772 (the commit **just before** the one that @mcb30 indicated might fix the problem); observed the same **failure** during WinPE boot (anticipated result)
4) Tested with 5b41b9a80ffb365376d8d522675a8d248a8717ab (the **exact** commit @mcb30 indicated might fix my observed issue); observed the same **failure** during WinPE boot
5) Tested with HEAD (9b25f6e5cf517f426de80ede618398ef01e385f9); observed the same **failure** during WinPE boot

So, it appears that 5b41b9a80ffb365376d8d522675a8d248a8717ab did not fix the problem, as was hoped.

aheusser, 2020-11-03:
NOTE: using 5b41b9a80ffb365376d8d522675a8d248a8717ab I re-built using `DEBUG=efi_driver:3` ... captured the following output just prior to the boot failure.

[image]

It appears to be the same debug output that I observed with the original breaking commit (c70b3e04e86cefca335e36f883829d89583a6921).

aheusser, 2020-11-03:
Taking @mcb30 's [comment from issue #164](https://github.com/ipxe/ipxe/issues/164#issuecomment-719822547) as hint/inspiration, I used iPXE to load `http://boot.ipxe.org/Shell.efi` in both the working commit (fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb) and the broken one (c70b3e04e86cefca335e36f883829d89583a6921).

Once in the shell, as suggested, I ran `devtree`, `devices`, and `drivers` to attempt to establish a difference.  I found a few:

In the broken version, I saw entries with a "#D" column for the following:
- "MNP Network Service Driver"
- "IP4 CONFIG Network Service Driver"
- "IP6 Network Service Driver"

These entries **do not** have "#D" column value in the prior **working** commit (fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb).

On a whim, I modified `interface/efi/efi_blacklist.c` on HEAD (9b25f6e5cf517f426de80ede618398ef01e385f9) to blacklist "MNP Network Service Driver" and tried my test once again. My boot **worked**.

My layman's supposition is that it appears that the logic change introduced in c70b3e04e86cefca335e36f883829d89583a6921 is somehow allowing "MNP Network Service Driver" to load, whereas is wasn't getting loaded before ... and this is what is inhibiting my WinPE boot to complete.

mcb30, 2020-11-06:
> My layman's supposition is that it appears that the logic change introduced in [c70b3e0](https://github.com/ipxe/ipxe/commit/c70b3e04e86cefca335e36f883829d89583a6921) is somehow allowing "MNP Network Service Driver" to load, whereas is wasn't getting loaded before ... and this is what is inhibiting my WinPE boot to complete.

It won't be causing the driver to load, but it will be causing the MnpDxe driver to bind to the handle exposed by iPXE.

> On a whim, I modified `interface/efi/efi_blacklist.c` on HEAD ([9b25f6e](https://github.com/ipxe/ipxe/commit/9b25f6e5cf517f426de80ede618398ef01e385f9)) to blacklist "MNP Network Service Driver" and tried my test once again. My boot **worked**.

Unfortunately globally blacklisting MnpDxe is not a viable solution, since this driver is required to allow the UEFI network stack to use the driver handles exposed by iPXE (e.g. in a situation where the user has an iPXE ROM but is trying to use the less-friendly UEFI built-in network stack to perform the boot).

Is the problem limited to booting into WinPE, or does it also arise when performing a generic Linux kernel+initrd boot (e.g. `chain http://boot.ipxe.org/demo/boot.php`)?

aheusser, 2020-11-07:
@mcb30 For sure, I was not intending to suggest that globally blocking MnpDxe was the correct solution. I was merely trying to report out that:
1) in the pre-recursion commit, my WinPE boot worked and the "MNP Network Service Driver", "IP4 CONFIG Network Service Driver", and "IP6 Network Service Driver" did not have "#D" entries in the `drivers` in my environment
2) in the post-recursion commit, they do have entries
3) blocking "MnpDxe" allowed my particular boot environment to succeed in the post-recursion commit codebase

What the actual correct fix should be, I have no idea ... as I don't have knowledge about MnpDxe.

In the next couple days I will see what I can do about responding to your question about generic Linux kernel+initrd boot ... and if I can find a reasonable way to share my setup with you (might require sharing the WinPE image too ... which is large).

aheusser, 2020-11-09:
@mcb30 I have encountered what appears to be the same problem when performing a generic Linux kernel+initrd boot  (e.g. chain http://boot.ipxe.org/demo/boot.php)?

[image]

Lukas-UAUX, 2020-11-17:
+1  by me, my iPXE has the same issue with both wimboot and linux. I will try to checkout the mentioned commit as a workaround...

aheusser, 2020-11-17:
Please see overlapping discussion in https://github.com/ipxe/ipxe/issues/164

Per my comment on 11/17, at this time I don't think I have a reliable repro other than using my one specific problematic VM. So perhaps this issue should be closed.

Lukas-UAUX, 2020-11-17:
Well, I do - I used a hyper-v gen2 2016 and deleted and re-created it like a dozen times and I also tried some random fujitsu esprimo - all behave exactly as your machine, and reverting to the mentioned commit works

aheusser, 2020-11-17:
@Lukas-UAUX  Excellent! As I said before, I am not able to re-CREATE a faulty VM that shows the problem ... but I do have one specific VM that shows the symptom.  But my environment is ESXi.

mcb30, 2020-11-23:
@aheusser @Lukas-UAUX This should now be fixed.  Please reopen the issue if needed.  Thank you!

closed 2020-11-23 by mcb30

aheusser, 2020-11-24:
Thank you @mcb30.

FYI: I just tested with HEAD of master (1295b4acff1f2014261c40d9f9d2107ffd668d92).  My previously failing test VM works again.

mcb30, 2020-11-24:
> FYI: I just tested with HEAD of master ([1295b4a](https://github.com/ipxe/ipxe/commit/1295b4acff1f2014261c40d9f9d2107ffd668d92)). My previously failing test VM works again.

Fantastic: thanks for letting me know!

PR #170 (2020-11-23): [efi] Allow initialisation via SNP interface even while claimed
Fixes: #164 
Fixes: #155 
Fixes: #159