ipxe/ipxe #156: Intel x550 nic not found

jordan3825, 2020-10-09:
Hello, 

I have a Supermicro X11SPG-TF motherboard with two Intel x550 nics. When booting iPXE neither of the nics are seen by iPXE. The device ID is located in intelx.c source file. 

https://github.com/ipxe/ipxe/blob/eecb75ba4809bd5d3e6d63413a45c5fccbda2bc2/src/drivers/net/intelx.c#L476

Here is the command I am running to build iPXE:

```
make bin-x86_64-efi/ipxe.usb DEBUG=intelx,intelxvf,main,dhcpopts,dhcppkt,intelxl
```

I have attached a screen shot of iPXE running and a screenshot of the NIC information from the BIOS. 

Any suggestions on how to debug this issue?

[image]
[image]

toreanderson, 2020-10-19:
I have the same problem, I believe. In my case, the server hardware is Intel S2600BPS (firmware version `02.01.0011`) with an Intel X722-DA2 OCP mezzanine NIC (firmware version `4.10`). The network interfaces are not detected by iPXE; `ifstat` yields no output, and the boot fails.

According to `git bisect`, the problem was introduced by commit 27e886c67b63b9a381f87238ed7a856775199fd9. Current Git master (6ccd5239b13ca92800ea81c2d6489e0ad8e9d5e6) plus `git revert 27e886c` works fine. @jordan3825, can you please test reverting that commit to confirm that we're seeing the same issue?

This is the console output shown when attempting to boot 27e886c67b63b9a381f87238ed7a856775199fd9, having built iPXE with `DEBUG=efi_driver,snpnet`:

```
>>Start PXE over IPv4.
  Station IP address is 10.27.88.85

  Server IP address is 87.238.33.6
  NBP filename is ipxe.efi
  NBP filesize is 985504 Bytes
 Downloading NBP file...

  NBP file downloaded successfully.
iPXE initialising devices...EFIDRV connecting our drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) has driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) disconnecting existing drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) connecting new drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) has driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) DRIVER_START
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) could not start driver "PCI": No such device (http://ipxe.org/2ccb4087)
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) is not supported by driver "NII": Error 0x7f684083 (http://ipxe.org/7f684083)
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) is not supported by driver "SNP": Error 0x7f684083 (http://ipxe.org/7f684083)
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) has driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) disconnecting existing drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) connecting new drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) has driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) DRIVER_START
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) could not start driver "PCI": No such device (http://ipxe.org/2ccb4087)
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) is not supported by driver "NII": Error 0x7f684083 (http://ipxe.org/7f684083)
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) is not supported by driver "SNP": Error 0x7f684083 (http://ipxe.org/7f684083)
ok

iPXE 1.20.1+ (g27e88) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

iPXE> ifstat                             
iPXE> 
```

For comparison, this is the console output from the last commit that works correctly (eecb75ba4809bd5d3e6d63413a45c5fccbda2bc2):

```
>>Start PXE over IPv4.
  Station IP address is 10.27.88.85

  Server IP address is 87.238.33.6
  NBP filename is ipxe.efi
  NBP filesize is 984480 Bytes
 Downloading NBP file...

  NBP file downloaded successfully.
iPXE initialising devices...EFIDRV connecting our drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) has driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) disconnecting existing drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) connecting new drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) has driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) DRIVER_START
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x0) using driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) has driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) disconnecting existing drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) connecting new drivers
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) has driver "PCI"
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) DRIVER_START
EFIDRV PciRoot(0x1)/Pci(0x0,0x0)/Pci(0x0,0x0)/Pci(0x3,0x0)/Pci(0x0,0x1) using driver "PCI"
ok

iPXE 1.20.1+ (geecb7) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

iPXE> ifstat                             
net0: a4:bf:01:60:f8:d2 using x722-sfp on 0000:1a:00.0 (closed)
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
net1: a4:bf:01:60:f8:d3 using x722-sfp on 0000:1a:00.1 (closed)
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
iPXE> 
```

jordan3825, 2020-10-20:
Hello! I can confirm that reverting `27e886c` fixed the problem with detecting the x550 nics.

mcb30, 2020-10-26:
@toreanderson @jordan3825 Could you please try building with `DEBUG=efi_pci` to see what is causing the error in `efipci_ioremap()`?  Thanks.

toreanderson, 2020-10-29:
@mcb30 Here you go:
```
>>Start PXE over IPv4.
  Station IP address is 10.27.88.85

  Server IP address is 87.238.33.6
  NBP filename is ipxe.efi
  NBP filesize is 979488 Bytes
 Downloading NBP file...

  NBP file downloaded successfully.
iPXE initialising devices...EFIPCI 0000:00:00.0 (8086:2020 class 060000) has no driver
EFIPCI 0000:00:04.0 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.1 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.2 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.3 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.4 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.5 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.6 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.7 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:05.0 (8086:2024 class 088000) has no driver
EFIPCI 0000:00:05.1 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:05.2 (8086:2025 class 088000) has no driver
EFIPCI 0000:00:05.4 (8086:2026 class 080020) has no driver
EFIPCI 0000:00:05.6 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:07.0 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:07.4 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:07.7 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:08.0 (8086:2014 class 088000) has no driver
EFIPCI 0000:00:08.1 (8086:2015 class 110100) has no driver
EFIPCI 0000:00:08.2 (8086:2016 class 088000) has no driver
EFIPCI 0000:00:11.0 (8086:a1ec class ff0000) has no driver
EFIPCI 0000:00:11.1 (8086:a1ed class ff0000) has no driver
EFIPCI 0000:00:11.5 (8086:a1d2 class 010601) has no driver
EFIPCI 0000:00:14.0 (8086:a1af class 0c0330) has no driver
EFIPCI 0000:00:14.2 (8086:a1b1 class 118000) has no driver
EFIPCI 0000:00:16.0 (8086:a1ba class 078000) has no driver
EFIPCI 0000:00:16.1 (8086:a1bb class 078000) has no driver
EFIPCI 0000:00:16.4 (8086:a1be class 078000) has no driver
EFIPCI 0000:00:17.0 (8086:a182 class 010601) has no driver
EFIPCI 0000:00:1c.0 (8086:a195 class 060400) has no driver
EFIPCI 0000:01:00.0 (1a03:1150 class 060400) has no driver
EFIPCI 0000:02:00.0 (1a03:2000 class 030000) has no driver
EFIPCI 0000:00:1f.0 (8086:a1c2 class 060100) has no driver
EFIPCI 0000:00:1f.2 (8086:a1a1 class 058000) has no driver
EFIPCI 0000:00:1f.4 (8086:a1a3 class 0c0500) has no driver
EFIPCI 0000:17:00.0 (8086:2030 class 060400) has no driver
EFIPCI 0000:18:00.0 (8086:37c0 class 060400) has no driver
EFIPCI 0000:19:03.0 (8086:37c5 class 060400) has no driver
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 9f000000+200000 is not within root bridge address space
EFIPCI 0000:1a:00.0 could not probe driver "x722-sfp": No such device (http://ipxe.org/2ccb4087)
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 9e000000+200000 is not within root bridge address space
EFIPCI 0000:1a:00.1 could not probe driver "x722-sfp": No such device (http://ipxe.org/2ccb4087)
EFIPCI 0000:17:05.0 (8086:2034 class 088000) has no driver
EFIPCI 0000:17:05.2 (8086:2035 class 088000) has no driver
EFIPCI 0000:17:05.4 (8086:2036 class 080020) has no driver
EFIPCI 0000:17:05.6 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.0 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.1 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.2 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.4 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.7 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:08.0 (8086:208d class 088000) has no driver
[... 380 lines cut ...]
EFIPCI 0001:d7:12.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:12.4 (8086:204c class 110100) has no driver
EFIPCI 0001:d7:12.5 (8086:204d class 110100) has no driver
EFIPCI 0001:d7:12.6 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:12.7 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:15.0 (8086:2018 class 088000) has no driver
EFIPCI 0001:d7:15.1 (8086:2088 class 110100) has no driver
EFIPCI 0001:d7:15.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:16.0 (8086:2018 class 088000) has no driver
EFIPCI 0001:d7:16.1 (8086:2088 class 110100) has no driver
EFIPCI 0001:d7:16.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:16.4 (8086:2018 class 088000) has no driver
EFIPCI 0001:d7:16.5 (8086:2088 class 110100) has no driver
EFIPCI 0001:d7:16.7 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:17.0 (8086:2018 class 088000) has no driver
EFIPCI 0001:d7:17.1 (8086:2088 class 110100) has no driver
EFIPCI 0001:d7:17.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
ok

iPXE 1.20.1+ (g27e88) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

iPXE> ifstat                             
iPXE> 
```

toreanderson, 2020-10-29:
By the way, it would appear that c70b3e04e86cefca335e36f883829d89583a6921 *kind-of* fixed this issue. The NICs are detected, but they're now using the NII driver instead of the x722-sfp driver (wich was last used in eecb75ba4809bd5d3e6d63413a45c5fccbda2bc2). See below for debug output from both cases.

## Debug output as of c70b3e04e86cefca335e36f883829d89583a6921

```
>>Start PXE over IPv4.
  Station IP address is 10.27.88.85

  Server IP address is 87.238.33.6
  NBP filename is ipxe.efi
  NBP filesize is 979488 Bytes
 Downloading NBP file...

  NBP file downloaded successfully.
iPXE initialising devices...EFIPCI 0000:00:00.0 (8086:2020 class 060000) has no driver
EFIPCI 0000:00:04.0 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.1 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.2 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.3 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.4 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.5 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.6 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:04.7 (8086:2021 class 088000) has no driver
EFIPCI 0000:00:05.0 (8086:2024 class 088000) has no driver
EFIPCI 0000:00:05.1 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:05.2 (8086:2025 class 088000) has no driver
EFIPCI 0000:00:05.4 (8086:2026 class 080020) has no driver
EFIPCI 0000:00:05.6 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:07.0 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:07.4 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:07.7 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:00:08.0 (8086:2014 class 088000) has no driver
EFIPCI 0000:00:08.1 (8086:2015 class 110100) has no driver
EFIPCI 0000:00:08.2 (8086:2016 class 088000) has no driver
EFIPCI 0000:00:11.0 (8086:a1ec class ff0000) has no driver
EFIPCI 0000:00:11.1 (8086:a1ed class ff0000) has no driver
EFIPCI 0000:00:11.5 (8086:a1d2 class 010601) has no driver
EFIPCI 0000:00:14.0 (8086:a1af class 0c0330) has no driver
EFIPCI 0000:00:14.2 (8086:a1b1 class 118000) has no driver
EFIPCI 0000:00:16.0 (8086:a1ba class 078000) has no driver
EFIPCI 0000:00:16.1 (8086:a1bb class 078000) has no driver
EFIPCI 0000:00:16.4 (8086:a1be class 078000) has no driver
EFIPCI 0000:00:17.0 (8086:a182 class 010601) has no driver
EFIPCI 0000:00:1c.0 (8086:a195 class 060400) has no driver
EFIPCI 0000:01:00.0 (1a03:1150 class 060400) has no driver
EFIPCI 0000:02:00.0 (1a03:2000 class 030000) has no driver
EFIPCI 0000:00:1f.0 (8086:a1c2 class 060100) has no driver
EFIPCI 0000:00:1f.2 (8086:a1a1 class 058000) has no driver
EFIPCI 0000:00:1f.4 (8086:a1a3 class 0c0500) has no driver
EFIPCI 0000:17:00.0 (8086:2030 class 060400) has no driver
EFIPCI 0000:18:00.0 (8086:37c0 class 060400) has no driver
EFIPCI 0000:19:03.0 (8086:37c5 class 060400) has no driver
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 9f000000+200000 is not within root bridge address space
EFIPCI 0000:1a:00.0 could not probe driver "x722-sfp": No such device (http://ipxe.org/2ccb4087)
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 9e000000+200000 is not within root bridge address space
EFIPCI 0000:1a:00.1 could not probe driver "x722-sfp": No such device (http://ipxe.org/2ccb4087)
EFIPCI 0000:17:05.0 (8086:2034 class 088000) has no driver
EFIPCI 0000:17:05.2 (8086:2035 class 088000) has no driver
EFIPCI 0000:17:05.4 (8086:2036 class 080020) has no driver
EFIPCI 0000:17:05.6 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.0 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.1 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.2 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0000:17:07.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
[... 868 lines cut ...]
EFIPCI 0001:d7:12.7 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:15.0 (8086:2018 class 088000) has no driver
EFIPCI 0001:d7:15.1 (8086:2088 class 110100) has no driver
EFIPCI 0001:d7:15.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:16.0 (8086:2018 class 088000) has no driver
EFIPCI 0001:d7:16.1 (8086:2088 class 110100) has no driver
EFIPCI 0001:d7:16.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:16.4 (8086:2018 class 088000) has no driver
EFIPCI 0001:d7:16.5 (8086:2088 class 110100) has no driver
EFIPCI 0001:d7:16.7 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
EFIPCI 0001:d7:17.0 (8086:2018 class 088000) has no driver
EFIPCI 0001:d7:17.1 (8086:2088 class 110100) has no driver
EFIPCI 0001:d7:17.3 cannot read PCI configuration: No such device (http://ipxe.org/2c044087)
ok

iPXE 1.20.1+ (geecb7) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

iPXE> ifstat                             
net0: a4:bf:01:60:f8:d2 using x722-sfp on 0000:1a:00.0 (closed)
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
net1: a4:bf:01:60:f8:d3 using x722-sfp on 0000:1a:00.1 (closed)
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
iPXE> 
```

mcb30, 2020-10-30:
> ```
> EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
> EFIPCI 0000:1a:00.0 9f000000+200000 is not within root bridge address space
> EFIPCI 0000:1a:00.0 could not probe driver "x722-sfp": No such device (http://ipxe.org/2ccb4087)
> ```

Thank you for the debug output.  It looks as though the UEFI BIOS is failing to provide the expected ACPI resource descriptors.  This should now be worked around in commit 9b25f6e5c (with credit to you) - please reopen if this doesn't fix the problem.  Thanks!

closed 2020-10-30 by mcb30

toreanderson, 2020-10-31:
I can confirm that 9b25f6e fixes the problem, the network interfaces on my Intel S2600BPS motherboard are detected and use the `x722-sfp` driver:

```
(…)
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 9f000000+200000 is not within root bridge address space
EFIPCI 0000:1a:00.0 a1008000+1000 is not within root bridge address space
EFIPCI 0000:1a:00.0 a1009000+1000 is not within root bridge address space
EFIPCI 0000:1a:00.0 using driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 9e000000+200000 is not within root bridge address space
EFIPCI 0000:1a:00.1 a1000000+1000 is not within root bridge address space
EFIPCI 0000:1a:00.1 a1001000+1000 is not within root bridge address space
EFIPCI 0000:1a:00.1 using driver "x722-sfp"
(…)
```

I further confirmed that the preceding commmit (16873703) was still broken in the sense that it detects the NICs but use the NII driver:

```
(…)
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.0 9f000000+200000 is not within root bridge address space
EFIPCI 0000:1a:00.0 could not probe driver "x722-sfp": No such device (http://ipxe.org/2ccb4087)
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 (8086:37d0 class 020000) has driver "x722-sfp"
EFIPCI 0000:1a:00.1 9e000000+200000 is not within root bridge address space
EFIPCI 0000:1a:00.1 could not probe driver "x722-sfp": No such device (http://ipxe.org/2ccb4087)
(…)
```

Thank you for the fix! :+1: