ipxe/ipxe #157: iPXE causes an LACP flood, DoS-ing the control plane of the switch

toreanderson, 2020-10-13:
The source code comment quoted below suggests that iPXE will only respond to an LACP partner's PDUs as they arrive.

https://github.com/ipxe/ipxe/blob/e7f67d5a4c6e9f06aa7a9db1b4245f5e16f00bb2/src/net/eth_slow.c#L39-L46

However, this does not seem to be the case. If the switch sends an LACP PDU to iPXE while its NIC is initialised, the iPXE starts spewing out a never-ending flood of LACP PDUs. This in turn may overload the control  plane on the switch.

I can easily reproduce this using the following lab setup:

* Switch model: Edge-Core AS5812-54X running Cumulus Linux v4.2.1 NOS
* [LACP bypass](https://docs.cumulusnetworks.com/cumulus-linux/Layer-2/LACP-Bypass/) enabled on the switch ports, which allows switch ports to pass traffic even in the absence of incoming LACP PDUs
* Dnsmasq running directly on the switch to provide PXE boot service: `dnsmasq -d -i vlan1409 --enable-tftp --tftp-root /srv/tftp --dhcp-range=10.21.9.10,10.21.9.20 --dhcp-boot=ipxe.efi`
* iPXE 1.20.1+ (gc70b) binary downloaded today from http://boot.ipxe.org/ipxe.efi (md5sum: `4f0268b10c1e54e7d2b7b74f02ceb35b`)
* Server model: HP ProLiant DL380 Gen9, BIOS Version: P89 v2.30 (09/13/2016), UEFI boot mode
* NIC model: HPE Ethernet 10G 2-port 546FLR-SFP+, which appears to be a rebranded Mellanox Technologies MT27520 Family [ConnectX-3 Pro]

When turning on the server, it starts EFI PXE boot, which works fine, chainloading into iPXE. I then use `CTRL+B` to enter the iPXE shell, and issue `dhcp net4` to configure the NIC (`net4` is the same NIC/port the EFI PXE boot uses). This goes fine as well, and the serial console output is shown below:

```
>> Booting Embedded FlexibleLOM 1 Port 1 : HPE Ethernet 10G 2-port 546FLR-SFP+ Adapter - NIC (PXE IPv4) 

>> Booting PXE over IPv4.
  Station IP address is 10.21.9.20

  Server IP address is 10.21.9.3
  NBP filename is ipxe.efi
  NBP filesize is 970752 Bytes
 Downloading NBP file...

  NBP file downloaded successfully.
iPXE initialising devices...ok

iPXE 1.20.1+ (gc70b) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

iPXE> dhcp net4                          
Configuring (net4 e0:07:1b:7f:52:10)...... ok
iPXE>
```

At this point I can leave it like this and observe the network traffic using `tcpdump` on the switch port. It is then clear that the moment the switch sends a single LACP PDU to the server, iPXE goes crazy and starts spamming the switch with an endless flood of LACP PDUs in return. There is no indication that anything is amiss on the console, it just sits there waiting for further commands (which it will happily accept, e.g., issuing `ifclose net4` will instantly stop the flood).

[ipxe-lacp-flood.pcap.gz](https://github.com/ipxe/ipxe/files/5370473/ipxe-lacp-flood.pcap.gz) is a PCAP file which contains an example of such a flood. It was captured with `tcpdump -vni swp24 -s0 -c 1000 -w ipxe-lacp-flood.pcap`, started seconds before issuing the `dhcp net4` command in the iPXE shell. Frame 16 is the initial LACP PDU sent by the switch; iPXE's flood starts at frame 17.

One thing worth noting here is that each successive LACP frame in the flood appears to increase in size by 14 bytes for some weird reason (until the link MTU is reached), cf. column 7 below:

```console
[:~] $ tshark -Y 'slow.subtype == 0x01' -r ipxe-lacp-flood.pcap.gz | head
   16  10.680019 Edgecore_6a:3b:94 → Slow-Protocols LACP 124 v1 ACTOR 44:38:39:ff:30:bf P: 1 K: 15 *F**SGSA PARTNER 00:00:00:00:00:00 P: 1 K: 1 *******A
   17  10.680682 HewlettP_7f:52:10 → Slow-Protocols LACP 138 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER 44:38:39:ff:30:bf P: 1 K: 15 *F**SGSA
   18  10.683697 HewlettP_7f:52:10 → Slow-Protocols LACP 152 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS*
   19  10.686696 HewlettP_7f:52:10 → Slow-Protocols LACP 166 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS*
   20  10.689726 HewlettP_7f:52:10 → Slow-Protocols LACP 180 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS*
   21  10.692671 HewlettP_7f:52:10 → Slow-Protocols LACP 194 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS*
   22  10.695682 HewlettP_7f:52:10 → Slow-Protocols LACP 208 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS*
   23  10.698753 HewlettP_7f:52:10 → Slow-Protocols LACP 222 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS*
   24  10.701667 HewlettP_7f:52:10 → Slow-Protocols LACP 236 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS*
   25  10.704667 HewlettP_7f:52:10 → Slow-Protocols LACP 250 v1 ACTOR e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS* PARTNER e0:07:1b:7f:52:10 P: 1 K: 1 **DCSGS*
```

I can also reproduce this problem with much older iPXE versions, e.g. `1.0.0+ (e5b4)`.

toreanderson, 2020-10-13:
I am attaching the console output when reproducing with an iPXE binary built from current Git master with LACP debugging (command line: `make bin-x86_64-efi/ipxe.efi DEBUG=eth_slow`). The output contains terminal colour codes, so it's best viewed with `less -R` or `bat`). With debugging output enabled, the packet flood is reduced to a trickle; presumably the 115200-baud serial console is slowing it down.

[ipxe-lacp-debug.txt](https://github.com/ipxe/ipxe/files/5370986/ipxe-lacp-debug.txt)

toreanderson, 2020-10-13:
As one would expect, disabling LACP support in iPXE (`sed -i /NET_PROTO_LACP/s/define/undef/ config/general.h`) prevents the bug from being triggered.

Disabling LACP might be a sensible default for build targets meant for chainloading, by the way. The initial UEFI/PCBIOS PXE firmware does generally not support LACP anyway, so in order for it to work the switch must support LACP bypass or equivalent, which in turn means that iPXE does not really need LACP support either.

mcb30, 2020-10-13:
@toreanderson It looks as though your switch (or some other component below iPXE's network stack) is echoing the LACP packets back to iPXE: you can see from the trace lines such as
```
SLOW net4 TX LACP actor (ffff,e0:07:1b:7f:52:10,0001,ff,0001) [aFGSCDlx]
SLOW net4 TX LACP partner (ffff,44:38:39:ff:30:bf,000f,ff,0001) [AFGScdLx]
SLOW net4 TX LACP collector 0000 (0 us)
```
followed immediately by
```
SLOW net4 RX LACP actor (ffff,e0:07:1b:7f:52:10,0001,ff,0001) [aFGSCDlx]
SLOW net4 RX LACP partner (ffff,44:38:39:ff:30:bf,000f,ff,0001) [AFGScdLx]
SLOW net4 RX LACP collector 0000 (0 us)
```
i.e. the packet transmitted by iPXE (with `TX LACP actor (ffff,e0:07:1b:7f:52:10,0001,ff,0001)` containing iPXE's MAC address) is then immediately received by iPXE.

We should be able to work around this loopback issue by detecting and ignoring looped-back packets.  Could you please try the commit in #158?

If it works, then please let me know if I have your name and e-mail address correct in the commit message.  Thanks!

mcb30, 2020-10-14:
> I can confirm that the commit in #158 does resolve the flooding problem.

Thank you!

> I did notice, however, that the curious behaviour of iPXE's LACP responses being consistently 14 bytes larger then those originating from the switch remain as before. I am attaching a packet capture ([ipxe-fixed.pcap.gz](https://github.com/ipxe/ipxe/files/5371587/ipxe-fixed.pcap.gz)) of the behaviour with #158 applied so you can see for yourself.

Will take a look.

> I do not believe it is the switch that is responsible for the looped LACP PDUs, as we make extensive use of LACP between servers and these switches have not seen this problem anywhere else outside of iPXE. I was also able to reproduce it in the lab while using different switch hardware than when we saw it happen in production. While reproducing I paid close attention to the hardware counters of the switch port; the Tx counters were increasing wildly, but the Rx counters were ticking along at a few pps (consistent with normal LACP and STP background chatter).
> 
> I suspect, therefore, that the looping occurs in the NIC's firmware or in its iPXE driver.

Thanks.  I'll update the commit message to note this.

> I just realised that the 24 errors on the fixed version were _RX_ errors, not TX errors, and that they had a different error code than the one seen during the flooding. That means my speculation doesn't necessarily make any sense, so feel free to ignore that.

The 0x202a608a error is the ELOOP reported by the code in #158 that detects and drops the looped-back LACP packets, so it is expected in the circumstances.

mcb30, 2020-10-14:
> Will take a look.

I don't see this same erroneous behaviour in my test setup (qemu VM running iPXE, using a host `tap0` interface made a member of a `bond0` port configured using `mode=4 miimon=100 lacp_rate=1`).  The transmitted packets are the correct length.

iPXE reuses the received I/O buffer to construct the response, without adjusting the length.  Given the odd and apparently LACP-specific behaviour arising from what seems to be the HPE/Mellanox UEFI NII driver, my best guess is that the UEFI NII driver is adding 14 bytes (suspiciously equal in length to an Ethernet header) on either the transmit or receive datapath.

I've added a patch to #158 that will reset the length of the I/O buffer before reusing it.  If the UEFI NII driver is adding the padding on the receive datapath then this should cure the problem.  If the UEFI NII driver is adding the padding on the transmit datapath then there's nothing iPXE can do about it.

closed 2020-10-14 by mcb30

mcb30, 2020-10-14:
> By the way, now that I have your attention, I was wondering if you would be willing to have a look at #84? Would be nice to have it merged before its two-year anniversary.

On it!

NiKiZe, 2020-10-14:
To try and avoid any possible missunderstanding
The card provides an EFI Driver for the NIC that is exposed as a NII interface
iPXE uses that NII interface. `net4: e0:07:1b:7f:52:10 using NII on NII-0000:04:00.0 (open)`

When running PCBIOS mode the undi driver probably comes from the same ROM on the NIC but is a completely different driver.
So with the information so far most likely this is a bug with the EFI driver on NIC, rather than with iPXE NII implementation.

You might want to tell HP/Mellanox about this.

mcb30, 2020-10-14:
> So it seems clear to me that this LACP loopback issue is something the NIC firmware/driver or iPXE (impossible for me to tell which) conjures up all on its own, it is _not_ caused by an improperly configured network topology.

Agreed; that's why I changed the comments and commit log wording to indicate that the faulty component is the UEFI NII driver.

PR #158 (2020-10-14): [lacp] Detect and ignore erroneously looped back LACP packets
Some network topologies seem to cause LACP packets transmitted by iPXE
to be looped back as received packets.  Since iPXE's trivial LACP
responder will send one response per received packet, this results in
an immediate LACP packet storm.

Detect looped back LACP packets (based on the received LACP actor MAC
address), and refuse to respond to such packets.

Fixes: #157 

Reported-by: Tore Anderson <tore@fud.no>