ipxe/ipxe #159: [EFI] Assertion failures when kernel is about to be booted

toreanderson, 2020-10-15:
Since commit c70b3e04e86cefca335e36f883829d89583a6921 iPXE is no longer able to boot my HP ProLiant DL380 Gen9 server with HPE Ethernet 10G 2-port 546FLR-SFP+ (rebranded Mellanox Technologies MT27520 Family [ConnectX-3 Pro]).

It fails like this, where it hangs until manually rebooted:

```
ASSERT u:\MdeModulePkg\Core\Dxe\Hand\Locate.c(492): Size >= 0
ASSERT u:\MdePkg\Library\BaseMemoryLibRepStr\CompareMemWrapper.c(62): (Length - 1) <= (0xFFFFFFFFFFFFFFFFULL - (UINTN)DestinationBuffer)
ASSERT u:\MdePkg\Library\BaseMemoryLibRepStr\CompareMemWrapper.c(63): (Length - 1) <= (0xFFFFFFFFFFFFFFFFULL - (UINTN)SourceBuffer)
ASSERT u:\MdeModulePkg\Core\Dxe\Hand\Locate.c(492): Size >= 0
ASSERT u:\MdePkg\Library\BaseMemoryLibRepStr\CompareMemWrapper.c(62): (Length - 1) <= (0xFFFFFFFFFFFFFFFFULL - (UINTN)DestinationBuffer)
ASSERT u:\MdePkg\Library\BaseMemoryLibRepStr\CompareMemWrapper.c(63): (Length - 1) <= (0xFFFFFFFFFFFFFFFFULL - (UINTN)SourceBuffer)
```

These messages appear after iPXE has completed the download the kernel and the initramfs image, and is about to hand over control to the Linux kernel.

Today's Git master with c70b3e04e86cefca335e36f883829d89583a6921 reverted works fine. When this build boots successfully, normal output from the Linux kernel appears instead of the ASSERT lines quoted above.

pmounsey, 2020-10-18:
Same Gen10, HP FlexFabric 10Gb 2port 534FLR-SFP+ Adapter, after a few moments, RSOD GeneralProtectionException.    Also reverted to an older binary resolved issue.

[image]

mcb30, 2020-10-26:
Commit https://github.com/ipxe/ipxe/commit/5b41b9a80 may fix the underlying issue.  @toreanderson: could you possibly retest with current master?  Thanks.

toreanderson, 2020-10-29:
I will, but it might take a few more days before I get the opportunity, as the server is currently back in production and cannot be easily rebooted.

NiKiZe, 2020-10-30:
Possible related to #164 (?)

Cellebyte, 2020-11-06:
@toreanderson I tested it today with the current master and still got an issue.
I will append all stacktraces I got.
[image]
[image]
[image]

Servers:
HPE ProLiant DL20 Gen10
HPE ProLiant DL325 Gen10 Plus

toreanderson, 2020-11-12:
@mcb30 I have finally been able to re-test, and I can confirm that 5b41b9a80ffb365376d8d522675a8d248a8717ab (and master as of 0e26220902592fb3066d2043dec212d957215b79) resolves the assertion failures I saw previously, and I can boot my server without any problems.

I am however not closing this issue in case you would like to track @Cellebyte's boot failure here as well.

mcb30, 2020-11-12:
> @mcb30 I have finally been able to re-test, and I can confirm that [5b41b9a](https://github.com/ipxe/ipxe/commit/5b41b9a80ffb365376d8d522675a8d248a8717ab) (and master as of [0e26220](https://github.com/ipxe/ipxe/commit/0e26220902592fb3066d2043dec212d957215b79)) resolves the assertion failures I saw previously, and I can boot my server without any problems.
> 
> I am however not closing this issue in case you would like to track @Cellebyte's boot failure here as well.

Fantastic; thank you for letting me know!

I will mark this issue as closed.  @Cellebyte please open a new issue if needed; thank you.

closed 2020-11-12 by mcb30

Cellebyte, 2020-11-12:
@mcb30 going back to v1.20.x from January solved the issue for me.

mcb30, 2020-11-23:
@Cellebyte The underlying issue should be fixed as of commit 1295b4acf - please try retesting with the current iPXE master.  Thank you!

Cellebyte, 2020-11-25:
@mcb30 thanks I will test it when I'am back from vacation ^^

kjetilho, 2020-12-02:
I had the same problem as @Cellebyte on a HPE DL380 gen10, upgrading to iPXE 1.20.1+ (gb6e2e) fixed the problem.

PR #170 (2020-11-23): [efi] Allow initialisation via SNP interface even while claimed
Fixes: #164 
Fixes: #155 
Fixes: #159