ipxe/ipxe #160: ARP request is sent unexpectedly during and iPXE boot phase.

kevintranbrcm, 2020-10-19:
Some of our customers reported an intermittent iPXE problem where iPXE fails to download a large image from the iPXE server with our 10G adapter using chain load (legacy UNDI driver).  After some analysis, I found that iPXE stack sent out an ARP request during TCP data transfer.  it then sent out TCP ACK packet after it received an ARP responses.  During this period, iPXE stack was dropping the incoming TCP packets.  iPXE stack shouldn't send out any ARP requests for the active connection.  ARP requests should be sent out after a long time of inactivities of a particular node where ARP entry is removed from the ARP table cache.  Based on the network trace,  the last good packet that iPXE stack received from the iPXE server is just 200uS earlier. 

They are using iPXE binary that was built in 2016.  Is this a known problem which has been fixed?

Thanks!
Kevin

mcb30, 2020-10-30:
Neighbour cache (ARP or NDP) discarding in iPXE is driven by memory pressure rather than by age, since iPXE is expecting to remain running for only a few seconds in total and so does not implement any of the usual long-running aging logic (e.g. renewing DHCP leases, or discarding old ARP entries).  The least recently used ARP cache entry will be discarded under extreme low memory conditions, and only after discarding less expensive cache entries (such as the TCP out-of-order receive queue).

It's not a known issue that immediately springs to mind, but there have been so many changes since 2016 that I would insist on attempting reproduction with a more recent build before debugging further.

NiKiZe, 2021-01-17:
Did you test a newer version and reproduce the issue? Or did it work as expected and we can close the issue?

NiKiZe, 2021-07-09:
@kevintranbrcm another ping, did upgrade solve the issue, or is it an issue with current master as well?

NiKiZe, 2026-01-16:
Closing as stale.

closed 2026-01-16 by NiKiZe