ipxe/ipxe #163: Cannot compile iPXE rom for Intel I219-V ethernet adapter

BrentFathom5, 2020-10-29:
When I try to compile an iPXE rom to burn onto my i219-V ethernet adapter, I run the following:

```bash
cd src
make bin/80860d4f
```
and I get the following error:

```bash
[VERSION] bin/version.80860d4f.rom.o
  [LD] bin/80860d4f.rom.tmp
--defsym:2: undefined symbol `obj_80860d4f' referenced in expression
Makefile.housekeeping:1188: recipe for target 'bin/80860d4f.rom.tmp' failed
make: *** [bin/80860d4f.rom.tmp] Error 1
rm bin/version.80860d4f.rom.o
```

is there something different I should be doing when compiling this? Thanks!

NiKiZe, 2020-10-29:
Is that your exact command? I would suggest that you check which targets are valid, see the app notes/buildtargets page

BrentFathom5, 2020-10-30:
Yes, this is my exact command. I already have iPXE on my NIC, but I am trying to upgrade to the latest build. Is it the case that I can build for a different target, but have the ROM still be compatible with my NIC? Sorry for a potentially noobie question, I'm fairly new to using iPXE.

mcb30, 2020-10-30:
Two things:

1. Your build command must I think be `make bin/808604df.rom` rather than just `make bin/808604df`.  I'm assuming this is just a typo.

2. The PCI ID 8086:04df is not defined within iPXE.  You may need to add the relevant `PCI_ROM()` line in `drivers/net/intel.c`.  If this works, please do submit a pull request to add the ID.

Thanks.

BrentFathom5, 2020-10-30:
@mcb30 Ah, yes that was a typo...I was putting **.rom** at the end. Thanks for the direction! I'll take a look at that intel file and see what progress I can make.

NiKiZe, 2020-11-09:
Where you able to solve this? as in update intel.c by adding the correct `PCI_ROM` line? and it work correctly for you nic?
(I would suggest testing a 80860d4f.pxe or .usb build first to know it works before trying to get it in to a rom.

reference to the pciid: https://pci-ids.ucw.cz/read/PC/8086/0d4f

mcb30, 2020-11-23:
Assuming closed; please reopen if needed.

closed 2020-11-23 by mcb30