ipxe/ipxe #164: exit_boot() failed! & efi_main() failed! with OptiPlex 9020

PicoMitchell, 2020-10-29:
This issue occurred with an iPXE build using commit 2091288eaa5b3b6144afba193f44cce985705e79 because I thought it was the newest version that was working properly for me (refer to issue https://github.com/ipxe/ipxe/issues/162).

But, we've just run into a new issues using that build when loading Linux Mint 20. The iPXE menu loads properly and everything, this failure is after the `boot` command is run.

Roughly, `initrd` is called first with the proper initrd.lz file and then `boot` is called with the proper vmlinuz file and kernel arguments which are all known working. Both command seem to run properly, but then we hit a black screen with the just the following messages:
```
exit_boot() failed!
efi_main() failed!
```

After running into this issue, I just rolled all the way back to commit fe69934191ca46c4948a71f416c21dcc5a29e63a from 2020-09-22 (which was just the last known good version that we used in production) and then booting into Linux Mint 20 from this exact OptiPlex 9020 works fine again.

PicoMitchell, 2020-10-29:
Also, this exact same error was hit on 3 identical OptiPlex 9020s.

mcb30, 2020-10-30:
As with #162 could I please ask you to retest with current master?  Thanks.

PicoMitchell, 2020-10-30:
I grabbed another OptiPlex 9020 to hold onto for testing for the time being.

With the latest build, I am still getting this same issue.

Is there anything I can do on my end to get more information about whats going on here? Such as build with extra debug logging or anything?

PicoMitchell, 2020-10-30:
Just to be clear, whatever is going on here does not affect all computers.

My ThinkCentre M700 Tiny test computer continues to boot into Linux Mint 20 via iPXE with the latest build.

One thing that I noticed, is that the font in iPXE looks slightly different between these two computers.

On the OptiPlex 9020, the font is thinner (not bold looking) while the font looks bold on the ThinkCentre M700. I know that's a cosmetic detail, but I thought it might give some clue into what different inner workings of iPXE may be at play here.

Also, this issue affects both `snponly.efi` & regular `ipxe.efi` builds. BIOS versions seem to be unaffected.

NiKiZe, 2020-10-30:
Your issue is most likely due to something specific in Dells implementation of EFI.

In regards to font's again that is due to graphic card implementations or EFI.
This also goes for your error messages, which comes from EFI firmare, not iPXE code.

You might want to open an issue with Dell here, they are actually quite famous (or rather infamous) for having bugs in their firmware.

PicoMitchell, 2020-10-30:
> Your issue is most likely due to something specific in Dells implementation of EFI.

While I don't doubt there is a chance that this may be some issue with Dell EFI, something changes in iPXE recently which contributes to this issue since commit fe69934191ca46c4948a71f416c21dcc5a29e63a works properly with booting into Linux Mint 20 from iPXE.

PicoMitchell, 2020-10-30:
And just to add more clarity, this issue is not specific to Linux Mint 20, that was just the first OS we ran into this issue on. I've just tried booting Debian 9.12 and Clonezilla 2.6.7-28 and got the same issue.

PicoMitchell, 2020-10-30:
Confirmed that commit fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb boots OSes properly AND that c70b3e04e86cefca335e36f883829d89583a6921 fails to boot with the same error described above.

I believe this likely confirms @NiKiZe's suspicion that this is related to issue https://github.com/ipxe/ipxe/issues/159.

mcb30, 2020-10-30:
Best guess: there is some driver in the Dell UEFI firmware that binds to one of the handles produced indirectly by iPXE, and this driver is subsequently incapable of being disconnected when it comes time to call `ExitBootServices()`.

There is precedent for this: commit 64b4452bc introduces the concept of a UEFI driver blacklist specifically to work around problems caused by a faulty Dell UEFI driver.  It is very plausible that we need to add some other Dell driver to this list.

Could you please try using iPXE to boot into the UEFI shell?  You can then use a combination of the `devtree`, `devices`, `drivers`, `dh`, and `openinfo` UEFI shell commands to identify which drivers have created and/or bound to the iPXE devices or child devices thereof.  This list of devices should differ between commit fbb5989fd (which precedes the change to use recursive connection) and current master.

Given this information, you could then try experimenting with unloading sets of drivers (e.g. by adding them in `interface/efi/efi_blacklist.c`) to see if you can identify a culprit to be permanently blacklisted.

PicoMitchell, 2020-10-30:
Thanks for this info. I'm not sure that I will have time to do this investigation today, but I can make time to do it next week.

Naive question though, in commit c70b3e04e86cefca335e36f883829d89583a6921 you said:
> There appears to be no reason for avoiding recursion when calling ConnectController()

But now it seems that there are least 3 distinct bugs that I've noticed that have been traced back to this commit (this one, https://github.com/ipxe/ipxe/issues/159, and https://github.com/ipxe/ipxe/issues/155).

Is this not a good reason to avoid recursion when calling `ConnectController()`. But of course I am unaware of what benefits using recursion there offers.

PicoMitchell, 2020-10-30:
For some context, I work in refurbishment and maintain our netboot setup which uses iPXE.

We take donations of computers and refurbish them so we see a huge variety of desktops and laptops of all different manufacturers and ages.

Thinking about the prospect of halting production on certain models when a technician runs into this problem and then needing to track down which driver is the cause to be added into iPXE and having iPXE updated to accommodate does not sound like a great prospect.

I'm not against putting in the leg work. Ideally this would be down to just a few common drivers that can be blacklisted, but maybe its tons of separate driver that will need to be tracked down an added.

I guess I'd just feel better about that leg work if I understand the purpose a bit better. Such as, does using recursion in `ConnectController()` avoid other issues that others have run into?

Or maybe a simple solution could be to add a build configuration option so that folks can choose wether or not they want to enable this recursion in `ConnectController()` if it's helpful for certain cases and harmful for others.

mcb30, 2020-10-30:
The least surprising effect is that connection is recursive: to do otherwise is to leave the system in a state where a future completely unrelated event (e.g. loading an unrelated UEFI driver that can't bind to any devices present in the system and so should be a no-op) may suddenly trigger existing drivers to bind to existing handles.  It is undesirable to leave the system in this kind of hair-trigger situation.

The change to use recursive connection (commit c70b3e04e) is exposing preexisting interoperability problems that were previously masked by sheer luck.  For example: the bug around not handling `UninstallMultipleProtocolInterfaces()` failures (fixed in commit 5b41b9a80) has always been present but had not been noticed until recursive connection made it more likely to happen.  I'd rather not revert to relying upon luck to hide problems.

If it turns out that the interoperability problems are more widespread than expected, then we can revisit the question in future.

PicoMitchell, 2020-10-30:
Thanks for this context. That makes good sense. I can see that it would be nice to nail down and blacklist these driver with issues instead of hoping to avoid them by chance.

As a related aside, can you give any guidance on how to boot into the UEFI Shell from iPXE? Is it possible to do that if there is not hard drive or OS on the computer?

mcb30, 2020-11-02:
> As a related aside, can you give any guidance on how to boot into the UEFI Shell from iPXE? Is it possible to do that if there is not hard drive or OS on the computer?

Yes, you can use the normal iPXE [`chain`](https://ipxe.org/cmd/chain) command to load any UEFI executable, including the UEFI shell.  For example:
```
#!ipxe
chain http://boot.ipxe.org/Shell.efi
```

PicoMitchell, 2020-11-02:
Thanks so much, I will add this to our administration tools in our iPXE menu for easy access.

I'm curious, what is the source for this Shell.efi?

From Googling, I found https://github.com/tianocore/edk2/tree/edk2-stable201903/ShellBinPkg/UefiShell/X64 but yours seems to be newer/larger and the latest master of this project does not have a built Shell.efi included. Is yours a newer build from this same project, or is it totally different than what I've found here?

Also, I won't be back in the office until Thurs to have access to the OptiPlex 9020 again to do this research.

PicoMitchell, 2020-11-05:
Finally back in the office to mess with this OptiPlex 9020 again and it looks like @aheusser did most of the heavy lifting already!

Following their lead, I looked in the **#D** column of the `drivers` command in `Shell.efi` and the _only_ difference listed was the same one that fixed their issue (#155) when blacklisted: "MNP Network Service Driver"

And, surprisingly (or maybe not so surprisingly) blacklisting that exact same driver fixed this issue as well!

That would be awesome if this one driver is all that is needed to be blacklisted to fix problems across a wide variety of computer models.

PicoMitchell, 2020-11-05:
It's nothing you couldn't do in a few moments time, but here are my modifications to `efi_blacklist.c` for reference:

```
/**
 * Blacklist MnpDxe driver
 *
 * @v binding		Driver binding protocol
 * @v loaded		Loaded image protocol
 * @v wtf		Component name protocol, if present
 * @ret blacklisted	Driver is the blacklisted driver
 */
static int
efi_blacklist_mnp ( EFI_DRIVER_BINDING_PROTOCOL *binding __unused,
			       EFI_LOADED_IMAGE_PROTOCOL *loaded __unused,
			       EFI_COMPONENT_NAME_PROTOCOL *wtf ) {
	static const CHAR16 mnp[] = L"MNP Network Service Driver";
	CHAR16 *name;

	/* Check driver name */
	if ( ! wtf )
		return 0;
	if ( wtf->GetDriverName ( wtf, "eng", &name ) != 0 )
		return 0;
	if ( memcmp ( name, mnp, sizeof ( mnp ) ) != 0 )
		return 0;

	return 1;
}
```

I did not notice (or did not know how to find) any Manufacturer associated with "MNP Network Service Driver" so I removed the manufacturer checking for this driver.

And of course:
```
/** Blacklisted drivers */
static struct efi_blacklist efi_blacklists[] = {
	{
		.name = "MnpDxe",
		.blacklist = efi_blacklist_mnp,
	},
	{
		.name = "Dell Ip4Config",
		.blacklist = efi_blacklist_dell_ip4config,
	},
};
```

PS. I'd like to suggest replacing the term _blacklist_ to something like _blocklist_ (for example) to move away from use of these kinds of racially charged terms.

aheusser, 2020-11-05:
@PicoMitchell FYI: my experiment with blocking `MnpDxe `was similar to yours.  I also didn't know what manufacturer to ascribe to the problematic driver, so I blocked it universally as well.

I was interested to observe that blocking "MNP Network Service Driver", also seemed to cause "IP4 CONFIG Network Service Driver" and "IP6 Network Service Driver" to **also** not have "#D" entries either ... similar to what I observed in the last working commit (fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb).

So my blocking of the MnpDxe driver appeared to cause **all three drivers** to not be present (which then ultimately results in the same `drivers` list as I saw in the last **working** commit (fbb5989fd9a5ddbbde6e273c7bb3ba75ec9f1ebb).

PicoMitchell, 2020-11-05:
@aheusser That is quite interesting. Maybe @mcb30 will have more insight about what exactly is going on here and also know more about the inner workings of this MnpDxe driver.

It's also interesting that you had "IP4 CONFIG Network Service Driver" present since that driver was the one that caused the addition of the blacklist in the first place. But, it was only blacklisted for Dells. For your issue, it sounds like you're running iPXE in a VM. Is the underlying hardware a Dell? Wondering if some of these drivers listed in a VM could be inherited from the physical hardware, but then the actual manufacturer wouldn't be known to iPXE since it would only get the VM manufacturer. Just thinking/wondering out loud here, but that driver being listed for you seems possibly worth investigating.

aheusser, 2020-11-05:
@PicoMitchell I am not sure what the **underlying** hardware is in my case. It is an ESXi server, that is all I know at the moment.

mcb30, 2020-11-06:
> Thanks so much, I will add this to our administration tools in our iPXE menu for easy access.
> 
> I'm curious, what is the source for this Shell.efi?
 
@PicoMitchell It was built from https://github.com/tianocore/edk2/ using `build -a X64 -p ShellPkg/ShellPkg.dsc`

mcb30, 2020-11-06:
> Following their lead, I looked in the **#D** column of the `drivers` command in `Shell.efi` and the _only_ difference listed was the same one that fixed their issue (#155) when blacklisted: "MNP [...]

As noted in https://github.com/ipxe/ipxe/issues/155#issuecomment-723300494 we unfortunately can't globally blacklist MnpDxe since it is required for some realistic boot scenarios.

However, the fact that you've found a common fix suggests that there may be a common root cause.

At this point, I think it would be most interesting for me to have a reproduction of the problem in a VM, so that I can debug it directly.  @aheusser Are you able to simplify your test setup to the point that I could reproduce it locally?

mcb30, 2020-11-06:
> PS. I'd like to suggest replacing the term _blacklist_ to something like _blocklist_ (for example) to move away from use of these kinds of racially charged terms.

The term "blocklist" (or "denylist") doesn't really cover the intended meaning here: unlike in an ACL use case, there is no action of "blocking" or "denying" taking place.  I'm very happy to take alternative suggestions, subject to the usual restrictions that names must be short, unambiguous, and obvious on first reading.  The word "veto" seems close to the concept in use, though "vetolist" is not acceptable since the word "vetolist" does not obviously decompose into "veto" and "list" on first reading.  Better ideas welcome!

PicoMitchell, 2020-11-06:
Thank you for all of your replies, info, and consideration.

We're currently using a build of iPXE with MnpDxe globally blacklisted with no issues so far, but I await your better solution. Hoping you'll be able to set up some local testing for this issue.

In regard to a better replacement term for blacklist, I also thought blocklist may not be great for iPXE since "block" has other very different meaning.

I'll continue trying to think of better terms, but on first though these come to mind:
avoidlist
bypasslist
rejectlist
ignorelist

I think of these option (if I am understanding the actual intended meaning correctly) votes would go to ignorelist.

PicoMitchell, 2020-11-06:
@mcb30, because of your question to @aheusser (https://github.com/ipxe/ipxe/issues/155#issuecomment-723300494), I was curious if I could wimboot on this OptiPlex 9020.

After rolling back to iPXE from 2020-10-30 with this issue present (which does not have MnpDxe blacklisted), I re-confirmed that I COULD NOT boot into Linux, but then confirmed that I COULD boot into WinPE via wimboot with this build of iPXE.

So, clearly something different is going on for me and @aheusser, but interesting that the same "solution" solves both of our problems.

mcb30, 2020-11-07:
> Another thought is that this name may not actually need a -list suffix. Could be called "skipdrivers" (or "skipdevices" if that's more accurate?) or any of the other prefix words proposed.

With a commit date manipulated to honour the day's events: https://github.com/ipxe/ipxe/commit/0e26220

ngrsdhbw, 2020-11-09:
I have a Dell T20 with Bios A20 in a remote location that I am accessing using Intel AMT.
When trying to reinstall Debian using the netboot.xyz-efi.iso mounted in the virtual cd drive I ran into exactly the same issue:
[image]
As requested I also took a look at the listed drivers
[image]
[image]
Since there is apparently some problems with the driver I disabled the UEFI Network stack in the bios:
[image]
After doing this the ArpDxe, SnpDxe and MnpDxe drivers are gone and **booting works**
[image]
[image]

aheusser, 2020-11-09:
@mcb30  @PicoMitchell I observe, what perhaps may be, the same problem when performing a generic Linux kernel+initrd boot (e.g. chain http://boot.ipxe.org/demo/boot.php).  The image downloads but the machine resets itself before boot completes.  So it appears to not be constrained to WinPE for me either.

Again, PXE booting to either environment **works** for me, using the same environment and VM if I use the pre-recursion commit ... or if I use latest and change the code to veto the MnpPxe.

I have been able to get my repro setup simplified down to the following:
1) install and configure "Tiny PXE" (http://mistyrebootfiles.altervista.org/documents/TinyPXEServer/files/pxesrv_about.htm)
2) copy files into Tiny PXE (PXE script, wimboot, iPXE firmware, WinPE image files)
3) create "ESXi 6.0 and later (VM version 11)" VM
 - Guest OS (Microsoft Windows 10 64-bit )
 - UEFI firmware; secure boot disabled
 - E1000E network adaptor
4) attach the VM to the same network as the Tiny PXE server and PXE boot the VM

I can share my pxe script and, if necessary, a zip of all my other files too (WinPE WIM image files) and detailed setup instructions.  I only need to know: a) if it is actually desired for me to take the time to pull all that information together, and b) what would be the recommended way for me to share the content with you. I guess, barring some other recommendation, I could put them up on a Google drive for you to pull down.

mcb30, 2020-11-09:
@aheusser Let me try the generic kernel+initrd boot in a VMware setup first.  I know it works without problems in QEMU (with OVMF firmware built from EDK2).  If I **can't** reproduce the problem in VMware then it would be great to have you assemble your reproduction files; thanks.

aheusser, 2020-11-09:
@PicoMitchell FYI: because you asked about this last week, I got the following info about my ESXi host hardware:

[image]

PicoMitchell, 2020-11-10:
@aheusser Cool, I really appreciate all the investigation you've done into this bug. Hoping this all helps @mcb30 find a solution.

This does seem like another sign that this issue may be because of Dell's buggy UEFI firmware as @NiKiZe pointed out early on. But, hoping that @mcb30 can find a fix/workaround within iPXE anyway.

At least for our refurbishment use-case, we need to be able to boot up any donated computer and be able to netboot it without having to first update any firmware (even if Dell were to fix any underlying bugs in the future).

mcb30, 2020-11-10:
> @aheusser Let me try the generic kernel+initrd boot in a VMware setup first. I know it works without problems in QEMU (with OVMF firmware built from EDK2). If I **can't** reproduce the problem in VMware then it would be great to have you assemble your reproduction files; thanks.

Updated to VMware Workstation Player version 16 and tried a boot as follows:

- VM type `windows9srv-64` ("Windows Server 2016")
- iPXE built as `make bin-x86_64-efi/ipxe.efi` from current master (https://github.com/ipxe/ipxe/commit/0e26220902592fb3066d2043dec212d957215b79) with no customisation, placed in TFTP server root
- DHCP server (ISC `dhcpd`) configured to hand out `ipxe.efi` or `http://boot.ipxe.org/demo/boot.php` via:
```
if exists user-class and option user-class = "iPXE" {
    filename "http://boot.ipxe.org/demo/boot.php";
} else {
    filename "ipxe.efi";
}
```

With this configuration:
- VM powers up and shows `Attempting to start up from.... EFI Network`
- VM downloads and runs `ipxe.efi`
- iPXE starts up as normal and shows `net0: 00:50:56:3d:61:32 using 82574l on 0000:03:00.0`
- iPXE downloads and executes http://boot.ipxe.org/boot/demo.php
- After a few seconds, Linux kernel+initrd starts up and reaches interactive shell

i.e. no errors occur for me in this setup.

I've tried to make this case as simple as possible.  @aheusser could you give this a try and see how it behaves in your ESXi setup?

aheusser, 2020-11-10:
Hi @mcb30 I approximated your (successful) setup & test with latest master.  Following are the details of what I did:

Using VMWare Workstation 15.5.1 build-15018445

Created a ubuntu 20.04.1 VM to be my dhcp and pxe server (and internet gateway for the private network):
- Has 2 network interfaces, one for private network (no internet) and one for internet access
- Installed and enabled TFTP
- Installed isc-dhcp-server and configured it as you did (to hand our ipxe.efi or http://boot.ipxe.org/demo/boot.php based on `user-class`
- Configured the machine to route from the private network to the internet
- Built tip of ipxe master and placed bin-x86_64-sb/ipxe.efi in the tftp root

Created a windows 10 x64 VM to be my PXE client:
- Has single network interface (for the private network that the PXE server machine is connected to)
- UEFI; non-secure boot

In this environment, iPXE downloads and executes http://boot.ipxe.org/boot/demo.php. After a few seconds, Linux kernel+initrd starts up and reaches interactive shell (i.e. no errors occur for me in this setup).

So, in this environment (i.e., non-ESXi) **it works fine**.  But when my VM is instead hosted in my ESXi server, it fails in the same way I previously detailed in this ticket.

Not all is the same between this latest (successful) test and the previous (failed) one. In this test, I am using a hand-rolled ubuntu as my dhcp, tftp, and pxe server ... and my PXE client is a VMWare workstation VM. In my previous test, I was using "Tiny PXE Server" as my dhcp proxy, tftp, and pxe server ... and my PXE client was an ESXI VM.

I guess I **could** spend time re-creating my hand-rolled ubuntu VM in ESXI (i.e., taking Tiny PXE out of the mix and reducing the variables), and see if that succeeds or fails. But my expectation is that it will again **fail** ... because Tiny PXE has no apparent problem delivering the firmware in the first place (iPXE loads and runs just fine). But I suppose that is a test I could do (now that I invested the time to figure out how to create the ubuntu dhcp, tftp, and pxe server). I don't expect it to yield an insightful result though.

mcb30, 2020-11-11:
@aheusser Thanks for your testing.

Does ESXi allow you to suspend and export a (diskless, and therefore small) VM that can then be imported into VMware Workstation?  If so, then that could be a fast route to reproduction at my end.

Stephenm64, 2020-11-11:
Started seeing this same issue on a system when attempting to boot Centos 8:
[image]

### System info
- X570 Phantom gaming 4 Motherboard
- Secure boot disabled
- UEFI boot
- CSM Enabled

### PXE info
- iPXE 1.20.1+ (gc70b)  - Boot fails
- iPXE 1.0.0+ (2afd6) - Boot successful

The system can boot into other operating systems fine such as WinPE and a quite old version of CloneZilla but attempts to boot into Centos 8 result in that boot failure instantly. reverting to the older version of iPXE allows for Centos 8 to load correctly.

This issue looks to also have been reported on the forums by another user here: https://forum.ipxe.org/showthread.php?tid=31946

aheusser, 2020-11-11:
@mcb30 I am not an ESXi expert, but the path I found to approximate your ask did not yield the observed failure.

Here is what I tried:
1) shutdown the PXE Client VM in ESXi
2) navigated to the ESXi datastore and copied the files (including the disk) for the VM to my local host
3) opened the VM in VMWare Workstation
4) changed the network adaptor assignment for the VM to my private network
5) start the VM

This test didn't PXE boot at all, because my attempts to change the network adaptor assignment would not seem to "stick". I would change it and restart the VM ... and then observe my assignment change was reverted (back to Bridged).  I tried this multiple times with the same result after starting the VM.

So, I instead added a **new** network adaptor with the correct assignment (to the private network) and removed the pre-existing one and tried the test again. PXE boot succeeds this time ... and it boots all the way into a generic Linux kernel+initrd boot interactive shell.

I don't know if the above workflow is "valid" ... but if it is, it would suggest that the problem is contributed to by running in ESXi (or running on the ESXi host hardware I have).

PicoMitchell, 2020-11-12:
@aheusser You mentioned the issue exists when using a Bridged network adapter in the VM and does not exist when using a different network adapter. I believe the Bridged network adapter means the VM is accessing the hardware network adapter of your Dell PowerEdge R740. This might be a clue that the issue may be specific to the hardware and may not be able to be replicated by @mcb30 in another VM running on different hardware.

aheusser, 2020-11-12:
Agreed @PicoMitchell , that has been one of my strong wonderings/suspicions from the beginning of this issue. But to be clear, in my original ESXi problem repro configuration ... my VM's were **not** using a Bridged adaptor (as such). Well, at least there doesn't appear to be any such configuration setting in the ESXi VM that I can find.  But when I moved the VM over to Workstation, the adaptor came up as "Bridged".

It was @mcb30 's, not my, suggestion that we try to see if the VM itself was somehow faulty and would continue to demonstrate the problem when extracted from ESXi and migrated to a different host.

I am merely trying to respond to @mcb30 's request for follow-up. It was an experiment he requested, and so I tried to execute it to the best of my ability.

If it is true that it is a hardware-specific problem ... then considering that the hardware in question in not some exotic or obtuse piece of kit ... it would suggest there are compatibility limitations with the iPXE codebase as it stands at the moment.  Limitations that, I would expect other users might encounter as well.

mcb30, 2020-11-12:
@PicoMitchell Using a "bridged" network adapter would normally mean that the VM is talking to a purely virtual NIC that is then bridged on the host side to the physical NIC.  It **is** possible to have the VM talking directly to the hardware, but that would require using a passed-through PCIe virtual function rather than the standard "bridged" configuration.

@aheusser Thank you for testing.  I should have been clearer about my intentions, sorry: the aim of a "suspend-export-import-resume" was to allow the UEFI firmware image as present within the ESXi VM to be run in VMware Workstation (and hence run locally by me).

I will see if I can get ESXi running on a test system here and hope that I can reproduce the issue.

aheusser, 2020-11-12:
@mcb30 I know my procedure wasn't identical to the "suspend-export-import-resume" you requested. I am no expert in ESXi, that's for sure ... but my procedure was the only one I could find to export VM to Workstation.

There is a VMWare vCenter Converter Tool ... but it seemed to not want the VM to be in a suspended state.

PicoMitchell, 2020-11-12:
@mcb30 Thanks for the clarification about Bridged network adapters in a VM. This thread has been very informative :-) I see also that I misunderstood exactly what was happening with @aheusser's VM setups as well and when and if a Bridged adapter was being used in which VM setup.

PicoMitchell, 2020-11-12:
An interesting note... We've been using the iPXE build I made with MnpDxe vetoed with no issue for a while until we tried to boot a OptiPlex 7010. Using iPXE with MnpDxe vetoed, the computer just hangs on "initializing devices" forever.

Booting the OptiPlex 7010 using an iPXE build with ConnectController() recursion enabled loads properly, but then hits this same "exit_boot() failed!" and "efi_main() failed!" error.

Rolling back all the way to the build prior to ConnectController() recursion being enabled and nothing vetoed works properly. I guess that's what we'll have to stick with for now.

PicoMitchell, 2020-11-13:
So that I can continue using the latest build of iPXE while avoiding this issue for our use case, I've added this snippet to my iPXE build script after I clone the latest iPXE from GitHub.

I've wrapped the block in a condition so I can easily turn it on or off for debugging and the snippet will exit my build script if something changes in the source which would result in the replacements not happening as expected (only expecting to replace a single instance instance of each match).

This snippet could also be useful for determining if an issue is caused by this ConnectController recursion while easily being able to use the latest build without having to roll back to a month old version of iPXE since lots of other changes have happened since then as well.

```
if true; then
    # DISABLED ConnectController() RECURSION TO TEMPORARILY WORKAROUND ISSUE #164 (https://github.com/ipxe/ipxe/issues/164)
    if [ "$(grep -c 'TRUE ) ) != 0 ) {' interface/efi/efi_driver.c)" == "1" ]; then
        sed -i 's/TRUE ) ) != 0 ) {/FALSE ) ) != 0 ) {/' interface/efi/efi_driver.c
    else
        echo 'DISABLE ConnectController() RECURSION ERROR'
        read line
        exit 1
    fi

    if [ "$(grep -c 'bs->ConnectController ( device, NULL, NULL, TRUE );' interface/efi/efi_driver.c)" == "1" ]; then
        sed -i 's/bs->ConnectController ( device, NULL, NULL, TRUE );/bs->ConnectController ( device, NULL, NULL, FALSE );/' interface/efi/efi_driver.c
    else
        echo 'DISABLE ConnectController() RECURSION ERROR'
        read line
        exit 1
    fi
fi
```

mcb30, 2020-11-15:
Managed to get ESXi 7.0b up and running (via nested virtualisation), and it annoyingly succeeds in the same way as VMware Player (i.e. no errors for an ESXi "Windows 10 x64" VM booting `ipxe.efi` and then http://boot.ipxe.org/boot/demo.php)

mcb30, 2020-11-15:
@aheusser Is there any chance of getting remote access to a known-failing VM within your ESXi setup, since I have not been able to reproduce the failure here?

mcb30, 2020-11-15:
> Managed to get ESXi 7.0b up and running (via nested virtualisation), and it annoyingly succeeds in the same way as VMware Player (i.e. no errors for an ESXi "Windows 10 x64" VM booting `ipxe.efi` and then http://boot.ipxe.org/boot/demo.php)

Also tried using the ESXi VM to boot into the CentOS 8 installer, again with no problems.

mcb30, 2020-11-15:
Also tested in ESXi using `snponly.efi` rather than `ipxe.efi`.  Works with no errors, after fixing an unrelated issue via commit 83b8c0e21

aheusser, 2020-11-17:
@mcb30 Thank you for attempting to recreate it on your side.

TL;DR  I would be happy to do something like screenshare with you or do whatever you need doing, if you can give instructions. But providing you direct remote access is going to be problematic as the VM is located within my corporate IT infrastructure.  But regardless, I am thinking there is nothing more needing to be done ... as I will explain below

But, I now think we should close this my issue (https://github.com/ipxe/ipxe/issues/155) as I myself am not able to re-CREATE a brand new VM that exhibits the problem I detailed.  I DO have an **existing** VM that shows it, but I cannot create a new one that does so. So I think we should chalk **my issue** up, for now, to some personal problem with this particular VM instance.

For the morbidly curious, following is some additional detail. 

The PXE Client VM I have here is the same VM I have been using to test iPXE all throughout 2020. It has worked fine until "the recursion commit" (c70b3e04e86cefca335e36f883829d89583a6921 ... made at the beginning of Oct).  At which point it started exhibiting the symptoms outlined in my original issue detail (https://github.com/ipxe/ipxe/issues/155).

This particular VM is "ESXi 6.0 and later (VM version 11)" ... which is somewhat down-rev'd from more current ESXi (maybe it was originally created within ESXi 5.x, I don't know?). I am confident that I originally created the VM on an older ESXi host that got upgraded last summer to our newer ESXi. So the VM has "been around the block for a while".

Just this morning I tried to create a brand NEW VM that was as close to identical to my problematic one as a I can make it (but a more current "ESXi 6.7 and later (VM version 15)" VM). But it works correctly. And then, I further tried to create a brand new down-rev'd v11 VM (thinking maybe it was the VM version itself that was the problem) ... but it also works correctly, despite the fact that this is the same VM version as my existing problematic one.

So, the problem I am observing now seems to be specific to **this particular VM**, as opposed to a general problem (at least in terms of my ability to re-create with a new VM). It is just interesting that this VM, whatever its problem is, **does work correctly** with the pre-recursion-commit iPXE.

I guess I should have re-created the VM before even submitting my issue. My apologies for any of your time I have wasted!!

NOTE: I **was** able to **suspend** the problematic VM, copy the datastore files, and open the VM again in VMWare Workstation (something that @mcb30 asked about the possibility of earlier).  But, I am not sure where to take it from there ... I don't think that this is going to allow me to actually reproduce the problem (as it would require more massaging of the VM in Workstation to get it properly configured for a end-to-end PXE boot test. I suspect that massaging will cause the problem to be unreproducible (as it was before when I tried to lift the VM into Workstation).

mcb30, 2020-11-17:
@aheusser Thank you for the detailed testing and reporting!

My best guess based on this new information is that the root cause is either:

1. Some interoperability issue between iPXE and the UEFI firmware version available when that specific virtual machine was created (on the assumption that VMware snapshots the firmware at the point of VM creation, rather than reloading from disk each time as with e.g. libvirt+KVM), or
2. Some issue that arises with UEFI drivers that are active only with the hardware configuration as defined by the older version of VMware (and hence present in that particular VM)

Either way, I would be very interested to receive the files that let you suspend and migrate the VM to VMware Workstation.  I can have a go at recreating the required bits of PXE infrastructure around this single VM, and I'd be keen to track down the problem since it seems to currently be my best shot at reproducing an issue that appears to affect multiple people.

aheusser, 2020-11-17:
@mcb30 I will gladly help. Could you provide a few additional details of what you want though?

So I could boot up the faulty ESXi VM into the UEFI shell, for example ... and then suspend and capture that state?  Or put another way:  what state would be best for for the VM to be suspended into?

I will have to play around a bit to see if I can remove the disk and reduce the memory and still have a repro (e.g., for linux) ... so that I can get the total file set down to a minimum size for you.

MaxPeal, 2020-11-17:
IMHO: If it's possible to create a VM that exhibits the problem with terraform or packer, than its a good way to integrate this in a test CI environment

LanceUMatthews, 2020-11-18:
> I myself am not able to re-CREATE a brand new VM that **exhibits** the problem I detailed. I DO have an existing VM that shows it, but I cannot create a new one that does so. So I think we should chalk **my issue** up, for now, to some personal problem with this particular VM instance.

> Just this morning I tried to create a brand NEW VM that was as close to identical to my problematic one as a I can make it (but a more current "ESXi 6.7 and later (VM version 15)" VM). But it works [...]

It's not clear if you're comparing the two VMs in the ESXi UI or by looking at their respective `.vmx` files, but if you haven't already I'd suggest copying the `.vmx` files from the datastore to your local system so you can compare them using a diff tool or, at the least, a plain text editor.  Obviously the name, version, disk identifiers, and MAC addresses will be different, but you can see if anything else jumps out at you as "hidden", potentially problematic differences.

mcb30, 2020-11-18:
@aheusser Thanks for the heads-up on Hyper-V.  I'm bringing my Hyper-V test setup up to date and will then give reproduction a try there.

mcb30, 2020-11-18:
@aheusser @Lukas-UAUX Thank you for suggesting Hyper-V.  I have a viable reproduction now.  Specifically, I have a Hyper-V VM that will reboot immediately after starting the Linux kernel, if the recursion from commit c70b3e04e is used, and will boot normally if the recursion is not used.

aheusser, 2020-11-18:
@mcb30 Fantastic! I am so relieved. It sounds symptomatically identical to my situation. So hopefully the right fix for that will address my scenario as well.

mcb30, 2020-11-22:
Update on this: within Hyper-V, using the driver veto mechanism: it is both necessary and sufficient to unload the "UDP6 Network Service Driver" to eliminate the problem, suggesting that the fault may lie within that driver.  This is also suggested by the screenshots in https://github.com/ipxe/ipxe/issues/159#issuecomment-722990278 which show a backtrace through the same Udp6Dxe driver.

The known-working EDK2 OVMF firmware (used for QEMU/KVM) does not include an IPv6 stack by default, and so does not provide the Udp6Dxe driver.  I have tried rebuilding with the IPv6 stack enabled.  The result is still able to boot successfully, but the debug console does show a failure at the time that `efi_snp_remove()` calls `UninstallMultipleProtocolInterfaces()`:

```
MnpServiceBindingDestroyChild: Failed to uninstall the ManagedNetwork protocol, Invalid Parameter.
```

The version of Udp6Dxe within OVMF will be substantially more recent than in Hyper-V (or on any physical hardware), so there is a chance that these are the same bug that happens to present as different behaviours.

I have tried forcing `efi_snp_remove()` to skip the call to `UninstallMultipleProtocolInterface()` (and hence leak memory, leaving behind nullified protocol interfaces).  This hack is sufficient to produce a successful boot on Hyper-V (even with Udp6Dxe left loaded), and also to avoid the above error message on OVMF.

I will continue to investigate further.

PicoMitchell, 2020-11-22:
Thanks for much for the update and for your hard work on this issue. It is very much appreciated.

mcb30, 2020-11-23:
A further update: the problem seems to arise from the fact that MnpDxe does not gracefully handle an SNP device that refuses to start.

iPXE has to navigate an awkward path between its own high-speed use of the network devices by its own internal network stack, and the legacy compatibility mechanism of providing an SNP protocol instance for use by the much slower and less capable built-in UEFI network stack.  This is handled via our `efi_snp_claim()` mechanism: the EFI_SIMPLE_NETWORK_PROTOCOL instances remain permanently installed but we retain a notion of whether the devices are currently reserved for use by iPXE.  Devices can be reserved for iPXE for multiple reasons: for example, it could be because iPXE is currently running as the main foreground application, or because an external caller has called us via an API such as EFI_BLOCK_IO_PROTOCOL to read from an iSCSI or FCoE disk provided by iPXE.

While devices are reserved for use by iPXE, we will unconditionally fail any EFI_SIMPLE_NETWORK_PROTOCOL API calls with EFI_NOT_READY.

With recursive connection enabled, the MnpDxe driver will attempt to attach to our EFI_SIMPLE_NETWORK_PROTOCOL instance as soon as it has been installed.  When the MnpDxe driver first attaches to our EFI_SIMPLE_NETWORK_PROTOCOL instance, it will attempt to call our `Start()` and `Initialize()` methods as part of its own initialisation sequence.  These will both currently be failed with EFI_NOT_READY.  The failure path through MnpDxe seems to leave a partially initialised handle in place, which cannot later be removed when we attempt to uninstall our EFI_SIMPLE_NETWORK_PROTOCOL instance as part of shutting down via `ExitBootServices()`.

The UEFI core, by design, handles a failure to uninstall a protocol instance in an incredibly unsafe way, and it is entirely unsurprising the such a failure leads to a system crash.

I have verified that allowing the `Start()` and `Initialize()` methods to succeed is sufficient to remove all traces of the problem in both OVMF and Hyper-V.

This workaround has the unwanted side-effect of allowing MnpDxe to open the network devices at a point when iPXE would otherwise default to leaving them closed.  I am searching for an alternative solution that allows MnpDxe to complete its initialisation cleanly without actually changing the state as seen by iPXE.

mcb30, 2020-11-23:
@aheusser @PicoMitchell A potential fix is available in #170 

If you have a chance to test it soon, that would be great.  If not, I'll merge it after some extra testing here.

Thanks!

PicoMitchell, 2020-11-23:
@mcb30 I'll be back in the office with access to the test computers on Thurs. I will be sure to test then, sorry I can't test any sooner!

[16 more replies]

PR #170 (2020-11-23): [efi] Allow initialisation via SNP interface even while claimed
Fixes: #164 
Fixes: #155 
Fixes: #159
closed 2020-12-19 by mcb30