ipxe/ipxe #165: Arm64 build with serial console missing symbols.

ericdraken, 2020-11-02:
`git pull origin master`

If I leave `//#define CONSOLE_SERIAL` then the `snp.efi` and `ipxe.efi` compile just fine.

However, when I edit src/config/console.h to enable console
```
#define CONSOLE_SERIAL          /* Serial port console */
```
And run make on an SoC with ArmV8 (Arm64) with the latest aarch64 tool chain
```
make ARCH=arm64 bin-arm64-efi/snp.efi -j8
```
The failure is
```
  ... snip ...
  [VERSION] bin-arm64-efi/version.snp.efi.o
  [HOSTCC] util/elf2efi64
  [AR] bin-arm64-efi/blib.a
ar: creating bin-arm64-efi/blib.a
  [LD] bin-arm64-efi/snp.efi.tmp
ld: bin-arm64-efi/blib.a(serial.o): in function `serial_init':
/ipxe/src/core/serial.c:145: undefined reference to `uart_select'
ld: bin-arm64-efi/blib.a(serial.o): in function `uart_data_ready':
/ipxe/src/include/ipxe/uart.h:112: undefined reference to `uart_read'
ld: /ipxe/src/include/ipxe/uart.h:112: undefined reference to `uart_read'
ld: bin-arm64-efi/blib.a(serial.o): in function `uart_receive':
/ipxe/src/include/ipxe/uart.h:124: undefined reference to `uart_read'
ld: bin-arm64-efi/blib.a(uart.o): in function `uart_transmit':
/ipxe/src/core/uart.c:54: undefined reference to `uart_read'
ld: /ipxe/src/core/uart.c:61: undefined reference to `uart_write'
ld: bin-arm64-efi/blib.a(uart.o): in function `uart_flush':
/ipxe/src/core/uart.c:75: undefined reference to `uart_read'
ld: /ipxe/src/core/uart.c:76: undefined reference to `uart_read'
ld: bin-arm64-efi/blib.a(uart.o): in function `uart_exists':
/ipxe/src/core/uart.c:95: undefined reference to `uart_write'
ld: /ipxe/src/core/uart.c:96: undefined reference to `uart_read'
ld: /ipxe/src/core/uart.c:98: undefined reference to `uart_write'
ld: /ipxe/src/core/uart.c:99: undefined reference to `uart_read'
ld: bin-arm64-efi/blib.a(uart.o): in function `uart_init':
/ipxe/src/core/uart.c:124: undefined reference to `uart_read'
ld: /ipxe/src/core/uart.c:126: undefined reference to `uart_write'
ld: /ipxe/src/core/uart.c:131: undefined reference to `uart_write'
ld: /ipxe/src/core/uart.c:132: undefined reference to `uart_write'
ld: /ipxe/src/core/uart.c:138: undefined reference to `uart_write'
ld: /ipxe/src/core/uart.c:141: undefined reference to `uart_write'
ld: bin-arm64-efi/blib.a(uart.o):/ipxe/src/core/uart.c:144: more undefined references to `uart_write' follow
ld: bin-arm64-efi/blib.a(uart.o): in function `uart_init':
/ipxe/src/core/uart.c:134: undefined reference to `uart_read'
ld: /ipxe/src/core/uart.c:135: undefined reference to `uart_read'
ld: bin-arm64-efi/snp.efi.tmp: hidden symbol `uart_select' isn't defined
ld: final link failed: bad value
```
From https://github.com/ipxe/ipxe/search?q=uart_select I can see that these missing UART methods are present in the x86 version of `uart.c`: `src/arch/x86/core/x86_uart.c` but no where else.

The problem I'm trying to solve is the iPXE menu doesn't recognize keyboard commands like up, down, nor enter. Even chain booting to an iPXE shell doesn't allow typing at all. Suggestions?

NiKiZe, 2020-11-02:
The serial implementation is one of those things that have not been ported to EFI land,
Does EFI have an serial interface/protocol?
(the implementation that exists uses legacy bios calls to open and manage the port, that needs to be re-implemented for EFI)

I think this is at most the second time it has even come up.

Thinking further, in regards to EFI it should be the efi console driver that should be responsible for the serial console as well. - maybe you should switch focus to the problem of keys not working in your environment and try and debug that instead?
Start with enabling debug of the efi console code in iPXE and see if there is anything we can do there?

mcb30, 2020-11-02:
Under UEFI, iPXE uses the ConIn and ConOut handles provided by the UEFI platform firmware for console I/O.  Most UEFI implementations will connect these to a multiplexer that in turn connects to a local framebuffer console, any local USB keyboards, and usually the first serial port.  The UEFI setup screen may offer the ability to enable/disable some of these console devices.

Either way, iPXE is not usually responsible for directly controlling console hardware devices under UEFI.  The exceptions are when using a console type that is not supported natively by UEFI (e.g. the syslog console).

It would be possible to implement UART support for the hardware found on your Arm64 SoC, but:
* it shouldn't be necessary, because the UEFI firmware should be handling the UART anyway
* if the UEFI firmware *is* trying to handle the UART, then having iPXE also drive the same hardware is likely to cause problems.

How are you creating the UEFI firmware build for your SoC?  Does serial console I/O work in any other UEFI applications (e.g. the UEFI shell)?

ericdraken, 2020-11-02:
Hello, @NiKiZe and @mcb30, thanks for responding. 

I'm convinced this isn't a true UART issue. It can't be. U-Boot PXE loads the snp.efi (and even the ipxe.efi), displays the iPXE info text, and continues on to request the boot.ipxe or menu.ipxe. 

At the iPXE menu, all serial console input is ignored. The arrow keys, enter, ctrl+b, esc+b, etc. are non-resultant. Even when I instruct boot.ipxe to immediately drop to shell, no input appears on the `iPXE>` prompt.

This only happens with iPXE. The PXE shell works as intended (accepts keyboard input). There are no changes to hardware: serial cable to UART0 on the SoC.

This could be an XY problem, but my starting point is to "enable serial console" to get iPXE continuing. Any suggestions? Is this a UART issue?

---

> How are you creating the UEFI firmware build for your SoC?

# On the physical Arm64 SoC with the aarch64 toolchain:
make ARCH=arm64 bin-aarch64-efi/snp.efi  # Compiles just fine, without enabling serial console

---

> maybe you should switch focus to the problem of keys not working in your environment and try and debug that instead?
> Start with enabling debug of the efi console code in iPXE and see if there is anything we can do there?

Certainly. I changed LOGS_NONE to LOGS_ALL but not much logging took place on recompilation. Using a boot.ipxe that is just

```
#!ipxe
shell
```

I get:
```
...
Found 0 disks
## Starting EFI application at 40080000 ....
efi_load_pe: Invalid DOS Signature
iPXE initialisating devices...ok

iPXE 1.20.1+ (g9b25) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP AoE EFI Menu
Configuring (net0 ...mac address...)...... ok
Booting from boot.ipxe
boot.ipxe... ok
efi_load_pe: Invalid DOS Signature
iPXE>
```

Again, no input is accepted.

file snp.ipxe
`MS-DOS executable PE32+ executable (DLL) (EFI application) Aarch64, for MS Windows`

I don't know what is up with `efi_load_pe: Invalid DOS Signature` shown twice, however.

NiKiZe, 2020-11-02:
You say no input is accepted, but you only list "special" keys.
So to be clear, if you get to a iPXE shell no input at all, not even a-zA-Z0-9 works?

can you try to add `DEBUG=efi_console:3` at the end of your make line?

ericdraken, 2020-11-02:
> So to be clear, if you get to a iPXE shell no input at all, not even a-zA-Z0-9 works?

Correct. Mashing on the keyboard does nothing either ;)

> can you try to add DEBUG=efi_console:3 at the end of your make line?

Interesting. Adding that line at the end of the make line added no new information. I even removed an EMBED I missed to make debugging easier. Now it is:

```
...
Found 0 disks
## Starting EFI application at 40080000 ....
iPXE initialisating devices...ok

iPXE 1.20.1+ (g9b25) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP AoE EFI Menu

Press Ctrl+B for the iPXE command line...
```
Again, no keyboard input is accepted.

`make ARCH=arm64 bin-aarch64-efi/snp.efi DEBUG=efi_console:3`

Stumped.

ericdraken, 2020-11-03:
I've isolated the execution flow to the prompt method
```
	/* Prompt user */
	printf ( "\n" );
	return ( prompt ( "Press Ctrl-B for the " PRODUCT_SHORT_NAME
			  " command line...",
			  ( ( BANNER_TIMEOUT * TICKS_PER_SEC ) / 10 ),
			  CTRL_B ) == 0 );
```
then
```
	/* Wait for key */
	key_pressed = getkey ( timeout );
```
then I get lost in `getkey.c`. 
```
	while ( ( timeout == 0 ) || ( ( currticks() - start ) < timeout ) ) {
		step();
		if ( iskey() )
			return getchar();
		cpu_nap();
	}
```
Could something be blocking input when compiled for ARM64?

ericdraken, 2020-11-03:
Update: So debug messages aren't helpful, so I put a whole bunch of printf() statements in get key.c, and here is where we are stuck:

```
iPXE 1.20.1+ (g9b25) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP AoE EFI Menu

getchar_timeout() - cpu_nap()mand line...
````
The whole execution flow halts at `cpu_nap()` and doesn't come back. Any ideas?

NiKiZe, 2020-11-03:
Have you tried to reach out to the makers of the EFI firmware for assistance? Maybe they can help to debug the issue, might be a good idea to add workarounds both for them and in iPXE if/when we can figure out how to fix it.

ericdraken, 2020-11-04:
I figured it out. I found an archived email thread (https://ipxe-devel.ipxe.narkive.com/wkDSSmnD/patch-1-1-efi-provide-possibility-to-disable-cpu-nap). The UEFI emulation of UBoot (just on ARM?) is single-threaded, so the event-driven interrupt of `cpu_nap()` will never fire. 

One contentious solution (read: hack) is to essentially remove that `cpu_nap()` line from a few places and let the CPU run hot for all of a few seconds.

Another is to adjust flags in `nap.h`:

```
#undef NAP_EFIX86
#undef NAP_EFIARM
#define NAP_NULL
```

The best solution is to get on the UEFI implementors to provide interrupt support. Thank you everyone for your replies! iPXE is wonderful.

closed 2020-11-04 by ericdraken