ipxe/ipxe #166: TX failure with i219-lm(4)

raphaelmeyer, 2020-11-06:
I ran into a problem with an _Intel i219-lm(4)_ based nic.
I have a setup where I use iPXE to boot a [Genode](https://github.com/genodelabs/genode) based OS. Genode uses a [port of iPXE](https://github.com/genodelabs/genode/tree/master/repos/dde_ipxe) as its generic network interface driver. Sometimes the nic then fails to transmit packets. The error occurs more often if the cable is removed while booting the OS and reconnected after the nic driver turned iPXE is running.
I found an [older issue](https://forum.ipxe.org/showthread.php?tid=10280) with similar behaviour but there already exists a [fix](https://github.com/ipxe/ipxe/commit/546dd51).

I came to the following conclusions while investigating the error:
When the OS is started and the link is up the network stack starts to send DHCP discover packets. The first transmit attempt is acknowledged with the TXDW bit set in the interrupt cause register. But I could not see any packet on the net (wireshark). So the packet was never sent or seriously corrupted. All following transmit attempts won't even be acknowledged and the TXDW bit is never set again.
The RX queue on the other hand seems to be working.

I could recover from this situation by executing the following sequence.

    netdev_link_down(...);
    netdev_close(...);
    netdev_open(...);
    netdev_irq(..., 1);

The error sometimes can also be reproduced by unplugging and then re-plugging the network cable. Therefore I think the error might be connected to the MAC being resetted without a power cycle. As far as I understood the PHY is not resetted because of other hardware related issues.

I suspect the issue to be caused by a hardware issue. The comments and workarounds in the iPXE and the linux source code regarding said hardware family may support my suspicion.

Is there anybody else who already observed and may confirm the issue? Is it feasible to add some kind of device reset (as shown above) to the intel driver?

NiKiZe, 2020-11-06:
And this is tested with the latest version of iPXE?

mcb30, 2020-11-06:
@raphaelmeyer Are you able to reproduce the problem with iPXE, or does this occur only when using iPXE's driver within Genode?

raphaelmeyer, 2020-11-10:
Thank you very much for your inputs.

Indeed, I could not reproduce the issue with iPXE itself. I tried by setting up an infinite iPXE loop, even with adding undionly.kpxe once before entering the loop as well.

I am not sure about how recent the iPXE code in genode is. The driver `drivers/net/intel.c` is basically the same as in iPXE master, but I found a few differences in `net/netdevice.c`. The issue was still present when I incoporated these differences into the genode code.

These insights will help to further narrow down the issue. Probably it's in how _netdevice_ is used by genode as a driver interface.

Therefore, I close the issue, but I would still be very happy to receive further feedback.

closed 2020-11-10 by raphaelmeyer

mcb30, 2020-11-10:
@raphaelmeyer Since you have a sequence of operations in Genode that is able to recover from the situation, I would suggest bisecting this sequence to isolate the minimum operation(s) required to effect a recovery.

For example, the call to `netdev_link_down()` should be superfluous since that will not touch the hardware in any way (it affects only the internally recorded link state).  Bisecting the operations performed by `netdev_close()` and `netdev_open()` is harder, but you can start with bypassing the netdevice core by calling `intel_close()` and `intel_open()` directly, then try various bits of those routines to see what minimally clears the error condition.