ipxe/ipxe #167: Initrd support for AArch64

ericdraken, 2020-11-12:
Is there any trick to getting `initrd` to load for AArch64?

`Kernel` loads fine. But I have a 100% failure rate for loading various Linux flavors of `initrd` and `initramfs` build for AArch64. iPXE was built as such:

```
# Fix the problem when PXE is single-threaded, but iPXE uses interrupts,
#  and prevent redefining cpu_nap() by avoiding efiarm_nap.h from loading.
    echo "#undef  NAP_PCBIOS"   >> config/local/nap.h &&\
    echo "#define NAP_NULL"     >> config/local/nap.h &&\
    echo ""                      > arch/arm/include/bits/nap.h
 
# Add more menu items
    echo "#define DOWNLOAD_PROTO_NFS"   >> config/local/general.h &&\
    echo "#define DOWNLOAD_PROTO_HTTPS" >> config/local/general.h &&\
    echo "#define PING_CMD"             >> config/local/general.h &&\
    echo "#define IPSTAT_CMD"           >> config/local/general.h &&\
    echo "#define NSLOOKUP_CMD"         >> config/local/general.h &&\
    echo "#define TIME_CMD"             >> config/local/general.h &&\
    echo "#define REBOOT_CMD"           >> config/local/general.h &&\
    echo "#define POWEROFF_CMD"         >> config/local/general.h &&\
    echo "#define IMAGE_SCRIPT"         >> config/local/general.h &&\
    echo "#define IMAGE_EFI"            >> config/local/general.h

make ARCH=arm64 bin-arm64-efi/ipxe.efi -j8 DEBUG=image:3
```

Attempts to load various AArm64 initrds:

```
iPXE> initrd http://10.10.0.200/images/debian-installer/arm64/initrd   
http://10.10.0.200/images/debian-installer/arm64/initrd... ok 
IMAGE initrd at [571a2000,5c1eb200) registered
Invalid DOS Signature
IMAGE initrd is not EFI: Error 0x7f048181 (http://ipxe.org/7f048181)
IMAGE initrd is not script: Exec format error (http://ipxe.org/2e028081)
IMAGE initrd is not DER: Exec format error (http://ipxe.org/2e088081)
IMAGE initrd is not PEM: No such file or directory (http://ipxe.org/2d09808e)
IMAGE initrd format not recognised

iPXE> initrd http://10.10.0.200/images/alpine-aarch64/initramfs-lts
http://10.10.0.200/images/alpine-aarch64/initramfs-lts... ok
IMAGE initramfs-lts at [54cb2000,5614e74e) registered
Invalid DOS Signature
IMAGE initramfs-lts is not EFI: Error 0x7f048181 (http://ipxe.org/7f048181)
IMAGE initramfs-lts is not script: Exec format error (http://ipxe.org/2e028081)
IMAGE initramfs-lts is not DER: Error 0x3e00e08e (http://ipxe.org/3e00e08e)
IMAGE initramfs-lts is not PEM: No such file or directory (http://ipxe.org/2d09808e)
IMAGE initramfs-lts format not recognised

iPXE> initrd http://10.10.0.200/images/debian/initrd.gz              
http://10.10.0.200/images/debian/initrd.gz... ok
IMAGE initrd.gz at [56152000,5719e3be) registered
Invalid DOS Signature
IMAGE initrd.gz is not EFI: Error 0x7f048181 (http://ipxe.org/7f048181)
IMAGE initrd.gz is not script: Exec format error (http://ipxe.org/2e028081)
IMAGE initrd.gz is not DER: Error 0x3e00e08e (http://ipxe.org/3e00e08e)
IMAGE initrd.gz is not PEM: No such file or directory (http://ipxe.org/2d09808e)
IMAGE initrd.gz format not recognised
```

This is the same error for the x86-64 initrd files:

```
iPXE> imgfetch http://boot.ipxe.org/1mb
http://boot.ipxe.org/1mb........... ok            
IMAGE 1mb at [52e39000,52f39000) registered
Invalid DOS Signature
IMAGE 1mb is not EFI: Error 0x7f048181 (http://ipxe.org/7f048181)
IMAGE 1mb is not script: Exec format error (http://ipxe.org/2e028081)
IMAGE 1mb is not DER: Error 0x3e00e08e (http://ipxe.org/3e00e08e)
IMAGE 1mb is not PEM: No such file or directory (http://ipxe.org/2d09808e)
IMAGE 1mb format not recognised
```

For AArch64, I'm tried renaming initrd to initrd.gz, uncompressing initrd.gz to initrd, and I've tried various netboot ISOs initrds for AArch64, but nothing succeeds.

I've noticed that all the initrd code (https://github.com/ipxe/ipxe/search?q=initrd) is under the x86 arch folder. Does that matter?

```
iPXE 1.20.1 (g8f15) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP HTTPS iSCSI NFS TFTP SRP AoE EFI Menu
```

NiKiZe, 2020-11-12:
Could you test without debug and then tell us what the error is in that build?, and also without modifying more than you actually need in the config, for example IMAGE_SCRIPT and IMAGE_EFI should not need to be changed, (and there is multiple examples where users have tried to set incompatible configs, and got a variety of issues)

ericdraken, 2020-11-13:
I have turned off those extra configs, ran the build again - no errors or warnings:

```
rm -f  bin-arm64-efi/*.*  bin-arm64-efi/.certificate.* bin-arm64-efi/.certificates.* bin-arm64-efi/.private_key.* bin-arm64-efi/errors	 bin-arm64-efi/NIC	 ./util/zbin ./util/elf2efi32 ./util/elf2efi64 ./util/efirom ./util/efifatbin ./util/iccfix ./util/einfo TAGS bin-arm64-efi/symtab
  [PARSEROM]
  [DEPS] arch/arm64/core/setjmp.S
  [DEPS] drivers/net/ath/ath9k/ath9k_eeprom_4k.c
  ... snip ...
  [BUILD] bin-arm64-efi/ath5k.ids.o
  [BUILD] bin-arm64-efi/ath9k.ids.o
  [HOSTCC] util/elf2efi64
  [VERSION] bin-arm64-efi/version.ipxe.efi.o
  [AR] bin-arm64-efi/blib.a
/usr/bin/aarch64-linux-gnu-ar: creating bin-arm64-efi/blib.a
  [LD] bin-arm64-efi/ipxe.efi.tmp
  [FINISH] bin-arm64-efi/ipxe.efi
rm bin-arm64-efi/version.ipxe.efi.o
```

I've run the iPXE initrd commands again on AArch64 images:

```
Type "exit" to return to menu.
iPXE> initrd http://10.10.0.200/images/debian-installer/arm64/initrd
http://10.10.0.200/images/debian-installer/arm64/initrd... ok 
Invalid DOS Signature
iPXE> initrd http://10.10.0.200/images/alpine-aarch64/initramfs-vanilla
http://10.10.0.200/images/alpine-aarch64/initramfs-vanilla... ok 
Invalid DOS Signature
iPXE> initrd http://10.10.0.200/images/debian/initrd.gz
http://10.10.0.200/images/debian/initrd.gz... ok
Invalid DOS Signature
iPXE> initrd http://10.10.0.200/images/debian-installer/arm64/initrd
http://10.10.0.200/images/debian-installer/arm64/initrd... ok 
Invalid DOS Signature
iPXE> initrd http://10.10.0.200/images/alpine-aarch64-lts/initramfs-lts    
http://10.10.0.200/images/alpine-aarch64-lts/initramfs-lts... ok 
Invalid DOS Signature
iPXE> 
```

I've rebuilt this so many times, so I've make a Dockerfile for building v1.20.1:

```
FROM debian:buster-slim
 
RUN apt update &&\
    apt install -qy gcc-aarch64-linux-gnu build-essential libssl-dev git ccache

# Do not checkout master as it is unstable
RUN git clone --depth=1 -b v1.20.1 https://github.com/ipxe/ipxe.git
 
WORKDIR /ipxe/src
 
# Use ccache to make subsequent recompilations faster
ENV make_cmd="make ARCH=arm64 CROSS_COMPILE='ccache aarch64-linux-gnu-' bin-arm64-efi/ipxe.efi -j8"
 
# Run the compilation once to help out ccache
RUN /bin/sh -c "$make_cmd"
 
# Fix the problem when PXE is single-threaded, but iPXE uses interrupts,
#  and prevent redefining cpu_nap() by avoiding efiarm_nap.h from loading.
RUN echo "#undef  NAP_PCBIOS"   >> config/local/nap.h &&\
    echo "#define NAP_NULL"     >> config/local/nap.h &&\
    echo ""                      > arch/arm/include/bits/nap.h

RUN mkdir /out
 
CMD /bin/sh -c "$make_cmd" &&\
    cd bin-arm64-efi &&\
    chown nobody:nogroup ipxe.efi &&\
    mv ipxe.efi ipxe-$(date +"%Y-%m-%d").efi &&\
    cp -p *.efi /out &&\
    cd /out &&\
    ls -al
```

Any suggestions to try next?

ericdraken, 2020-11-13:
Additionally, if I try to build the latest version from master with absolutely no editing of configs...

```
FROM debian:buster-slim
 
RUN apt-get update &&\
    apt-get install -qy gcc-aarch64-linux-gnu build-essential libssl-dev git ccache

RUN git clone --depth=1 https://github.com/ipxe/ipxe.git
 
WORKDIR /ipxe/src
 
# Use ccache to make subsequent recompilations faster
ENV make_cmd="make ARCH=arm64 CROSS_COMPILE='ccache aarch64-linux-gnu-' bin-arm64-efi/ipxe.efi -j8"

RUN mkdir /out
 
CMD /bin/sh -c "$make_cmd" &&\
    cd bin-arm64-efi &&\
    chown nobody:nogroup ipxe.efi &&\
    mv ipxe.efi ipxe-$(date +"%Y-%m-%d").efi &&\
    cp -p *.efi /out &&\
    cd /out &&\
    ls -al
```

there are indeed errors

``` 
... snip ...
  [BUILD] bin-arm64-efi/rtl8185.ids.o
  [BUILD] bin-arm64-efi/rtl8180.ids.o
  [BUILD] bin-arm64-efi/ath5k.ids.o
  [BUILD] bin-arm64-efi/ath9k.ids.o
  [HOSTCC] util/elf2efi64
  [VERSION] bin-arm64-efi/version.ipxe.efi.o
In file included from include/ipxe/features.h:6,
                 from core/version.c:33:
core/version.c:48:47: error: expected expression before ',' token
 FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
                                               ^
include/ipxe/dhcp.h:529:58: note: in definition of macro 'DHCP_OPTION'
 #define DHCP_OPTION( ... ) VA_ARG_COUNT ( __VA_ARGS__ ), __VA_ARGS__
                                                          ^~~~~~~~~~~
include/ipxe/features.h:71:2: note: in expansion of macro '__DHCP_FEATURE'
  __DHCP_FEATURE ( _name, feature_opt, __VA_ARGS__ )
  ^~~~~~~~~~~~~~
include/ipxe/features.h:69:2: note: in expansion of macro '_DHCP_FEATURE'
  _DHCP_FEATURE ( OBJECT, feature_opt, __VA_ARGS__ )
  ^~~~~~~~~~~~~
include/ipxe/features.h:106:2: note: in expansion of macro 'DHCP_FEATURE'
  DHCP_FEATURE ( DHCP_ENCAPSULATED ( DHCP_EB_VERSION ), __VA_ARGS__ )
  ^~~~~~~~~~~~
core/version.c:48:1: note: in expansion of macro 'FEATURE_VERSION'
 FEATURE_VERSION ( VERSION_MAJOR, VERSION_MINOR, VERSION_PATCH );
 ^~~~~~~~~~~~~~~
core/version.c:66:48: error: expected expression before ';' token
 const int product_minor_version = VERSION_MINOR;
                                                ^
core/version.c:66:48: error: initialization of 'int' from 'uint8_t *' {aka 'unsigned char *'} makes integer from pointer without a cast [-Werror=int-conversion]
core/version.c:66:48: error: initializer element is not constant
cc1: error: unrecognized command line option '-Wno-address-of-packed-member' [-Werror]
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:1177: bin-arm64-efi/version.ipxe.efi.o] Error 1
make: *** Waiting for unfinished jobs...
```

ericdraken, 2020-11-13:
Ah, I see what was changed since the 1.20.1....

```
###############################################################################
#
# Version number calculations
#
ifneq ($(wildcard ../.git),)
VERSIONS := $(shell git describe --tags --always --long --abbrev=1 --match "v*")
VERSION_TUPLE	:= $(subst ., ,$(subst -, ,$(patsubst v%,%,$(VERSIONS))))
VERSION_MAJOR	:= $(word 1,$(VERSION_TUPLE))
VERSION_MINOR	:= $(word 2,$(VERSION_TUPLE))
VERSION_PATCH	:= $(word 3,$(VERSION_TUPLE))
...
```
Then `git clone --depth=1 https://github.com/ipxe/ipxe.git` will not work. It has to be the full `git clone https://github.com/ipxe/ipxe.git`. Now the bleeding-edge version compiles, but is stuck on the iPXE loading screen:

```
iPXE 1.20.1+ (g0e262) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

Press Ctrl-B for the iPXE command line...   <-- Stuck here
```

NiKiZe, 2020-11-13:
I think you are mixing things..
All errors from iPXE has an ipxe.org URL.
There is no ipxe.org URL, thus there is no iPXE errors.

I think you are jumping to the conclusion that "Invalid DOS Signature" is an error, but you are not clear if that is what you do mean.

Try again to do a initrd call, followed by a imgstat call, and let's look at the relevant parts only.

In regards to build errors, it should have been a separate issue, and be clearer about what your build command is and what the error is. Also avoid other things such as (for us) irrelevant scripts etc. But happy that you got to know what the issue was and a fix.

ericdraken, 2020-11-13:
> I think you are mixing things..

I would love to be mixing things up so I know where to start. I've been at this for so long.

> Try again to do a initrd call, followed by a imgstat call, and let's look at the relevant parts only.

Certainly.
```
iPXE> imgstat                                                        
initrd : 84185600 bytes
initramfs-vanilla : 74651436 bytes
initrd.gz : 17089470 bytes
initramfs-lts : 30879370 bytes
vmlinuz-lts : 6385244 bytes
linux : 18761584 bytes [EFI]
vmlinuz-vanilla : 12102144 bytes [EFI] [SELECTED]
iPXE> 
```
> I think you are jumping to the conclusion that "Invalid DOS Signature" is an error

That may very well be. When I had logs turned on, I had

```
iPXE> initrd http://10.10.0.200/images/debian/initrd.gz              
http://10.10.0.200/images/debian/initrd.gz... ok
IMAGE initrd.gz at [56152000,5719e3be) registered
Invalid DOS Signature
IMAGE initrd.gz is not EFI: Error 0x7f048181 (http://ipxe.org/7f048181)
IMAGE initrd.gz is not script: Exec format error (http://ipxe.org/2e028081)
IMAGE initrd.gz is not DER: Error 0x3e00e08e (http://ipxe.org/3e00e08e)
IMAGE initrd.gz is not PEM: No such file or directory (http://ipxe.org/2d09808e)
IMAGE initrd.gz format not recognised
```

The underlying symptom is that an `initrd` cannot be selected so that `boot` works here. For example,
```
http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/vmlinuz-vanilla.... ok                      
http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/initramfs-vanilla... ok 
Invalid DOS Signature
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map... <-- frozen here
```

However, without `initrd` I can start a boot up until kernel panic (of course):

```
iPXE> kernel http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/vmlinuz-vanilla
http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/vmlinuz-vanilla.... [dl-cdn.alpinelinux.or... ok                      
iPXE> imgstat
vmlinuz-vanilla : 12102144 bytes [EFI] [SELECTED]
iPXE> boot
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map...
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 4.19.118-0-vanilla (buildozer@build-3-9-aarch64) (gcc version 8.3.0 (Alpine 8.3.0)) #1-Alpine SMP Thu Apr 23 11:18:42 UTC 2020
[    0.000000] Machine model: linux,dummy-virt
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: EFI v2.80 by Das U-Boot
[    0.000000] efi:  SMBIOS=0x5dddd000 
...
[    0.639934] No filesystem could mount root, tried: 
[    0.639951] 
[    0.640458] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    0.641421] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.19.118-0-vanilla #1-Alpine
[    0.641954] Hardware name:  , BIOS 2021.01-rc1-g896cc5aa 11/06/2020
[    0.642487] Call trace:
[    0.642830]  dump_backtrace+0x0/0x168
[    0.642994]  show_stack+0x24/0x30
[    0.643176]  dump_stack+0xa0/0xc8
[    0.643304]  panic+0x134/0x280
[    0.643489]  mount_block_root+0x240/0x2c8
[    0.643639]  mount_root+0x84/0x90
[    0.643843]  prepare_namespace+0x19c/0x1ac
[    0.644052]  kernel_init_freeable+0x324/0x3a4
[    0.644230]  kernel_init+0x18/0x10c
[    0.644391]  ret_from_fork+0x10/0x18
[    0.644940] Kernel Offset: disabled
[    0.645214] CPU features: 0x0,24002004
[    0.645426] Memory Limit: none
[    0.645890] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0) ]---
```

NiKiZe, 2020-11-13:
Lovely, we see that you have loads of thing loaded, great! and no errors!

initrd tries to parse the files, in that case you see return codes, are you expecting the initrd file that you have to be parsable as any of those formats?

Are you sure you want to select your initrd for boot? (normally you want to boot the kernel which loads the initrd, and not boot the initrd)

What is your kernel commandline? (I don't think you have one?)
With efi you always have to tell the kernel which file it should load from EFIFS as initrd,
See http://boot.ipxe.org/demo/ for a working example of script (for x86 but you still need the initrd commandline)

NiKiZe, 2020-11-13:
Searching for: ipxe "not syncing: VFS: Unable to mount root" gives you lots of explanations on this.

ericdraken, 2020-11-14:
Nice. I will follow up on that search shortly.

> Are you sure you want to select your initrd for boot? (normally you want to boot the kernel which loads the initrd, and not boot the initrd)

Just for reproducibility, I'm using https://boot.alpinelinux.org/boot.ipxe. The templated iPXE commands are:

```
imgfree
kernel ${img-url}/vmlinuz-${flavor} ${cmdline} alpine_repo=${repo-url} modloop=${modloop-url} ${console} ${acpi} ${ssh_key}
initrd ${img-url}/initramfs-${flavor}
boot
```
which then become
```
imgfree
kernel http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/vmlinuz-vanilla modules=loop,squashfs alpine_repo=http://dl-cdn.alpinelinux.org/alpine/v3.9/main modloop=http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/modloop-vanilla console=tty0 acpi=force
initrd http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/initramfs-vanilla
boot
```
By the way, if I try type this manually, there seems to be a character limit of 255 on the CLI so I cannot fully test the kernel command by hand:
```
kernel http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/vmlinuz-vanilla modules=loop,squashfs alpine_repo=http://dl-cdn.alpinelinux.org/alpine/v3.9/main modloop=http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9 <-- cannot type more than this
```

NiKiZe, 2020-11-14:
Still missing initrd=filename on the kernel line

That script will only work as-is on pcbios, not efi

ericdraken, 2020-11-14:
> Still missing initrd=filename on the kernel line

Understood.

I've tried that as well, I'm afraid.

```
#!ipxe
kernel http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/vmlinuz-vanilla modules=loop,squashfs alpine_repo=http://dl-cdn.alpinelinux.org/alpine/v3.9/main modloop=http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/modloop-vanilla console=tty0 acpi=force initrd=initramfs-vanilla
initrd http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/initramfs-vanilla
boot
```
but fortune is still not on my side.  Back to the same Invalid DOS signature:

```
iPXE 1.20.1+ (g0e262) -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

net0: 52:55:00:d1:55:06 using SNP on SNP-0x5eeff350 (open)
  [Link:up, TX:0 TXE:0 RX:0 RXE:0]
Configuring (net0 52:55:00:d1:55:06)....... ok
net0: 10.10.0.115/255.255.255.0 gw 10.10.0.128
net0: fe80::5055:ff:fed1:5506/64
Next server: 10.10.0.128
Filename: /tftp/menu.ipxe
tftp://10.10.0.128//tftp/menu.ipxe... ok
Invalid DOS Signature
menu.ipxe : 4766 bytes [script]
http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/vmlinuz-vanilla..... ok                      ]
http://dl-cdn.alpinelinux.org/alpine/v3.9/releases/aarch64/netboot-3.9.6/initramfs-vanilla... ok 
Invalid DOS Signature
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map... <-- frozen
```
I'll keep experimenting. It's been weeks.

NiKiZe, 2020-11-14:
And as already explained, your DOS signature line is not from iPXE, and there is nothing that says it is an error. It is a message nothing else (you need to debug the EFI firmware if you want more info). What does the kernel say during boot?

ericdraken, 2020-11-16:
You're quite right.
```
	dos = efi;
	if (dos->e_magic != IMAGE_DOS_SIGNATURE) {
		log_err("Invalid DOS Signature\n");
		ret = EFI_LOAD_ERROR;
		goto err;
	}
```
That error message is coming from U-Boot. Perhaps I was under the misunderstanding that U-Boot handed off bootloading completely to iPXE. I appreciate your responses. I will keep looking into this.

closed 2020-11-16 by ericdraken

NiKiZe, 2020-11-16:
Again, a message about it not being a DOS image, NOT an error as your prev log shows that everything is loaded correctly.

If you try to boot the kernel with initrd=filename on the kernel line so that it knows which file to load as United, what is your last 10 lines of output?

mcb30, 2021-01-01:
> ```
> tftp://10.10.0.128//tftp/menu.ipxe... ok
> Invalid DOS Signature
> ```

For the record: this will be coming from U-Boot's implementation of the UEFI `LoadImage()` boot services call.  iPXE will call this as part of detecting the format of the loaded `menu.ipxe` file.  The failure is expected (since the image genuinely isn't a DOS/UEFI image: it's an iPXE script) and the UEFI firmware would not normally produce any kind of console output.

So, the `Invalid DOS Signature` message is basically a stray debug message coming from U-Boot that can be safely ignored.  The fact that you get as far as

> ```
> EFI stub: Booting Linux Kernel...
> EFI stub: Using DTB from configuration table
> EFI stub: Exiting boot services and installing virtual address map
> ```

indicates that iPXE has already completed execution and passed control to the loaded Linux kernel.