ipxe/ipxe #168: Compiling iPXE for VMXNET3 does not work

MasterWayZ, 2020-11-18:
Hi,

Today I tried cloning the repo and building the VMXNET3 driver from source.

The VM building is Ubuntu 18.04 Server.

I run `make bin/15ad07bo.rom` and this creates the rom file. However, when attempting to use it in a VM, I get this in the VMware log:
`BIOS 15ad07b0.rom has unexpected file size 0x10600`

Whenever I download and use a rom file that someone else made years ago, then it does work, so this is really weird.
Am I doing something wrong?

Also, a question: is there a way to build this for UEFI?

NiKiZe, 2020-11-19:
Documentation on building can be found at https://ipxe.org/appnote/buildtargets

MasterWayZ, 2020-11-19:
Thanks.
That still leaves me with that I get that error. On an older ROM file it works, but if I clone and make, I get this error as I mentioned above:
`BIOS 15ad07b0.rom has unexpected file size 0x10600`
Which is very odd. It may seem like an ESXi setting, however there's no setting I believe for it for VMXNET3. The suspected only-E1000 setting, even if I change that to the size of the ROM file in bytes, it says that.

NiKiZe, 2020-11-19:
Have you read https://ipxe.org/howto/vmware ?
Have you tried searching for that message?

A page which seams to mention your message, https://thewayeye.net/2012/april/1/pxe-booting-virtual-machines-using-vmware-fusionworkstation-and-gpxe-or-ipxe/ easily found with a quick search on the net ;)

MasterWayZ, 2020-11-19:
I have read that and tried that yes.
Like I mentioned, if I download a file that has already been build ( https://s3.amazonaws.com/twe-public-fs/15ad07b0.rom for example), it gives no errors. But whenever I use my build it gives that error.
Even if I set ethernet0.opromsize to the filesize in bytes as it says so there it gives the error. Which I believe is ignored for VMXNET3. But just as a sanity check, I did it anyway to the filesize if the .rom file in bytes, but same error.

mcb30, 2020-12-28:
A quick experiment suggests that the VMXNET3 virtual hardware implements only a 64kB expansion ROM BAR, which is slightly too small to support the current iPXE ROM image.  You could try disabling some unwanted features (e.g. iSCSI) via `config/general.h`.

This should be a trivial fix in VMware, most likely just requiring a single `#define` to be updated from 64kB to 128kB.  If you have a commercial relationship with VMware then you could try opening a support request with them.

Closing since this cannot be fixed from the iPXE side, short of reducing the default feature set even further.

Michael

closed 2020-12-28 by mcb30

woa7, 2021-01-11:
I can Open a support request at VMware,
If I get a test scrip or the other need infos for the ticket.
(With govc for the vm creation or so)

mcb30, 2021-01-14:
@MasterWayZ @woa7 Thanks for the offer.  I've made contact with someone on VMware's firmware team, so there's a chance we might get the BAR change made on the VMware side.  I'll update this issue if there's more to report.

mcb30, 2021-01-22:
> @MasterWayZ @woa7 Thanks for the offer. I've made contact with someone on VMware's firmware team, so there's a chance we might get the BAR change made on the VMware side. I'll update this issue if there's more to report.

FYI: this change is now planned to be made and should hopefully appear in the next VMware release.

Elegant996, 2021-02-12:
@mcb30 Did this change make it into ESXi 7.0U1d? We're delaying our update at the moment but this might be push us to do it sooner rather than later. Thanks!

mcb30, 2021-02-12:
@Elegant996 I don't know, sorry.

woa7, 2021-06-17:
can this be connected? https://github.com/ipxe/ipxe/issues/349