ipxe/ipxe #172: Issues to make the kernel read filesystem.squashfs loaded imgfetch

kaleal, 2020-11-25:
Hey there,

I am willing to enable imgverify to check signatures for _vmlinuz_, _initrd.img_ and _filesystem.squashfs_ and I am facing issues with squashfs on Debian.

On the current setup (with no signing verification) the fetch parameter is passed to vmlinuz with the filesystem.squash url and everything works fine. The issue appears when I stop using _fetch_ and start using _imgfetch_ in order to check the signature first.

Old setup:
```
set install_params boot=live live-netdev=eth0 ethdevice=eth0 ethdevice-timeout=120 dhcp ...

imgfree
echo Boot parameters: ${install_params}
kernel ${mirror_address}${mirrordir}vmlinuz fetch=${clevosmirror}${mirrordir}filesystem.squashfs ${install_params}
initrd ${mirror_address}${mirrordir}initrd.img
boot
```

New setup (which fails):
```
set install_params boot=live live-netdev=eth0 ethdevice=eth0 ethdevice-timeout=120 dhcp ...

imgfree
echo Boot parameters: ${install_params}
kernel ${mirror_address}${mirrordir}vmlinuz root=/dev/ram0 ${install_params}
imgfetch ${mirror_address}${mirrordir}initrd.img
imgfetch ${mirror_address}${mirrordir}filesystem.squashfs filesystem.squashfs
imgverify vmlinuz ${mirror_address}${mirrordir}vmlinuz.sig
imgverify initrd.img ${mirror_address}${mirrordir}initrd.img.sig
imgverify filesystem.squashfs ${mirror_address}${mirrordir}filesystem.squashfs.sig
imgstat
sleep 5
boot
```

When the systems boots up it drops to busybox because the squashfs couldn't be found. I tried to add the path but when I do it the file filesystem.squashfs doesnt appear withing the filesystem anymore.
```
imgfetch ${mirror_address}${mirrordir}filesystem.squashfs /live/filesystem.squashfs
```

Any suggestion? Has anyone a similar setup?
I suspect that if I was able to add filesystem.squashfs into /live directory it would work.

NiKiZe, 2020-11-25:
You are using pcbios and not EFI correct?
your issue is not imgfetch vs initrd, if you visit https://ipxe.org/cmd/initrd you will find that you end up on a different page.

instead of jumping all the way in one step, only change one thing at a time.
if you want to use imgverify, maybe you should do that after getting the squashfs to work?

So your issue is to get your dist to find and use the squashfs when you provide it this way.
- To do that you will to find the documentation, or try to dissect the initrd.
If you want other help then you should start with the screen that you see when it fails to find the squashfs.

kaleal, 2020-11-25:
Yes, I did one step at a time. I just jumped to imgverify just to be sure it works. But actually I am stuck trying to make squashfs work without imgverify.

Yes, my current issue is to make it to find the squashfs.
I will organize the ideas and come up with more information.

Thanks.

kaleal, 2020-11-26:
Not na ipxe issue.
Closing it.

closed 2020-11-26 by kaleal