ipxe/ipxe #173: EFI PCI configuration access not working on system with more PCI segments

pe-bo, 2020-12-02:
I tried iPXE for my own custom PCI Express card on a Dell PowerEdge R7515 server.
I found an interesting problem - the iPXE driver will only find the card in the first PCIe slot.
I examined this issue in more detail and discovered the following:

1) The server uses 4 PCI Roots and these roots share same PCI segment (0)
 
2) First slot belongs PCI Root with index 0

3) Other slots had a different PCI Root

4) The device binding fails in the `efipci_read` function (see the efi_pci.c file) because` root-> Pci.Read` returns an INVALID PARAMETER error.

5) The `efipci_root_open` function pairs the PCI Root with the PCI segment number and thus always finds the first PCI Root

I think this pairing method is bad. The UEFI specification in chapter **PCI Root Bridge I/O Overview** says:

> A PCI Host Bridge and PCI Root Bridge are different than a PCI Segment. A PCI Segment is a
collection of up to 256 PCI busses that share the same PCI Configuration Space. Depending on
the chipset, a single EFI_PCI_ROOT_BRIDGE_IO_PROTOCOL may abstract a portion of a PCI
Segment, or an entire PCI Segment. A PCI Host Bridge may produce one or more PCI Root
Bridges. When a PCI Host Bridge produces multiple PCI Root Bridges, it is possible to have
more than one PCI Segment.

I have no idea how reliably fix this issue.

thwalker3, 2020-12-08:
As noted on the mail list, I ran into this exact same problem on the R7515 (single socket version) and R6525 (dual socket version) of this platform back in May and submitted a patch (which we've been using in production on both platforms for ~6 months) but failed to follow up on it to get it accepted upstream.
I'm not at all a UEFI developer, so I can't guarantee that this is the correct approach, but it has worked for us.
https://lists.ipxe.org/pipermail/ipxe-devel/2020-May/007058.html

rustydb, 2020-12-16:
I've also been hitting this issue on HPE Proliants with Mellanox. I've applied @thwalker3 's fix and to my surprise I was able to read/scan all of my PCI devices.

Thanks!

pe-bo, 2020-12-16:
Finally I used much simpler solution. The EFI_PCI_IO_PROTOCOL itself has functions for PCI configuration space access, but requires an open instance of the protocol. This approach has been introduced recently in the 36dde9b commit. Of course, this also requires that the struct pci_device be always contained in the struct efi_pci_device.

mcb30, 2020-12-16:
@pe-bo That won't work, sorry.  A `struct pci_device` may be constructed for a PCI device that has no corresponding `struct efi_pci_device`.  For example, the `${pci/...}` settings mechanism that provides access to PCI configuration space for arbitrary PCI devices (including those with no iPXE driver) will do this.

mcb30, 2020-12-31:
@pe-bo @thwalker3 @rustydb Thank you for all the information.  I have pushed a patch derived from the patch that @thwalker3 posted on the mailing list as commit https://github.com/ipxe/ipxe/commit/988d2c13c

Please reopen if the issue is not fixed by this patch.

Thanks!

closed 2020-12-31 by mcb30

rustydb, 2021-01-05:
I've since pulled the latest iPXE and haven't had the issue anymore. Thank you!