ipxe/ipxe #174: Support for Data URIs

displague, 2020-12-03:
IPXE support for HTTP/HTTPs URLs is very extensible, but places additional dependencies on the networking infrastructure, requiring a host site.

In environments where the IPXe URL can be provided through automation, and a URL is the preferred or only method of entry, users may benefit from data-uri support.

In order to keep data-uri support simple, the supported encoding types could be restricted to plain text, utf-8, and base64.

NiKiZe, 2020-12-03:
What kind of content do you have in mind, and what is the maximum data size that would need to be supported?

displague, 2020-12-03:
Equinix Metal (formerly Packet) offers a IPXEScriptURL parameter during device creation.  I have a gist conversation at https://gist.github.com/displague/5282172449a83c7b83821f8f8333a072 where I have been exploring the capabilities of this feature.

The contents of the data-uri, in this case, would be additional configuration, for example:

```
dhcp net0

kernel https://mirror.openshift.com/pub/openshift-v4/x86_64/dependencies/rhcos/pre-release/latest-4.6/rhcos-live-kernel-x86_64 console=ttyS1,115200n8 initrd=rhcos-live-initramfs.x86_64.img coreos.live.rootfs_url=http://mirror.openshift.com/pub/openshift-v4/x86_64/dependencies/rhcos/pre-release/latest-4.6/rhcos-live-rootfs.x86_64.img coreos.inst.install_dev=/dev/sda rd.net.timeout.carrier=30 coreos.inst.platform_id=packet

initrd https://mirror.openshift.com/pub/openshift-v4/x86_64/dependencies/rhcos/pre-release/latest-4.6/rhcos-live-initramfs.x86_64.img

boot
```

I don't know how large configurations like this typically get, in the gist I had several commented out lines and variable declarations that I've removed in this example.

1K seems like too little, does 4K seem sufficient?  I can't imagine trying to shim binary images in as a data-uri.

Am I misrepresenting IPXE's role here? Is the existing HTTP support used for chaining additional configuration, or is it only used for pulling binary boot resources?

displague, 2020-12-03:
I'm also exploring this data-uri concept from the other side at https://github.com/tinkerbell/boots/issues/110.

detiber, 2020-12-03:
Related RFC for Data URLs: https://tools.ietf.org/html/rfc2397

It doesn't appear to require a maximum data size, but does specify that some applications that use URLs may impose length limits, so it seems like there is flexibility here to provide limits.

AWS user-data is limited to 16KB, a similar limitation here might make sense.

NiKiZe, 2021-07-09:
You can chain into a different script yes: https://ipxe.org/cmd/chain
That is how the first script is loaded and executed, just that the chain is internal.
If you have a script, it would not make much sense to embed another script as data-uri

I could see a usage such as `startnet.cmd` when using wimboot,
Or maybe chainloaded configurations for other kernels.
But not as an embeded script inside a script.

Size might be an issue, on different levels, not sure how IPXEScriptURL is done, but if it is in dhcp packet, then you have a somewhat to small limit there.