ipxe/ipxe #175: Request: Provide full featured builds

schlomo, 2020-12-07:
First of, I am really happy to see that iPXE is still going on!

Would it be possible to provide full featured builds at http://boot.ipxe.org/ instead of the default builds? Or in addition to the default builds?

I see two benefits for that:
1. easy place for users to download an iPXE version that can do (almost) everything that is there, helping to show case iPXE abilities and reducing the effort for users who don't need further customization
2. demonstrate which features can be combined in a build without troubles

I got to this point when setting up a universal network boot server for myself. I tried to find precompiled iPXE binaries that would be fairly recent and support HTTPS booting and framebuffer console. No luck.

I know that I can build my own but that is an extra step while I would like to provide easy to follow instructions for setting up such a universal boot server.

Since you in any case have an automated setup for building iPXE and uploading it to http://boot.ipxe.org/, I am asking you to add there also a "big" version of iPXE.

MaxPeal, 2021-01-07:
+1 for a "big" version of iPXE with https hosted on http://boot.ipxe.org/big or so

NiKiZe, 2021-01-07:
You have to define "full" because there is features that can't be enabled at the same time.

I agree that ipxe+Https should be great to have and maybe also an ipxe+https+IPv6 (mostly true for pcbios since it's not enabled by default)

schlomo, 2021-01-07:
What would be an "as much as possible" build? Maybe without some exotic options? I am sure that if you let us know about the mutually exclusive features then we can come to a meaningful (= useful for many users) compromise. Just ask.

For example, features that require a matching setup on other systems probably won't be so important in a "big build". Or features that are closely related to seldom or special hardware.

berlin4apk, 2021-06-03:
@NiKiZe  @mcb30 
Can you give us a hint to find all mutually exclusive features?

NiKiZe, 2021-06-03:
I'd say no, don't include more than you need.
Sure a "bigger" version with often used extra features, such as vlan and framebuffer, but we need that information.

NiKiZe, 2021-07-09:
Instead of arguing about a "big" version, how about that we at least start with a "more" build, where we request features that we know is useful?
* [https](https://ipxe.org/buildcfg/download_proto_https)
* [framebuffer](https://ipxe.org/buildcfg/console_framebuffer) along with console and png
* [param](https://ipxe.org/buildcfg/param_cmd)
* [vram](https://ipxe.org/cfg/vram) (pcbios only)

what else?

schlomo, 2021-07-11:
Yes, this is a great idea!

PicoMitchell, 2021-07-29:
I find these very useful:

For UEFI & BIOS:
poweroff Command: https://ipxe.org/buildcfg/poweroff_cmd
ping Command: https://ipxe.org/buildcfg/ping_cmd
cpuvendor and cpumodel Settings: https://github.com/ipxe/ipxe/issues/419 (undocumented)

And for BIOS only (since it doesn't work in UEFI last I checked):
memsize Setting: https://ipxe.org/buildcfg/memmap_settings

berlin4apk, 2021-07-30:
I find this useful, together with the already mentioned, options:

NSLOOKUP_CMD
PCI_CMD
REBOOT_CMD
TIME_CMD
VLAN_CMD
IMAGE_TRUST_CMD
IMAGE_ARCHIVE_CMD
DOWNLOAD_PROTO_FILE
NVO_CMD
DIGEST_CMD
IMAGE_MEM_CMD

closed 2026-09-09 by mcb30