ipxe/ipxe #176: HP 840 EFI ipx boot failure

maiee004, 2020-12-07:
Currently failing to EFI pxe boot a HP Z840.  I believe it to be a IPV6/EFI issue.  When booting legacy I am able to pxe boot no problem but when switching to UEFI, iPXE initialize devices.  I currently have these debug build flags `DEBUG=efi_driver,snpnet,pci,efi_pxe,efi_pci,ipv6`

[image]

I have tried commenting out this include from this MR. https://github.com/ipxe/ipxe/pull/84/files#diff-8d5c900320c89e96f4a0e5efbf4ae6450ee32eafb5189890897fb00830959bfbR26   But still run into the above error.  Any assistance would be greatly appreciated.

maiee004, 2020-12-07:
After further investigation.  The file were it was error on was  was `src/interface/efi/efi_pci.c`  (reading error message).  After a couple of permutations the commit the build fails on is https://github.com/ipxe/ipxe/commit/9b25f6e5cf517f426de80ede618398ef01e385f9

Any recommendations for getting around the error?

mcb30, 2020-12-28:
> After further investigation. The file were it was erroring on was was `src/interface/efi/efi_pci.c` (reading error message). After a couple of permutations the commit the build fails on is [9b25f6e](https://github.com/ipxe/ipxe/commit/9b25f6e5cf517f426de80ede618398ef01e385f9)

Are you saying that it works before commit [9b25f6e](https://github.com/ipxe/ipxe/commit/9b25f6e5cf517f426de80ede618398ef01e385f9) but fails on that commit or later?  That would be very odd, since that particular commit is itself restoring an earlier behaviour.

maiee004, 2021-01-06:
Yes,  the ipxe build before commit https://github.com/ipxe/ipxe/commit/9b25f6e5cf517f426de80ede618398ef01e385f9 succeeds. but any commit later it is failing on.

mcb30, 2021-02-15:
May plausibly be the same issue as #250

maiee004, 2021-02-22:
Once https://github.com/ipxe/ipxe/issues/250 is closed I can re-run tests on the Z840 to confirm.

mcnaugha, 2021-05-27:
We're seeing this issue with several HP models when in UEFI mode. No issues with legacy mode. We use Quest KACE SDA which utilises iPXE. They recently incorporated the current release and it's led to this for us.

NiKiZe, 2022-07-04:
Could we verify if this is still a current issue? 

I assume this was with ipxe.efi ?, Did you ever try with snponly.efi?

maiee004, 2022-09-19:
I was able to reproduce the error on a hp Z8 machine.  snponly.efi succeeds to boot but then both nic are unseen in ipxe, so I am unable to chain https boot after.  I booted the intel.ipxe and it succeeds but is missing one of the onboard nics.  On the z8 it would appear that ipxe doesnt like teh intel x722 nic.

maiee004, 2022-09-20:
This error is still persistent in the latest version of ipxe

stappersg, 2022-09-21:
> This error is still persistent in the latest version of ipxe

and

> appear that ipxe doesnt like the intel x722 nic

Using that _x722_ for further steps:
```text
stappers@myhost:~/src/mailinglists/ipxe
$ git grep x722
src/drivers/infiniband/linda.c: PCI_ROM ( 0x1077, 0x7220, "iba7220", "QLE7240/7280 HCA driver", 0 ),
src/drivers/net/intelxl.c:      PCI_ROM ( 0x8086, 0x37ce, "x722-kx", "X722 10GbE backplane", 0 ),
src/drivers/net/intelxl.c:      PCI_ROM ( 0x8086, 0x37cf, "x722-qsfp", "X722 10GbE QSFP+", 0 ),
src/drivers/net/intelxl.c:      PCI_ROM ( 0x8086, 0x37d0, "x722-sfp", "X722 10GbE SFP+", 0 ),
src/drivers/net/intelxl.c:      PCI_ROM ( 0x8086, 0x37d1, "x722-1gt", "X722 1GBASE-T", 0 ),
src/drivers/net/intelxl.c:      PCI_ROM ( 0x8086, 0x37d2, "x722-10gt", "X722 10GBASE-T", 0 ),
src/drivers/net/intelxl.c:      PCI_ROM ( 0x8086, 0x37d3, "x722-sfp-i", "X722 10GbE SFP+", 0 ),
src/drivers/net/intelxlvf.c:    PCI_ROM ( 0x8086, 0x37cd, "x722-vf", "X722 VF", 0 ),
src/drivers/net/intelxlvf.c:    PCI_ROM ( 0x8086, 0x37d9, "x722-vf-hv", "X722 VF (Hyper-V)", 0 ),
stappers@myhost:~/src/mailinglists/ipxe
$ 
```

The basenames without the `.c` are `intelxl` and `intelxlvf`.

Use those basenames in:

```text
make [platform]/[driver].[extension] DEBUG=intelxl,intelxlvf
```

`[platform]`, `[driver]` and `[extension]` inspired by https://ipxe.org/appnote/buildtargets
If only `[driver].[extension]` is good enough, say so. Express explicite which command is used for building the _iPXE executable_.

stappersg, 2022-09-22:
_OCR_, Optical Character Recognitation. of the above photo:
```text
iPXE initialising devices...INTELXL 0x654b40f0 PF 7 using port 3 queues [0x07ff-0x07ff]
MSI-X 0x654b4130 has 129 vectors for 0000:04:00.2
MSI-X 0x654b4130 table could not find BAR3
INTELXL 0x654b40f0 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
```

stappersg, 2022-09-23:
To me it looks that the original issue is solved. At least there is a complete different error.

This snippet is from https://github.com/ipxe/ipxe/issues/731#issuecomment-1237907800
```text
INTELXL 0x9ea1b910 PF 7 using port 3 queues [0x07ff-0x07ff]
MSI-X 0x9ea1b950 has 129 vectors for 0000:44:00.3
MSI-X 0x9ea1b950 table could not find BAR3
INTELXL 0x9ea1b910 could not enable MSI-X: No such file or directory (https://ipxe.org/2dcc408e)
```

Yes, that is the same https://ipxe.org/2dcc408e

maiee004, 2022-09-26:
Hmm.. Ok if its a different issue i would have to go back to the commit before https://github.com/ipxe/ipxe/commit/a202de385dcb9c15a1b2b3fd339abbe721ed5c86 and if my builds work I'm running into the same error as #731 
I'll try and confirm that today.  Thank you.

stappersg, 2022-10-14:
> I'll try and confirm that today.

How did that go?       (BTW: What is the _ipx_ in the current title of this issue?)

stappersg, 2022-10-26:
[image]

This issue started as a good issue. It was about a single aspect of _iPXE_. Something worth to be an _iPXE issue_.

The
> Yes, the ipxe build before commit https://github.com/ipxe/ipxe/commit/9b25f6e5cf517f426de80ede618398ef01e385f9 succeeds. but any commit later it is failing on.

was also a good thing.

Follow up question _Could we verify if this is still a current issue?_ got after about **ten weeks** a blunt _This error is still persistent in the latest version of ipxe_

Then there was _the original error is gone_ and this issue became _iPXE does NOT work for me, you must fix it._

This comment is for asking to close this issue.  At least to prevent another vague comment like
> On the z8 it would appear that ipxe doesnt like the intel x722 nic. This is still an issue.

expressing **you must fix my problem**

stappersg, 2022-11-14:
Merge request #801 closed #731. Those who can close this issue, _please do_.

closed 2022-11-14 by mcb30