ipxe/ipxe #178: Cannot Boot to the IPXE After Restart/Shutdown the System on Comet Lake H

ctony1, 2020-12-15:
Dear sir
We met the issue of supporting ipxe on CML platform with ipxe rom embedded in BIOS. The LAN chip used was Intel i219v with Vendor ID Device ID 80860d4d. The platform is H470.
When first power on, system could boot to ipxe server. But if we restart the system from Ubuntu 20.04, or shutdown the system from Ubuntu 20.04 and then power on the system, system could no longer be able to boot into ipxe server unless we remove the power supply and connect the power again. 

_**Restart failure information:**_ 
iPXE 1.20.1+ (gb6e2e)[0m -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP AoE ELF MBOOT NBI PXE bzImage Menu PXEXT
net0: f8:75:a4:21:94:be using i219v-10 on 0000:00:1f.6 (open)
[Link:down, TX:0 TXE:0 RX:0 RXE:0]
[Link Status: Down (http://ipxe.org/38086101)]
Waiting for link-up on net0................. Down (http://ipxe.org/38086101)
net0: f8:75:a4:21:94:be using i219v-10 on 0000:00:1f.6 (open)
[Link:down, TX:0 TXE:0 RX:0 RXE:0]
[Link Status: Down (http://ipxe.org/38086101)]
[TXE: 1 x "Operation not supported (http://ipxe.org/3c654003)"]
...

_**Shutdown from ubuntun and then power on the system failure information:**_
iPXE 1.20.1+ (gb6e2e)[0m -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP AoE ELF MBOOT NBI PXE bzImage Menu PXEXT
net0: f8:75:a4:21:94:be using i219v-10 on 0000:00:1f.6 (open)
[Link:up, TX:0 TXE:0 RX:0 RXE:0]
Configuring (net0 f8:75:a4:21:94:be)....... No configuration methods succeeded (http://ipxe.org/040ee119)
net0: f8:75:a4:21:94:be using i219v-10 on 0000:00:1f.6 (open)
[Link:up, TX:0 TXE:0 RX:0 RXE:0]
[TXE: 1 x Operation not supported (http://ipxe.org/3c654003)]
...

**_How to recover it when the problem happens_**
Platform reset or global reset or power off and then power on cannot recover it. It can only be recovered by removing the power adaptor and plug it again.

**PS: There is no problem on Corporation SKU Q470.(The LAN chip used was Intel i219v with Vendor ID Device ID 80860d4d. )**

Can you kindly give some comments on this issue? Thanks.

NiKiZe, 2020-12-15:
What link speed does the switch support and detect?

ctony1, 2020-12-15:
Hi NiKiZe
Can you kindly help tell how to check it. We tried different switch or connect the server directly with LAN cable, all failed. 
We have traces as attached. Can you help review them? Thanks.

mcb30, 2020-12-15:
@ctony1 There is no `i219v-10` PCI ID recorded in the iPXE codebase.  Please share the modifications that you have made to iPXE, since we cannot help you debug unknown code.

ctony1, 2020-12-15:
Hi Michael

ctony1, 2020-12-15:
> @ctony1 There is no `i219v-10` PCI ID recorded in the iPXE codebase. Please share the modifications that you have made to iPXE, since we cannot help you debug unknown code.

HI Michael
Sorry I closed the case by mistake. 
I added the PCI ID manually to intel.c. 
	PCI_ROM ( 0x8086, 0x15e3, "i219lm-5", "I219-LM (5)", INTEL_I219 ),
	**PCI_ROM ( 0x8086, 0x0d4c, "i219lm-10", "I219-LM (10)", INTEL_I219 ),
	PCI_ROM ( 0x8086, 0x0d4d, "i219v-10", "I219-V (10)", INTEL_I219 ),**
	PCI_ROM ( 0x8086, 0x1f41, "i354", "I354", INTEL_NO_ASDE ),
	PCI_ROM ( 0x8086, 0x294c, "82566dc-2", "82566DC-2", 0 ),
	PCI_ROM ( 0x8086, 0x2e6e, "cemedia", "CE Media Processor", 0 ),
};

Also change the general.h to support PXE image.
#define	IMAGE_NBI		/* NBI image support */
#define	IMAGE_ELF		/* ELF image support */
//#define	IMAGE_MULTIBOOT		/* MultiBoot image support */
#define	IMAGE_PXE		/* PXE image support */
#define	IMAGE_SCRIPT		/* iPXE script image support */
#define	IMAGE_BZIMAGE		/* Linux bzImage image support */
//#define	IMAGE_COMBOOT		/* SYSLINUX COMBOOT image support */
//#define	IMAGE_EFI		/* EFI image support */
//#define	IMAGE_SDI		/* SDI image support */
//#define	IMAGE_PNM		/* PNM image support */
//#define	IMAGE_PNG		/* PNG image support */
#define	IMAGE_DER		/* DER image support */
#define	IMAGE_PEM		/* PEM image support */

Please refer to the attachment about the code change.
[code.zip](https://github.com/ipxe/ipxe/files/5696854/code.zip)

mcb30, 2020-12-15:
@ctony1 Thank you for clarifying.  The i219 has an extremely broken reset mechanism: see commit https://github.com/ipxe/ipxe/commit/546dd51de for details.

It is very possible that the i219v-10 has a subtle variation on this brokenness that would require further elaboration of the existing workaround.

ctony1, 2020-12-16:
@mcb30 Hi Michael, thank you for sharing the information. We use the latest code. The native code has already implemented the workaround you share. 
	_**/* Set undocumented bit in FEXTNVM11 to work around an errata
	 * in i219 devices that will otherwise cause a complete
	 * datapath hang at the next device reset.
	 */**_
Is there anything else we can try?

mcb30, 2020-12-16:
@ctony1 Based on prior experience with the i219, this is the kind of problem that would likely take me a couple of days of debugging with direct access to hardware that reproduces the problem.  It's not something I can viably remotely guide you through fixing yourself, sorry.

daberkow, 2021-02-02:
Hi,

I got a Dell Precision 7550, it wanted in on the fun. It has an Intel i219lm-11... PCI 0x8086, 0x0d4c... I was able to get this card to work by embedding a script. It would be nice to get this fixed, but I understand how hard this remote debugging can be. I have tried marking it as INTEL_I219, INTEL_NO_PHY_RST, if you have any other things I can try let me know. My C++ skills are not super fantastic, but I am happy to try things while I have a system here from work.

`#!ipxe

echo Opening Connection
ifopen net0
sleep 2

:retry_dhcp
echo Attempting DHCP
dhcp && isset ${filename} || goto retry_dhcp
echo Booting from ${filename}
boot ${filename}
`

mcb30, 2021-02-02:
@daberkow From what you've described, your issue sounds more like a DHCP server configuration problem than a driver problem.  To confirm, could you share a screenshot of a boot (using your embedded script) that shows `Attempting DHCP` appearing more than once?

daberkow, 2021-02-02:
@mcb30 It looks like that, but I have run iPXE on this network without issue for 7 years or so. I have an old Precision M6800 that boots fine with BIOS or EFI boot files. I have booted hundreds of servers on this network without issue on the stock image. I am getting some of https://ipxe.org/err/3c654083 and https://ipxe.org/err/380f60 . (Sidenote thanks for all you do here)

mcb30, 2021-02-02:
> @mcb30 It looks like that, but I have run iPXE on this network without issue for 7 years or so. I have an old Precision M6800 that boots fine with BIOS or EFI boot files. I have booted hundreds of [...]

Thank you for providing the error URLs.  The https://ipxe.org/err/380f60 is harmless (indicating just some broadcast UDP packets on your network), but the https://ipxe.org/err/3c654083 points directly to an unexpected interrupt happening in `intel_poll()` on line https://github.com/ipxe/ipxe/blob/5bdb75c9d/src/drivers/net/intel.c#L886

Could you build with `DEBUG=intel` and report the `INTEL 0x.... unexpected ICR 0x....` debug message?

Thanks.

daberkow, 2021-02-02:
``INTEL 0x405166f0 unexpected ICR 00000106`` I have photos of it working, and of it not working. It seems random when it is open or up enough to get a proxydhcp setting, vs when it doesnt.

mcb30, 2021-02-15:
@daberkow can you suggest a cheap (sub-$100) machine I can source that would reproduce this problem?  Alternatively, are you able to offer IP-KVM access for a few days to a suitable machine you have available?

daberkow, 2021-02-15:
@mcb30 Sadly I had to return the laptop to someone else to do some work on. It seems this chipset is part of the Z490 series of intel motherboards, and some 10th gen Intel Dell systems like Dell OptiPlex 7080 according to [intels forum](https://community.intel.com/t5/Ethernet-Products/Intel-R-Ethernet-Connection-11-I219-V-Unable-to-install-intel/td-p/1187251/page/5). I was running it on a Precision 7550. I can not find a PC let alone just a motherboard under $300 right now with this chipset.

I think we maybe stuck for now unless later I get one and can get you access or something comes up with its chipset. Thanks for the help, it seems to be an Intel 10th/11th gen ethernet chipset that is used more starting around August 2020. Someone told me it was a 2.5gbps interface, but I can not find any information to back that and Dells website seems to say its 1gbps, when it was here I had it on a 1gbps switch.

I currently have a working solution with retrying DHCP with an embedded script until it returns a bootfile. Appreciate all the work you do here.

a31amit, 2021-04-15:
I believe I also have a similar issue but the model is i219v-2. I tried this machine with undionly, snponly, ipxe.efi both in bios and UEFI, but all of them have the same problem. other machines work fine in the same network. 

[image]

mcb30, 2021-04-15:
Again: this issue is probably stuck unless someone is able to provide remote access to an affected system (or ship an affected system to me).

On the subject of remote access: I have a project in progress at https://github.com/ipxe/keyflogger to make it possible to use any Linux machine as a remote IP-KVM, which should help with remote debugging in future.  The initial run of 10 prototype PCBs is currently being assembled.