ipxe/ipxe #188: Chainloaded DHCP options for UEFI

mcb30, 2020-12-28:
## Description

As a user, I want UEFI chainloading to work similarly to BIOS chainloading, to minimise my surprise.  In particular, I would like options from the original vendor PXE DHCP request to be available within UEFI iPXE as they are in BIOS iPXE.

## Acceptance criteria

- [ ] Ability to chainload `snponly.efi`, to break immediately into the iPXE shell via Ctrl-B, and to see via `route` that iPXE already has an IPv4 address configuration
- [ ] Ability to chainload `snponly.efi`, to break immediately into the iPXE shell via Ctrl-B, and to see `show filename` show the path to `snponly.efi` itself.

etnestad, 2021-02-11:
> * [ ]  Ability to chainload `snponly.efi`, to break immediately into the iPXE shell via Ctrl-B, and to see `show filename` show the path to `snponly.efi` itself.

What do you mean by path in this context? tftp://<server-ip>/snponly.efi ?

mcb30, 2021-02-11:
> > * [ ]   Ability to chainload `snponly.efi`, to break immediately into the iPXE shell via Ctrl-B, and to see `show filename` show the path to `snponly.efi` itself.
> 
> What do you mean by path in this context? tftp://server-ip/snponly.efi ?

It would be whatever was present in the boot filename as provided by DHCP (i.e. matching the behaviour that already exists for BIOS builds).

etnestad, 2021-02-11:
> It would be whatever was present in the boot filename as provided by DHCP (i.e. matching the behaviour that already exists for BIOS builds).

Ah, got it!

etnestad, 2021-02-13:
A couple of questions:

For now I have just written some test code in efi_init.c to invoke pxe_base_code_protocol to get the cached DHCP packet. Where is the "right" location for this code?

What mechanism should I use to configure the network interface with the information from the cached DHCP packet?

mcb30, 2021-02-17:
@etnestad Thanks.  It required some slightly complex restructuring to allow code to be shared between BIOS and UEFI, so I've implemented it myself.

PR #265 (2021-02-17): Add UEFI support for cached DHCPACK
Fixes: #188
closed 2021-02-17 by mcb30