ipxe/ipxe #203: Do not hijack BOOTP filename to provide iPXE URL

vincentbernat, 2021-01-12:
Hey!

Currently, to load iPXE payload, we can provide an URL as a BOOTP filename. When chainloading, this requires the DHCP server to be able to give a different value depending on the user class. Not all DHCP servers are able to do that. I would propose to implement an additional option that would be preferred to BOOTP filename for the same purpose. If this option is present, then its value is used and BOOTP filename is ignored.

Is there any opposition for this approach?

Thanks.

NiKiZe, 2021-01-12:
You should switch to a DHCP server that allows it.
There is also the possibility to embedd a script into iPXE and other methods. See ipxe.org "breaking the infinitive loop"

vincentbernat, 2021-01-12:
The embedded script is OK for me. Thanks.

closed 2021-01-12 by vincentbernat