ipxe/ipxe #210: drivers/bus/isa.c doesn't compile bin-x86_64-efi/ipxe.efi on NixOS's gcc

mmlb, 2021-01-14:
I'm trying to build ipxe efi for x86_64 and came across this build failure. I first came across this with `gcc version 9.3.0 (GCC)` and can reproduce with newer GCCs also. I haven't tried going back to see when it broke but can do so if needed.

I can reproduce as follows:

```
[manny@dellnix:~/cloned/ipxe/ipxe]$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

[manny@dellnix:~/cloned/ipxe/ipxe]$ git describe --tags
v1.21.1-6-ga5fb4187

[manny@dellnix:~/cloned/ipxe/ipxe]$ cat ../shell.nix | tee shell.nix
let _pkgs = import <nixpkgs> { };
in { pkgs ? import (_pkgs.fetchFromGitHub {
  owner = "NixOS";
  repo = "nixpkgs";
  #branch@date: nixpkgs@2020-11-24
  rev = "6625284c397b44bc9518a5a1567c1b5aae455c08";
  sha256 = "1w0czzv53sg35gp7sr506facbmzd33jm34p6cg23fb9kz5rf5c89";
}) { } }:

with pkgs;

mkShell {
  buildInputs = [
    gcc
  ];
}

[manny@dellnix:~/cloned/ipxe/ipxe]$ nix-shell
bash: /home/manny/.fzf/shell/key-bindings.bash: No such file or directory

[nix-shell:~/cloned/ipxe/ipxe]$ cd src

[nix-shell:~/cloned/ipxe/ipxe/src]$ make bin-x86_64-efi/ipxe.efi -j$(nproc)
make: /bin/echo: No such file or directory
make: /bin/echo: No such file or directory
rm -f bin-x86_64-efi/*.*  bin-x86_64-efi/.certificate.* bin-x86_64-efi/.certificates.* bin-x86_64-efi/.private_key.* bin-x86_64-efi/errors	bin-x86_64-efi/NIC	./util/zbin ./util/elf2efi32 ./util/elf2efi64 ./util/efirom ./util/efifatbin ./util/iccfix ./util/einfo TAGS bin-x86_64-efi/symtab
  [DEPS] arch/x86/transitions/librm.S
  [DEPS] arch/x86/transitions/libkir.S
  [PARSEROM]
  [DEPS] arch/x86/transitions/libpm.S
  [DEPS] arch/x86/prefix/usbdisk.S
  [DEPS] arch/x86/transitions/liba20.S
...
  [BUILD] bin-x86_64-efi/mca.o
  [BUILD] bin-x86_64-efi/pcibackup.o
  [BUILD] bin-x86_64-efi/pci.o
  [BUILD] bin-x86_64-efi/pciea.o
drivers/bus/isa.c: In function ‘isabus_probe’:
drivers/bus/isa.c:52:4: error: array subscript [-2147483648, -1] is outside array bounds of ‘isa_probe_addr_t[0]’ {aka ‘short unsigned int[0]’} [-Werror=array-bounds]
   52 |    *( isa_extra_probe_addrs + (ioidx) + ISA_EXTRA_PROBE_ADDR_COUNT ) )
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/bus/isa.c:113:18: note: in expansion of macro ‘ISA_IOADDR’
  113 |    isa->ioaddr = ISA_IOADDR ( driver, ioidx );
      |                  ^~~~~~~~~~
drivers/bus/isa.c:34:25: note: while referencing ‘isa_extra_probe_addrs’
   34 | static isa_probe_addr_t isa_extra_probe_addrs[] = {
      |                         ^~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:943: bin-x86_64-efi/isa.o] Error 1
make: *** Waiting for unfinished jobs....

[nix-shell:~/cloned/ipxe/ipxe/src]$ 
```

This requires having `nix` installed, alternatively with docker:
```
[manny@dellnix:~/cloned/ipxe/ipxe]$ docker run --rm -ti -v $PWD:/ipxe -w /ipxe -e NIX_PATH=nixpkgs=https://github.com/NixOS/nixpkgs/archive/6625284c397b44bc9518a5a1567c1b5aae455c08.tar.gz nixos/nix:2.3.6

f732121730e2:/ipxe# nix-shell -p gcc
unpacking 'https://github.com/NixOS/nixpkgs/archive/6625284c397b44bc9518a5a1567c1b5aae455c08.tar.gz'...
these paths will be fetched (54.75 MiB download, 243.85 MiB unpacked):
  /nix/store/1n34g5xxfy395xzh8d5xk6dq2c2rnx2b-glibc-2.32-dev
  /nix/store/20i4573zsrxagw7mcn5adyd5m4f6a11l-ncurses-6.2
  /nix/store/31hnbrb3b3j6wlp43shjwa688mfzif2r-bzip2-1.0.6.0.1
  /nix/store/53nwh281zm213kg505mrp3g5qwdsnjbs-patch-2.7.6
  /nix/store/58q0ygc01jba5cf5zghc0dlhx187mskk-gcc-9.3.0
...
copying path '/nix/store/79rshjqnibwcx468m4i31wnmasaj6nf4-binutils-2.31.1' from 'https://cache.nixos.org'...
copying path '/nix/store/58q0ygc01jba5cf5zghc0dlhx187mskk-gcc-9.3.0' from 'https://cache.nixos.org'...
copying path '/nix/store/rqhvcy5zmlypr9qfa43npsyxsxjg7jxa-binutils-wrapper-2.31.1' from 'https://cache.nixos.org'...
copying path '/nix/store/n3r351j7vfr2bv95fp6miz3l8741jdb4-gcc-wrapper-9.3.0' from 'https://cache.nixos.org'...
copying path '/nix/store/frh7ir9rcv19b3ym33ck64s813yr3xrr-stdenv-linux' from 'https://cache.nixos.org'...

[nix-shell:/ipxe]# make -C src/ bin-x86_64-efi/ipxe.efi
make: Entering directory '/ipxe/src'
/bin/sh: git: not found
/bin/sh: perl: not found
rm -f bin-x86_64-efi/*.*  bin-x86_64-efi/.certificate.* bin-x86_64-efi/.certificates.* bin-x86_64-efi/.private_key.* bin-x86_64-efi/errors	bin-x86_64-efi/NIC	./util/zbin ./util/elf2efi32 ./util/elf2efi64 ./util/efirom ./util/efifatbin ./util/iccfix ./util/einfo TAGS bin-x86_64-efi/symtab
  [PARSEROM]
/bin/sh: perl: not found
make: *** Deleting file 'bin-x86_64-efi/.rom.defs'
  [DEPS] arch/x86/transitions/librm.S
  [DEPS] arch/x86/transitions/libpm.S
  [DEPS] arch/x86/transitions/libkir.S
  [DEPS] arch/x86/transitions/liba20.S
  [DEPS] arch/x86/prefix/usbdisk.S
  [DEPS] arch/x86/prefix/unlzma16.S
...
  [BUILD] bin-x86_64-efi/cdc.o
  [BUILD] bin-x86_64-efi/eisa.o
  [BUILD] bin-x86_64-efi/isa.o
drivers/bus/isa.c: In function 'isabus_probe':
drivers/bus/isa.c:52:4: error: array subscript [-2147483648, -1] is outside array bounds of 'isa_probe_addr_t[0]' {aka 'short unsigned int[0]'} [-Werror=array-bounds]
   52 |    *( isa_extra_probe_addrs + (ioidx) + ISA_EXTRA_PROBE_ADDR_COUNT ) )
      |    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/bus/isa.c:113:18: note: in expansion of macro 'ISA_IOADDR'
  113 |    isa->ioaddr = ISA_IOADDR ( driver, ioidx );
      |                  ^~~~~~~~~~
drivers/bus/isa.c:34:25: note: while referencing 'isa_extra_probe_addrs'
   34 | static isa_probe_addr_t isa_extra_probe_addrs[] = {
      |                         ^~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile.housekeeping:943: bin-x86_64-efi/isa.o] Error 1
make: Leaving directory '/ipxe/src'
```

mmlb, 2021-01-14:
I tried tracking this down, but by C chops aren't up to snuff any more. I did manage to get a passing build with a teeny hack that I don't think is the correct fix but may help someone else:

```diff
[nix-shell:/ipxe]# git diff | cat
diff --git a/src/drivers/bus/isa.c b/src/drivers/bus/isa.c
index 6b360a44..083cf26d 100644
--- a/src/drivers/bus/isa.c
+++ b/src/drivers/bus/isa.c
@@ -95,7 +95,7 @@ static void isa_remove ( struct isa_device *isa ) {
 static int isabus_probe ( struct root_device *rootdev ) {
 	struct isa_device *isa = NULL;
 	struct isa_driver *driver;
-	int ioidx;
+	long int ioidx;
 	int rc;
 
 	for_each_table_entry ( driver, ISA_DRIVERS ) {
```
```bash

[nix-shell:/ipxe]# make -C src/ bin-x86_64-efi/ipxe.efi
make: Entering directory '/ipxe/src'
/bin/sh: perl: not found
  [PARSEROM]
/bin/sh: perl: not found
make: *** Deleting file 'bin-x86_64-efi/.rom.defs'
  [DEPS] drivers/bus/isa.c
/bin/sh: perl: not found
  [PARSEROM]
/bin/sh: perl: not found
make: *** Deleting file 'bin-x86_64-efi/.rom.defs'
  [BUILD] bin-x86_64-efi/isa.o
  [BUILD] bin-x86_64-efi/isa_ids.o
...
  [BUILD] bin-x86_64-efi/librm.o
  [HOSTCC] util/elf2efi64
  [VERSION] bin-x86_64-efi/version.ipxe.efi.o
  [AR] bin-x86_64-efi/blib.a
ar: creating bin-x86_64-efi/blib.a
  [LD] bin-x86_64-efi/ipxe.efi.tmp
/bin/sh: perl: not found
/nix/store/79rshjqnibwcx468m4i31wnmasaj6nf4-binutils-2.31.1/bin/ld:--defsym:0: syntax error
make: *** [Makefile.housekeeping:1189: bin-x86_64-efi/ipxe.efi.tmp] Error 1
rm bin-x86_64-efi/version.ipxe.efi.o
make: Leaving directory '/ipxe/src'

[nix-shell:/ipxe]# 
````

mcb30, 2021-01-15:
@mmlb Thanks for debugging.  I was able to reproduce the problem using your container instructions.

I tried various modifications to avoid the spurious compiler warning, but couldn't find anything sensible that didn't instead trigger a `-Wtype-limits` warning.  I've applied your patch as commit https://github.com/ipxe/ipxe/commit/8e3826aa1 after verifying that it does seem to result in a clean build with the NixOS version of gcc.

Thanks!

closed 2021-01-15 by mcb30

mmlb, 2021-01-15:
No problem!

baloo, 2021-10-02:
That fix did not work for me, I ended up writing this:
```
diff --git a/src/drivers/bus/isa.c b/src/drivers/bus/isa.c
index 94d4ce99..5d924c09 100644
--- a/src/drivers/bus/isa.c
+++ b/src/drivers/bus/isa.c
@@ -110,7 +110,10 @@ static int isabus_probe ( struct root_device *rootdev ) {
                        }
                        memset ( isa, 0, sizeof ( *isa ) );
                        isa->driver = driver;
+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Warray-bounds"
                        isa->ioaddr = ISA_IOADDR ( driver, ioidx );
+#pragma GCC diagnostic pop

                        /* Add to device hierarchy */
                        snprintf ( isa->dev.name, sizeof ( isa->dev.name ),
```

Not sure this is the correct fix, but hopefully that would help someone else.

mcb30, 2021-10-04:
> ```
> diff --git a/src/drivers/bus/isa.c b/src/drivers/bus/isa.c
> index 94d4ce99..5d924c09 100644
> --- a/src/drivers/bus/isa.c
> +++ b/src/drivers/bus/isa.c
> @@ -110,7 +110,10 @@ static int isabus_probe ( [...]

Thanks for the follow-up.  I try hard not to pollute the codebase with preprocessor directives because it makes the code hard to read, so this patch unfortunately can't be applied as-is.  I'm happy to accept a cleaner patch.

baloo, 2021-10-04:
Yeah, I didn't submit a PR because I don't think this is an acceptable solution either. I mostly pushed it here as I hit it before and I lost my "fix" since.

I tried to understand isa addressing but I really don't have any experience with it nor any hardware to test it. Not sure I can get a better fix.

mmlb, 2021-10-04:
I've seen this come up again when building from different versions of Nixpkgs/gcc but haven't been able to track it down. I've also been told that this happened with Arch's GCC10. Either way, the proper fix remains out of my level of expertise with C I think.

baloo, 2021-10-04:
Interestingly, error is ... not happening anymore with a newer version of GCC.

I'm not entirely sure why.