ipxe/ipxe #212: iPXE script file does not execute

upl1nk-v01d, 2021-01-17:
Hello there!
I have an issue executing my boot.ipxe file. Script file permissions are 755. File content is simple 'hello world':

#!ipxe
:loop
echo hello world
goto loop

Script file is stored on USB stick drive within /boot/grub/ipxe directory.
I don't know why this doesn't execute. Please help!

NiKiZe, 2021-01-17:
Try imgfetch path/to/file
When that succeeds do imgstat and make sure that it says script, if not your #!ipxe is not the first bytes of the file.

The only way to have scripts on usb is in efi, and relies on the firmware support for the filesystem on the device, which in efi spec is only fat.

Let us know what you use to try and run the script, but my guess is that you want to do something that you have not yet described which has better solutions.

upl1nk-v01d, 2021-01-17:
Tried with imgfetch /boot/grub/ipxe/boot.ipxe, but console throwed back: Could not start download. Operation is not supported.

NiKiZe, 2021-01-17:
Why are you using a file on disk in the first place? Instead of embedded or from the network?

Did you read the documentation for using files from disk in efi, and are you sure you are using efi and not pcbios?

upl1nk-v01d, 2021-01-17:
First of all, thank you for replying to my dummy problem :D 
My desire was to boot ipxe from usb drive for experimenting and simplicity sake. 
My struggle was to automaticly execute boot.ipxe script file.
What do you mean by 'embedded'? I was using pcbios.
My catch was to boot in ipxe via usb drive on laptop to automaticly connect to my external OS (win, linux etc) server.

NiKiZe, 2021-01-17:
As I wrote, file on local disk is not an option in pcbios 

See https://ipxe.org/embed

upl1nk-v01d, 2021-01-17:
Eureka!! I found solution. :) 
I appended this line: initrd16 /boot/grub/ipxe/boot.ipxe

menuentry 'ipxe' {
    set root='hd0,msdos1' 
    linux16 /boot/grub/ipxe/ipxe.lkrn
    initrd16 /boot/grub/ipxe/boot.ipxe
}

...and the console throws out bunch of 'hello world'.
Thank you for trying to help me. :)

NiKiZe, 2021-01-17:
Well, that is not a ipxe script, but happy you got your scenario working. This is not a "solution" that I would recommend to anyone.

Since this is not an issue with ipxe itself, it would have been better to post it under discussion rather than issue.

closed 2021-01-17 by NiKiZe