ipxe/ipxe #213: v1.21.1 EFI local disk boot fail on vmware machines.

skippie81, 2021-01-20:
Hi

We have a setup where we have an iPXE build snponly.efi and undionly.kpxe  that is used to boot systems over the network by loading iPXE via pxe.

The build has a default embedded #!ipxe script that basically chain loads a ipxe script that it gets over http. This one is dynamically created using the mac address of the system that requests the the ipxe script.

There are some options that loaded script could boot for example: provisioning systems, testing,  ... 
But the default would most of the time be just to exit (boot form disk). The embedded script will continue on exit of the dynamically loaded script.

the relevant part of the embedded script:

```
:menuexit
echo ========================================
echo menu has exited, booting from local disk
echo
echo platform : ${platform}
echo ========================================
sleep 2
goto diskboot

:diskboot
iseq ${platform} efi && goto efidiskboot || goto biosdiskboot

:biosdiskboot
sanboot --no-describe --drive 0x80 || goto menu

:efidiskboot
exit 1
```

This worked perfectly with iPXE versions up to v1.20.1.

As we build again from iPXE version v1.21.1 this still works on most of our systems (various hardware machines).
But we also have a small amount of vmware virutal servers setup in EFI mode and booting over the network, mostly to test changes on this setup. For those virtual machines this worked until v1.20.1 but as I build from v1.21.1 the machine displays the entry that we are  going to boot from local disk, but after the 2seconds the system virtual server just powers down.
So somehow the exit 1 causes the system to power-down a vmware virtual machine. all other hardware  the exit 1 still causes EFI to hop to the next boot target: a bootloader on a local disk.

information about the virtual server:
ESXi 6.7 and later (VM version 14)

omniproc, 2022-04-13:
likely related to https://github.com/ipxe/ipxe/issues/349 ?

NiKiZe, 2022-04-13:
> likely related to #349 ?

I'm reading this as being an issue with disk boot in efi mode, while #349 is an "SNP issue"

NiKiZe, 2022-04-13:
> Dunno @NiKiZe . What's strange is that both issues seem to affect VMware systems only (the one linked the ESXi hypervisor and the one mentioned here vSphere VMs) and both have been introduced with 1.21.1. Could be a coincidence of course but since I haven't seen anyone pointing to the actual source thats causing both issues I just wanted to highlight that this at least sounds like it could be related.

Basically this issue is about the efi firmware implementation used in vmware vms not trying the next boot device when a efi programs returns with an error code, which most (but not all) real hardware does.
This seems to have worked previously, but something changed that made vmware not behave as wanted in this particular case.

#349 is about running ESX server (the hypervisor) on real hardware, specifically HP firmware killing itself.