ipxe/ipxe #233: iPXE not calling UNDI.stop()

bipedstites, 2021-01-28:
I am finding that iPXE 1.20.1 is leaving my UEFI UNDI driver in a running state which causes a RSOD.  The solution is for iPXE to call UNDI.stop() when it is done with its operations.  The following is an iPXE scenario that I run into:

1.   UEFI UNDI driver is loaded

2.   BIOS SNP calls undi.start() <- from this point driver is using SnpUndi32CallbackDelay as a default Delay function

3.   Downloading snponly.efi

4.   BIOS is loading snponly.efi

5.   iPXE stops original SNP which calls undi.stop() <- from this point UNDI is stopped and drivers is using native Delay function.

6.   iPXE snponly.efi calls undi.start() <- from this point driver is using iPXE delay function (nii_delay?) as a default Delay function

7.   Waiting for iPXE booting timeout

8.   BIOS is unloading snponly.efi <- iPXE should call undi.stop() but it doesn't. 

9.   Coming back to Boot Manager <- iPXE delay function is still set as default delay function even though snponly.efi is already unloaded. That's because undi.stop() was not called.

10. user tries to perform any action which requires Delay function <- RSOD occurs, UEFI driver tried to call iPXE delay function.

Conclusion:  iPXE needs to call undi.stop() during unloading and leave UNDI driver in stopped state.

mcb30, 2021-01-28:
There is no `undi.stop()` function pointer: all UNDI calls go via a single entry point as defined by the UEFI specification.  I assume that you mean an UNDI call with `PXE_OPCODE_STOP` as the opcode.

iPXE will already issue `PXE_OPCODE_STOP` when exiting back to the UEFI shell.  Tested just now using OVMF with the Intel E1000 UEFI drivers by booting into `snponly.efi`, using the `dhcp` command to open the interface and obtain an IP address, then the `exit` command to return (successfully) to the UEFI boot menu.

Here is the debug transcript from `DEBUG=efi_driver,nii:3` after issuing the `exit` command while the interface was still open:

```
EFIDRV disconnecting our drivers
EFIDRV PciRoot(0x0)/Pci(0x3,0x0)/MAC(525400123456,0x0) DRIVER_STOP child PciRoot(0x0)/Pci(0x3,0x0)/MAC(525400123456,0x0)/MAC(525400123456,0x1)                                                                                                                                
NII NII-0000:00:03.0 issuing 07:0000 ifnum 0
NII NII-0000:00:03.0 issuing 02:0000 ifnum 0                                                                                           
EFIDRV PciRoot(0x0)/Pci(0x3,0x0)/MAC(525400123456,0x0) DRIVER_STOP
EFIDRV disconnecting our drivers
EFIDRV reconnecting old drivers
```

As the transcript shows, iPXE issues opcode 07 (`PXE_OPCODE_SHUTDOWN`) followed by opcode 02 (`PXE_OPCODE_STOP`).

So, iPXE is leaving the driver in the stopped state, as required.  It is possible that your HP(?) BIOS has a bug that is preventing drivers from being disconnected.  Check the output when `snponly.efi` is built with `DEBUG=efi_driver,nii:3` (which you will probably need to capture via serial port).

mcb30, 2021-01-31:
Marking as closed since iPXE does not exhibit the issue as described.  Please comment and reopen if you still have a problem.

closed 2021-01-31 by mcb30

yayunliu90, 2021-02-12:
BIOS is unloading snponly.efi <- iPXE should call undi.stop() but it doesn't.
Coming back to Boot Manager<- iPXE delay function is still set as default delay function even though snponly.efi is already unloaded. That's because undi.stop() was not called.
User tries to perform any action which requires Delay function <- RSOD occurs, UEFI driver tried to call iPXE delay function.

Does iPXE disagree that undi.stop() should called by their iPXE driver? Calling undi.stop() will inform all drivers that might be hooked into iPXE to not use calls to iPXE anymore. That is the problem as diagnosed by Ivy. undi.stop() not called, but Boot Manager still has the iPXE delay function as the default.. so RSOD occurs when the iPXE delay function is called at anytime after snponly.efi is unloaded.
What is iPXE recommendation if they think BIOS is responsible?

yayunliu90, 2021-02-12:
[image]
The snapshot might be helpful to give you insights what i am talking...

mcb30, 2021-02-12:
@yayunliu90 iPXE **does** call `PXE_OPCODE_STOP`: see the debug transcript that I sent above.

If you disagree with this statement, please provide instructions that actually reproduce the alleged problem.

mcb30, 2021-02-12:
> [image]

Specifically: iPXE issues the call to `PXE_OPCODE_STOP` **before** it even returns to the BIOS.  Step 8 in the above sequence is incorrect: iPXE has already cleaned up **before** exiting back to the BIOS.

mcb30, 2021-02-12:
@yayunliu90 It might be helpful to conduct this conversation directly, rather than relaying information between parties.  You can CC me in to your email thread as `mcb30@ipxe.org`.

yayunliu90, 2021-02-12:
@mcb30  Thanks! I will add you to the existing thread. :)