ipxe/ipxe #239: Enable IPv6 by default

zajdee, 2021-01-31:
Hi,
thank you for your effort on developing iPXE. It's proven to be the most useful network boot tool for me since years ago. Much appreciated.

Now there's one thing that bothers me a lot. It's 2021 and there's still
```
//#define NET_PROTO_IPV6	/* IPv6 protocol */
```
in [general.h](https://github.com/ipxe/ipxe/blob/ba20ba42731b8b03a5db2f81f876fccd56257b0f/src/config/general.h#L38).

Isn't it time already to enable IPv6 permanently?

**As a user, I do not want to recompile new iPXE builds myself just to enable IPv6.**

Thanks again for all the work and time spent on the development of iPXE.

NiKiZe, 2021-01-31:
If you did a quick search you should have found #84
It already is enabled by default in `efi` builds, and will not be default in `pcbios`.

It is 2021, are you still using legacy pcbios? ;)

zajdee, 2021-01-31:
I've done a quick search among open tickets. ;-)
Yes, I am using legacy BIOS boots in old environments (chainloading the iPXE undionly images there). I guess I'm then left with enabling the option myself there.
What I was not aware of is that the option is enabled in the UEFI builds. That one is on me.

thanks.

zajdee, 2021-01-31:
Let's close this issue then.

closed 2021-01-31 by zajdee