ipxe/ipxe #246: Wrong  410de18f TLS cipher mismatch

wernerb, 2021-02-02:
I am trying for iPXE to accept https://fedorapeople.org.

It only accepts TLS 1.2
Nmap scan:
```
$ nmap --script ssl-enum-ciphers.nse -p 443 fedorapeople.org
Starting Nmap 7.91 ( https://nmap.org ) at 2021-02-02 22:00 CET
Nmap scan report for fedorapeople.org (152.19.134.199)
Host is up (0.096s latency).
Other addresses for fedorapeople.org (not scanned): 2610:28:3090:3001:5054:ff:fea7:9474
rDNS record for 152.19.134.199: vm20.fedora.ibiblio.org

PORT    STATE SERVICE
443/tcp open  https
| ssl-enum-ciphers: 
|   TLSv1.2: 
|     ciphers: 
|       TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256 (secp256r1) - A
|       TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384 (secp256r1) - A
|       TLS_DHE_RSA_WITH_AES_128_GCM_SHA256 (dh 4096) - A
|       TLS_DHE_RSA_WITH_AES_256_GCM_SHA384 (dh 4096) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 (secp256r1) - A
|       TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA (secp256r1) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384 (secp256r1) - A
|       TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA (secp256r1) - A
|       TLS_DHE_RSA_WITH_AES_128_CBC_SHA256 (dh 4096) - A
|       TLS_DHE_RSA_WITH_AES_128_CBC_SHA (dh 4096) - A
|       TLS_DHE_RSA_WITH_AES_256_CBC_SHA256 (dh 4096) - A
|       TLS_DHE_RSA_WITH_AES_256_CBC_SHA (dh 4096) - A
|     compressors: 
|       NULL
|     cipher preference: server
|     warnings: 
|       Key exchange (secp256r1) of lower strength than certificate key
|_  least strength: A

Nmap done: 1 IP address (1 host up) scanned in 7.98 seconds
```

Using the following build with all ciphers enabled (except MD5) 
```
make bin-arm64-efi/ipxe.efi DEBUG=tls:3
```

My error is:

```
boot.ipxe : 104 bytes [script]
ipxe... ok
TLS 0x33650188 created session fedorapeople.org
https://fedorapeople.org/groups/fcos-images/builds/33.20210111.dev.0/aarch64/fed
ora-coreos-33.20210111.dev.0-live-kernel-aarch64...Sending plaintext data:
3364f828 : 01 00 00 80 03 03 bb 82-15 60 80 0f dd cb f8 d3 : .........`......
3364f838 : ca 9b 95 38 48 fc 36 17-b9 4e 9a 15 4a 9e ac f8 : ...8H.6..N..J...
3364f848 : a0 75 56 46 57 85 20 bb-82 15 60 80 0f dd cb f8 : .uVFW. ...`.....
3364f858 : d3 ca 9b 95 38 48 fc 36-17 b9 4e 9a 15 4a 9e ac : ....8H.6..N..J..
3364f868 : f8 a0 75 56 46 57 85 00-04 00 3c 00 3d 01 00 00 : ..uVFW....<.=...
3364f878 : 33 00 00 00 15 00 13 00-00 10 66 65 64 6f 72 61 : 3.........fedora
3364f888 : 70 65 6f 70 6c 65 2e 6f-72 67 00 01 00 01 04 00 : people.org......
3364f898 : 0d 00 08 00 06 04 01 05-01 06 01 ff 01 00 01 00 : ................
3364f8a8 : 00 23 00 00                                     : .#..            
Received plaintext data:
3364e386 : 02 28                                           : .(              
TLS 0x33650188 received fatal alert 40
 Operation not permitted (http://ipxe.org/410de18f)
Could not boot: Operation not permitted (http://ipxe.org/410de18f)
Could not boot image: Operation not permitted (http://ipxe.org/410de18f)
No more network devices
```

To me this seems strange because the AES-CBC SHA1/SHA256 are as far as I can see supported in iPXE.

amjanoni, 2021-07-23:
I see the same issue when I'm trying to run iPXE on web servers that are behind Cloudflare, looks there's no compatible cipher.

https://developers.cloudflare.com/ssl/ssl-tls/cipher-suites

amjanoni, 2021-07-26:
> I see the same issue when I'm trying to run iPXE on web servers that are behind Cloudflare, looks there's no compatible cipher.
> 
> https://developers.cloudflare.com/ssl/ssl-tls/cipher-suites

JFYI: In my case, I found that newly edge certificates in Cloudflare are using ECDSA and some old domains have RSA, using an RSA certificate fixes the problem.

kevclarx, 2022-02-08:
Are there any ipxe forks that address this(ecdhe cipher support)?  I saw on a old post it was mentioned packet.com but I was not able to find it there.  Pretty much unusable with any modern https server.

bantu, 2022-02-08:
> JFYI: In my case, I found that newly edge certificates in Cloudflare are using ECDSA and some old domains have RSA, using an RSA certificate fixes the problem.

That's a separate issue. This issue here relates to the cryptographic primitive of key agreement, for example ephemeral diffie hellman (DHE) or ephemeral diffie hellman using elliptic curves (ECDHE) whereas the ECDSA vs. RSA is related to signatures and certificates.

bantu, 2022-02-08:
> Are there any ipxe forks that address this(ecdhe cipher support)? I saw on a old post it was mentioned packet.com but I was not able to find it there. Pretty much unusable with any modern https server.

My recommendation for resolving this issue would be to implement ECDHE with only the x25519 named curve. Doing so should require only minimal binary space and should allow to support `TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256 (x25519)` in TLS 1.2. RFC8422 seems relevant to this.

brokenjacobs, 2022-03-16:
Adding support for SHA1 with TLS1.2 would also help compatiblity. For instance google's list on gcp:
includes:
TLS_RSA_WITH_AES_128_CBC_SHA
TLS_RSA_WITH_AES_256_CBC_SHA
but does not include:
TLS_RSA_WITH_AES_128_CBC_SHA256
TLS_RSA_WITH_AES_256_CBC_SHA256

braindevices, 2022-05-10:
hit by this too. need something like TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384 or other ECDHE cipher, more an more services drop the old ciphers.

leeyc0, 2022-08-29:
I think the title is better to change to support modern cipher, newer TLS protocols and ECDSA certificates. Cryptography security policy is now more stringent and server may no longer support older and less secure ciphers.

To be exact, iPXE need to support these cryptography algorithms:

- TLS 1.3 and mandatory ciphers
  - TLS_AES_256_GCM_SHA384
  - TLS_AES_128_GCM_SHA256
  - TLS_CHACHA20_POLY1305_SHA256
- NIST P-256 and x25519 ECDH key exchange
- ECDSA certificates

leeyc0, 2022-08-29:
> My recommendation for resolving this issue would be to implement ECDHE with only the x25519 named curve. Doing so should require only minimal binary space and should allow to support [...]

Unfortunately NIST P-256 is a MUST in TLS 1.3 RFC, while x25516 is a SHOULD only.

NiKiZe, 2022-10-13:
Is this resolved by 6b2c94d3a7d93a8fc47fcb0b895477d4dafca5f0 ?

mcb30, 2022-10-27:
DHE is now supported in iPXE as of commit https://github.com/ipxe/ipxe/commit/2c7824273

I have verified that iPXE is able to negotiate the `dhe-rsa-aes_cbc-128-sha256` cipher suite with `https://fedorapeople.org`

closed 2022-10-27 by mcb30

bantu, 2022-10-27:
> DHE is now supported in iPXE as of commit [2c7824273](https://github.com/ipxe/ipxe/commit/2c7824273)

Very nice! You might want to keep this issue open as per the "broader scope" suggestion in https://github.com/ipxe/ipxe/issues/246#issuecomment-1229690029

stappersg, 2022-10-27:
> Someone update the docs here to list DHE. Tq! https://ipxe.org/crypto

Part of current source of that wiki page.
```
===== Supported configurations =====

^ Protocol versions        | TLSv1.0 TLSv1.1 TLSv1.2    |
^ Public-key algorithms    | RSA                        |
^ Block cipher algorithms  | AES-128-CBC AES-256-CBC    |
^ Hash algorithms          | MD5 SHA-1 SHA-224 SHA-256 SHA-384 SHA-512 SHA-512/224 SHA-512/256  |
```

Where should _DHE_ be added?

joernheissler, 2022-10-27:
> Where should DHE be added?

New line, "Key exchange algorithms"?