ipxe/ipxe #250: [BUG] src/interface/efi/efi_snp.c get stuck UEFI/iPXE boot(Dell R620 in current example)

banuchka, 2021-02-05:
I've catch a bug in scenario when I've tried to boot with UEFI mode on Dell R620.
iPXE work ok on any stage before https://github.com/ipxe/ipxe/commit/1295b4acf. I've tried few commits between v1.20.1 and v1.21.1.
Yesterday we discussed an issue on IRC and I'm here to show some debug info on it

I've build iPXE with
`$ make bin-x86_64-efi/ipxe.efi DEBUG=efi_driver:3,efi_pci:3,snp:3`
Here is a result with hanging on boot - https://gist.github.com/banuchka/92ac97053af1a0eef0ff6018f962c636

Some quote from IRC channel as well:
> - It looks as though what happens is this:
mcb30
> - iPXE disconnects "Broadcom Gigabit Ethernet Driver" from the PCI device handle
mcb30
> - iPXE installs its own driver on the PCI device handle
mcb30
> - iPXE successfully installs a new network interface using its own driver
mcb30
> - The platform firmware creates a child interface of iPXE's interface
mcb30
> - iPXE attempts to connect to this child interface
mcb30
> - Things go wrong
mcb30
> This is definitely a bug. 

Feel free to ping me if I may help with some additional debug/output on current issue.

Thanks

mcb30, 2021-02-05:
Copying the debug output here from the gist for easier access:

```
iPXE initialising devices...EFIDRV connecting our drivers
EFIPCI 0000:00:00.0 (8086:3c00 class 060000) has no driver
EFIPCI 0000:00:01.0 (8086:3c02 class 060400) has no driver
EFIPCI 0000:02:00.0 (14e4:165f class 020000) has driver "14e4-165f"
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) has driver "PCI"
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) before disconnecting:
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) HiiConfigAccess supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) HiiConfigAccess opened 2x (H) by DxeCore(?) for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) HiiConfigAccess opened 4x (T) by Dell DCS Convert Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) 86c77a67-0b97-4633-a187-49104d0685c7 supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) af1291d0-4594-4fd6-84df-dfd2a643d0ed supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) 75e757d5-8571-4718-9e13-affbba5cf38d supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) BusSpecificDriverOverride supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) BusSpecificDriverOverride opened 4x (H) by DxeCore(?) for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo opened 37x (H) by DxeCore(?) for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo opened 1x (G) by DCSHostCom for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo opened 1x (D) by Broadcom Gigabit Ethernet Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo opened 1x (C) by Broadcom Gigabit Ethernet Driver for PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo opened 1x (G) by Simple Network Protocol Driver for PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 2366x (H) by DxeCore(?) for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 1x (D) by Broadcom Gigabit Ethernet Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 4x (T) by Platform Console Management Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 4x (T) by Simple Network Protocol Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 4x (T) by Platform Console Management Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 21x (G) by Dell DCS PCI Driver
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) disconnecting existing drivers
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) after disconnecting:
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) BusSpecificDriverOverride supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) BusSpecificDriverOverride opened 4x (H) by DxeCore(?) for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo opened 37x (H) by DxeCore(?) for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo opened 1x (G) by DCSHostCom for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) PciIo opened 1x (G) by Simple Network Protocol Driver for UNKNOWN<0xc4217798>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath supported
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 2366x (H) by DxeCore(?) for <NULL>
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 4x (T) by Platform Console Management Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 4x (T) by Simple Network Protocol Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 4x (T) by Platform Console Management Driver
HANDLE PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DevicePath opened 21x (G) by Dell DCS PCI Driver
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) connecting new drivers
EFIPCI 0000:02:00.0 (14e4:165f class 020000) has driver "14e4-165f"
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) has driver "PCI"
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) DRIVER_START
EFIPCI 0000:02:00.0 (14e4:165f class 020000) has driver "14e4-165f"
EFIPCI 0000:02:00.0 d90a0000+10000 is not within root bridge address space
EFIPCI 0000:02:00.0 using driver "14e4-165f"
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0) using driver "PCI"
SNP PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) is an SNP device
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) has driver "SNP"
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) DRIVER_START
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) is not supported by driver "PCI": Error 0x7f10e183 (http://ipxe.org/7f10e183)
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) is not supported by driver "NII": Error 0x7f684083 (http://ipxe.org/7f684083)
SNP PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) is an SNP device
EFIPCI 0000:00:00.0 (8086:3c00 class 060000) has no driver
EFIPCI 0000:00:01.0 (8086:3c02 class 060400) has no driver
SNP PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) is an SNP device
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) has driver "SNP"
[... cut ...]
EFIDRV PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) is not supported by driver "NII": Error 0x7f684083 (http://ipxe.org/7f684083)
SNP PcieRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(C81F66D06D4F,0x1)/IPv4(0.0.0.0) is an SNP device

Exception Inside the Exception Handler
Type: General Protection Fault (13) Source: Software (UEFI0011)
AX=AFAFAFAFAFAFAFAF BX=00000000CCADFAC0 BP=00000000C209E898 SP=00000000CCA858C0
CX=0000000000000010 DX=00000000C305D288 R8=0000000000000000 R9=00000000CCAE201F
SI=0000000000000000 DI=00000000C594E728 SS=0018 CS=0038     Flags=00010202
IP=00000000CCAC97DE
```

daberkow, 2021-02-24:
Not surprisingly, the Dell R720 is also seeing this issue

TimRots, 2021-02-25:
Are you observing similar issues when using ``snponly.efi`` ?

daberkow, 2021-02-25:
@TimRots I just pulled the `snponly.efi` that one worked correctly!