ipxe/ipxe #251: efi gdbstub crash

bozax, 2021-02-07:
I build ipxe like : make bin-x86_64-efi/ipxe.efi, but when I debug it with gdbstub serial, ipxe crash at line "__asm__ __volatile__ ( "int $3" )" in function gdbmach_breakpoint. What's wrong ? Does ipxe.efi can debug with gdbstub? If not ,how can I debug it?
Thanks you very much!

mcb30, 2021-02-07:
I don't think the gdbstub was ever extended to handle UEFI, sorry.  I don't even know if the BIOS version will still be working: I haven't personally tried using it in years.

You can perform most debugging tasks via the `DEBUG=...` mechanism for enabling debug prints on a per-object basis.  Most of the code should give a verbose debug message for any error that occurs in operation.  If you follow the URL from any iPXE error message it will take you to a page that suggests an appropriate `DEBUG=...` option to start with.

mcb30, 2021-02-07:
Closing since this is not a bug: the gdbstub is not expected to work in UEFI.  Do please open another issue (or discussion) for whatever problem you found that led you to try the debugger in the first place!

closed 2021-02-07 by mcb30