ipxe/ipxe #255: Doesn't work on AWS EC2 Nitro instances

remcovanmook, 2021-02-10:
Even though the ena driver is in the repo, trying to boot an AWS EC2 Nitro instance using ipxe (ipxe.lkrn by way of Grub) doesn't work - no PCI devices are detected. Compiling with DEBUG=pci,ena shows no information.

'config pci' shows no devices recognised even though a booted Linux kernel shows the following devices as available:
00:00.0 Host bridge: Intel Corporation 440FX - 82441FX PMC [Natoma]
00:01.0 ISA bridge: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II]
00:01.3 Non-VGA unclassified device: Intel Corporation 82371AB/EB/MB PIIX4 ACPI (rev 08)
00:03.0 VGA compatible controller: Amazon.com, Inc. Device 1111
00:04.0 Non-Volatile memory controller: Amazon.com, Inc. Device 8061
00:05.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
[image]

NiKiZe, 2021-02-10:
lspci -nn from Linux would give the pciid

remcovanmook, 2021-02-10:
Here's the lspci output. The PCI ID matches with the one in the ena driver.
# lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation 440FX - 82441FX PMC [Natoma] [8086:1237]
00:01.0 ISA bridge [0601]: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II] [8086:7000]
00:01.3 Non-VGA unclassified device [0000]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 08)
00:03.0 VGA compatible controller [0300]: Amazon.com, Inc. Device [1d0f:1111]
00:04.0 Non-Volatile memory controller [0108]: Amazon.com, Inc. Device [1d0f:8061]
00:05.0 Ethernet controller [0200]: Amazon.com, Inc. Elastic Network Adapter (ENA) [1d0f:ec20]

NiKiZe, 2021-02-10:
Not that much debug statements, but try with DEBUG=pci:3,ena:3 and check if that gives anything.

One other longshot is to test with older version of iPXE? Right now it's not clear if it is any hardware change or iPXE change that is causing this.

remcovanmook, 2021-02-10:
Looking at the ipxe-devel mailing list it seems like this has never worked with the newer Amazon EC2 instances. Maybe @mcb30 can help shed some light? He added the ena driver back in 2018 so I'm assuming it was working at that point?

I'll recompile with DEBUG=pci:3,ena:3 and report back.

remcovanmook, 2021-02-10:
No output from DEBUG=pci:3,ena:3 either. I tried exploring the PCI space with pciscan but that also didn't return anything.

mcb30, 2021-02-10:
@remcovanmook what instance type can I use to recreate this problem?

remcovanmook, 2021-02-10:
t3a.nano, t3a.micro for example.

mcb30, 2021-02-13:
So, initial investigation shows that using `PCIAPI_DIRECT` rather than `PCIAPI_PCBIOS` does at least detect the devices on the PCI bus.  It looks as though the (modified) SeaBIOS in this instance family always returns an error from the `INT 1A,B10x` call that we use to read from PCI configuration space.

Extracting and disassembling the SeaBIOS image to see what's going on.

mcb30, 2021-02-13:
> So, initial investigation shows that using `PCIAPI_DIRECT` rather than `PCIAPI_PCBIOS` does at least detect the devices on the PCI bus. It looks as though the (modified) SeaBIOS in this instance [...]

Confirmed my hypothesis: the disassembly shows that the `INT 1A,B1xx` implementation is just a stub that always sets `CF` to indicate an error.  It looks as though the only option is to use `PCIAPI_DIRECT` (in `config/ioapi.h`).

There's also some sort of timing issue that causes the ENA driver to fail probing (unless debug output is enabled).  Looking at this now...

mcb30, 2021-02-13:
> There's also some sort of timing issue that causes the ENA driver to fail probing (unless debug output is enabled). Looking at this now...

Commit https://github.com/ipxe/ipxe/commit/004924336 fixes this issue.  With `PCIAPI_DIRECT` enabled, I now reliably get a working ENA NIC on both `t3.nano` and `t3a.nano`

remcovanmook, 2021-02-13:
That's amazing. I'll try and report back. Thanks @mcb30 !

mcb30, 2021-02-13:
I've also pushed commit https://github.com/ipxe/ipxe/commit/83516ba7f which enables `PCIAPI_DIRECT` automatically when building with `CONFIG=cloud`.

@remcovanmook There is an AMI `ami-0cd0bc62f064331cb` that I've made public in `eu-west-1` that you can try out if you like.  This is commit https://github.com/ipxe/ipxe/commit/83516ba7f built using
```
make CONFIG=cloud EMBED=config/cloud/aws.ipxe bin/ipxe.usb
```
It will boot whatever iPXE script you provide via the instance user-data.

remcovanmook, 2021-02-13:
> I've also pushed commit [83516ba](https://github.com/ipxe/ipxe/commit/83516ba7f) which enables `PCIAPI_DIRECT` automatically when building with `CONFIG=cloud`.
> 
> @remcovanmook There is an AMI `ami-0cd0bc62f064331cb` that I've made public in `eu-west-1` that you can try out if you like. This is commit [83516ba](https://github.com/ipxe/ipxe/commit/83516ba7f) built using
> 
> ```
> make CONFIG=cloud EMBED=config/cloud/aws.ipxe bin/ipxe.usb
> ```
> 
> It will boot whatever iPXE script you provide via the instance user-data.

Tested on t3a.micro in eu-central-1 -works like a charm. Thanks again for taking the time to investigate and fix!

closed 2021-02-13 by remcovanmook