ipxe/ipxe #258: ISO creation fails

JuniorJPDJ, 2021-02-12:
```
$ mkisofs --version
mkisofs 3.02a09 (x86_64-unknown-linux-gnu)

Copyright (C) 1993-1997 Eric Youngdale
Copyright (C) 1997-2017 Joerg Schilling
Copyright (C) 1997-2001 James Pearson
Copyright (C) 2006      HELIOS Software GmbH

$ genisoimage --version
mkisofs 3.02a09 (x86_64-unknown-linux-gnu)

Copyright (C) 1993-1997 Eric Youngdale
Copyright (C) 1997-2017 Joerg Schilling
Copyright (C) 1997-2001 James Pearson
Copyright (C) 2006      HELIOS Software GmbH

$ ./util/genfsimg -o ~/Downloads/netboot.iso ~/Downloads/netboot.xyz.lkrn ~/Downloads/netboot.xyz-snp.efi
./util/genfsimg: cannot find a suitable mkisofs or equivalent
```

mcb30, 2021-02-12:
You are trying to build an EFI-bootable ISO (since you include `~/Downloads/netboot.xyz-snp.efi`), and your installed version of `mkisofs` does not have the ability to generate EFI-bootable ISO images.  Try installing `xorrisofs`.

(See commit https://github.com/ipxe/ipxe/commit/7c3d186a1 for details)

closed 2021-02-12 by mcb30

mcb30, 2021-02-12:
@JuniorJPDJ I've pushed commit https://github.com/ipxe/ipxe/commit/df16df2c8 to make the error message more useful: with this code, you would have seen:

```
./util/genfsimg: xorrisofs: not installed
./util/genfsimg: mkisofs genisoimage: cannot handle -J -R -no-emul-boot -eltorito-boot isolinux.bin -boot-load-size 4 -boot-info-table -eltorito-alt-boot -no-emul-boot -e esp.img
./util/genfsimg: cannot find a suitable mkisofs or equivalent
```

NiKiZe, 2021-02-12:
Not sure how, but should the requirements at https://ipxe.org/download be updated in some way?

mcb30, 2021-02-12:
@NiKiZe Updated to mention the possible alternatives