ipxe/ipxe #269: Provide/generate a label for EFI devices that don't have one for file:// URI

patlefort, 2021-02-21:
As it stands, if a device doesn't have a label, it can't be accesses by a ipxe script with a "file://" URI. As this [commit message](https://github.com/ipxe/ipxe/commit/9913a405ea151706770559d42e758bc2c1da8525) say, mappings like "fs0:" are not exposed, so their label is empty, so maybe ipxe could generate one to make them accessible?

It could be useful to control what a machine should boot remotely, maybe fetch an image from network, maybe boot something locally.

Also, allow the use of the complete device path as the volume label. That path uses forward slashes, so maybe the URI should be written with them replaced by backslashes or some other character.

mcb30, 2021-02-22:
> Also, allow the use of the complete device path as the volume label. That path uses forward slashes, so maybe the URI should be written with them replaced by backslashes or some other character.

This part is unfortunately not possible: the `EFI_DEVICE_PATH_TO_TEXT_PROTOCOL` and `EFI_DEVICE_PATH_FROM_TEXT_PROTOCOL` are optional and there are systems that do not provide them.

mcb30, 2021-02-22:
> As it stands, if a device doesn't have a label, it can't be accesses by a ipxe script with a "file://" URI. As this [commit [...]

It would definitely be useful, but the question is: what can sensibly be used as an alternative means of identifying an unlabelled filesystem?

patlefort, 2021-02-22:
I'm not sure what would be the best way, but maybe simply generating one with a fixed prefix and an increasing number? It would at least make them accessible by index, it could be useful if the index is consistent across reboots. I would avoid "fs" or "blk" to avoid confusion as they might not have the same numbering and it's just a matter of using a prefix that would have no collision. Maybe it could be a special notation in the URI, like "file://device[0]/" that's simply parsed for an index?

mcb30, 2021-02-22:
The problem is that the "index" is not a well-defined concept.  We can enumerate all filesystem handles (as we already do in order to support access by volume name), but the UEFI firmware is allowed to change the order in which handles are returned between calls.

In practice, the order would most likely be stable except when something happens to cause a filesystem to be added or removed (e.g. using the [`sanhook`](https://ipxe.org/cmd/sanhook) command).  However, there's likely to be at least one system out there that does something weird with the ordering, which will lead to users complaining that `file://device0/...` keeps changing which drive it boots.

NiKiZe, 2021-02-22:
Is there anything in efi shell that makes that implementation more consistent? If not, random IDs might be better than no ID. Anyone looking for specific FS will have to enumerate and check for file existence or similar.

mcb30, 2021-02-22:
> Is there anything in efi shell that makes that implementation more consistent? If not, random IDs might be better than no ID. Anyone looking for specific FS will have to enumerate and check for file existence or similar.

From memory, the shell builds a persistent list of devices and registers for notifications when filesystems are added or removed.  We could do the same, but the overhead is reasonably large and the order would still be unstable between invocations of iPXE.

MikeKlem, 2021-02-22:
What about using a random Hex four character identifier?
filesegmentFEED
filesegAC1E
filesBEEF

mcb30, 2021-02-22:
> What about using a random Hex four character identifier?
> filesegmentFEED
> filesegAC1E
> filesBEEF

How would that be used?

MikeKlem, 2021-02-22:
file://filesegment[ABCD]/path/to/file

if it’s addressable, then at least it could be searched.
A la your file://device0/... naming and the mention that "fs0:" is unstable

mcb30, 2021-02-22:
> file://filesegment[ABCD]/path/to/file
> 
> if it’s addressable, then at least it could be searched.
> A la your file://device0/... naming and the mention that "fs0:" is unstable

I'm sorry, I have no idea what you are thinking the `[ABCD]` would represent, or how the user would know what value to provide in order to address a specific device.

It sounds as though you may be thinking of volume labels, which iPXE already supports.  If this is not what you mean, then could you give any kind of concrete example?

MikeKlem, 2021-02-22:
I was thinking it would represent unlabeled filesystems, where ABCD was either randomly or deterministicly generated, and thus iterable.

I’ll try to come up with an example

patlefort, 2021-02-23:
Here's an idea for generating a label: checking for the existance of a file on the device that would start with some prefix and use the filename as the label.

Otherwise, the only way I can think of is to search each devices for the queried file.

mcb30, 2021-02-23:
> Here's an idea for generating a label: checking for the existence of a file on the device that would start with some prefix and use the filename as the label.
> 
> Otherwise, the only way I can think of is to search each devices for the queried file.

We could viably treat an empty volume label (i.e. `file:///path`) as meaning "use the first[*] filesystem that contains this file".  That would be very similar code to what we currently have for iterating over filesystems and checking the volume label.

[*] where "first" means "first one we come across, in whatever undefined order the platform chooses to present the handles to us"

patlefort, 2021-02-23:
> We could viably treat an empty volume label (i.e. `file:///path`) as meaning "use the first[*] filesystem that contains this file". That would be very similar code to what we currently have for [...]

That'd be fine with me.