ipxe/ipxe #276: Request: [slirp] Add --help with the full --help Text for Linux

MaxPeal, 2021-03-03:
Please add a full help text via --help
And --version or --dump to get all infos out of the compiled in values.

As it really helps to use the ipxe Linux in a CI environment for testing.

As the info in the commit is not easy to find
https://github.com/ipxe/ipxe/commit/2b5d3f582f718ca11488fb6d92ea39dd22b8ffed

mcb30, 2021-03-03:
The code is one day old and someone has already found a use for it :slightly_smiling_face:

I'm hoping to get as far as implementing `kexec` within iPXE, so that the Linux userspace build can be used to provide fast rebooting within an existing iPXE-based setup.  If it reaches this point, I'll write up some documentation for it.

MaxPeal, 2021-03-03:
Actually I compiled and played with it, 
2h after the commit. ☺

Have you already some plans or a working CI environment? I was thinking about GH actions,
Together with packer or vagrant