ipxe/ipxe #288: Boot occasionally getting stuck

konstantin-kelemen, 2021-03-11:
Hi,

I'm facing an issue that happens periodically because of which the boot process simply hangs during PXE boot.
[image]
_it stays like this forever_

Currently, it is only happening on Dell R6515 and Dell R7525 servers equipped with Broadcom Adv. Dual 10GBASE-T Ethernet in NIC Mezzanine 1 slot.

This happens roughly every second boot and can be fixed by rebooting the server or by pressing Ctrl+C and retrying the boot.

During my debugging attempts I have also entered the iPXE command line, configured the network with `dhcp` command and tried to ping the pxe server. It was OK. Then I attempted the boot the kernel and the boot got stuck. I canceled the boot and tried to ping the pxe server again. This time here was no response even though on the pxe server itself I was able to see with tcpdump both requests and replies.

Also, I've built the undionly.kpxe with the following options: `make bin/undionly.kpxe DEBUG=tcp,ipv4`
The only thing I've noticed is that during unsuccessful boot I get the following message: 

> IPv4 length too long at 1500 bytes (packet is 1488 bytes)

[image]

I don't think this message is somehow can be linked to the root cause of the issue since I was able to get rid of it by changing the package length on the pxe server and still get the freeze during boot.

Any help and suggestions on how to debug and fix this issue would be highly appreciated.

NiKiZe, 2021-03-11:
Maybe add debug for the nic driver to try and find more what is actually going on?
Could you try to catch the traffic with tcpdump, wireshark or similar to try and see what happens.

Also try to run ifstat before the fetch starts, and after you cancel and see if there is anything that stands out. (In your manual example, use ifstat instead of ping)

Is there any VLAN, or other configuration involved on the network here? Maybe mismatching MTUs somewhere?

mcb30, 2021-03-11:
It does sound very suspiciously like an MTU issue.  You are able to receive without any problem until the first expected MTU-sized packet (as part of downloading the kernel, which is the first thing you attempt to fetch that is larger than one MTU).  The `IPv4 length too long` message indicates that a packet has been truncated somewhere on the receive datapath.  The 0% progress combined with `sending keepalive` is precisely what you will see if iPXE believes that the remote end is simply not sending anything.

With `undionly.kpxe`, the receive datapath is unfortunately quite opaque to us, because the NIC driver is being provided by the Broadcom PXE ROM rather than by iPXE.

As an experiment, you could try forcing iPXE's MTU to a small value using e.g. `set net0/mtu 1024` and then attempt the download.  This will cause iPXE to request a TCP MSS of 1024 (minus 40 bytes allowed for TCP/IP headers without options), which should in theory cause the server to send packets no larger than 1024 (plus the 14-byte Ethernet link-layer header).  There's still the possibility that TCP segmentation offload on the server side could violate this, though.

lukasstockner, 2021-03-11:
We are seeing similar issues with Broadcom NICs (in particular, P225p and M225p), also with UNDI.

In our case, it's hanging (just printing dots for hours) when downloading the initial script via HTTP, so I doubt that it's a MTU issue (the entire response is just 300 bytes).

If you think it helps, I can provide tcpdump output and screenshots - from what I remember, it seems like iPXE is not receiving the response from the server since it keeps re-ACKing the initial handshake.

I also tried with `ipxe.pxe` and `bnxt.pxe`, but couldn't get it working at all, so for now I'm stuck to UNDI.

konstantin-kelemen, 2021-03-11:
Hi @NiKiZe, @mcb30, thanks for your replies.

> Maybe add debug for the nic driver to try and find more what is actually going on?

I've added both bnxt and bnx2 (`DEBUG=tcp,ipv4,bnxt,bnx2`) but don't see any new messages in the output. Am I doing it wrong?

> Could you try to catch the traffic with tcpdump, wireshark or similar to try and see what happens.

Already tried that on the pxe end. Sadly no noticeable differences between successful and unsuccessful boots.

> Also try to run ifstat before the fetch starts, and after you cancel and see if there is anything that stands out. (In your manual example, use ifstat instead of ping)

I can't see much difference in ifstat output as well. The only thing I see is that the RX counter doesn't grow but TX does.
Also I can't reproduce that in any other OS booted from livecd.

> Is there any VLAN, or other configuration involved on the network here? Maybe mismatching MTUs somewhere?

No there is nothing that would raise suspicion.

> It does sound very suspiciously like an MTU issue. You are able to receive without any problem until the first expected MTU-sized packet (as part of downloading the kernel, which is the first thing [...]

I understand what you're saying but it is also strange that it works in 50% of cases and fails in the rest. Also there is no noticeable difference in tcpdump output on pxe server end between successful and unsuccessful boots.

> As an experiment, you could try forcing iPXE's MTU to a small value using e.g. `set net0/mtu 1024` and then attempt the download. This will cause iPXE to request a TCP MSS of 1024 (minus 40 bytes [...]

I've tried this solution but unfortunately it keeps hanging.
[image]

mcb30, 2021-03-11:
> I understand what you're saying but it is also strange that it works in 50% of cases and fails in the rest. Also there is no noticeable difference in tcpdump output on pxe server end between [...]

What is the size of the transmitted HTTP packets as shown in the wireshark capture?  (Most likely the server is using segmentation offloading, in which case the wireshark capture will show a handful of pre-segmentation giant packets, rather than showing you the actual packets as transmitted on the wire.)

konstantin-kelemen, 2021-03-12:
> What is the size of the transmitted HTTP packets as shown in the wireshark capture? (Most likely the server is using segmentation offloading, in which case the wireshark capture will show a handful [...]

MTU in iPXE is set to 1500, it is the same on pxe server with tcp segmentation offloading disabled.

Here you can compare captures of good and bad boot sequences: https://gist.github.com/konstantin-kelemen/3696c6b26a0d35cb5e056aa91118abaa
Note how packet size is never getting larger than set MTU.

Here is a nginx log record for successful boot:
`192.168.101.179 - - [12/Mar/2021:13:37:15 +0000] 0.039 "GET /some_path/vmlinuz-vanilla HTTP/1.1" 200 5191328 "-" "iPXE/1.21.1+ (g65bd)" "-" "-"`

and unsuccessful boot (note the smaller amount of bytes sent):
`192.168.101.179 - - [12/Mar/2021:13:39:22 +0000] 19.098 "GET /some_path/vmlinuz-vanilla HTTP/1.1" 200 89776 "-" "iPXE/1.21.1+ (g65bd)" "-" "-"`

And here is my unsuccessful attempt at booting the kernel from the iPXE command line
[image]

mcb30, 2021-03-12:
Thanks.  I'm going to ask for one more MTU test, just because the failure symptoms **precisely** match those of an MTU issue.

Could you please build iPXE with `PING_CMD` enabled and find the largest `<N>` for which
```
ping -c 20 -s <N> <your-http-server-address>
```
does not fail?

konstantin-kelemen, 2021-03-12:
> Could you please build iPXE with `PING_CMD` enabled and find the largest `<N>` for which
> 
> ```
> ping -c 20 -s <N> <your-http-server-address>
> ```
> 
> does not fail?

ping works with number of up to 1486 bytes and stops working at 1487.

During test I've also got some new unexpected results.

During ping with packet size of 1472 bytes I've got the same `IPv4 length too long at 1500 bytes (packet is 1488 bytes)` message after which the network stops working at all.
[image]

And these are the dumps of the ping command running before and after network breaks: https://gist.github.com/konstantin-kelemen/3263d8317cb82d1d4b7a713ca05310d0

Also same happened again with packet size of 1483 bytes:
[image]

Dump at that moment: https://gist.github.com/konstantin-kelemen/971d19fe549241e0c8d107997118dc38

mcb30, 2021-03-12:
Thanks; that's really useful to have a single identifiable event that occurs at the point that the Dell PXE ROM receive datapath locks up!  Could you please try with this patch?
```diff
diff --git a/src/net/ipv4.c b/src/net/ipv4.c
index b9ce5e7f7..de0075071 100644
--- a/src/net/ipv4.c
+++ b/src/net/ipv4.c
@@ -521,6 +521,8 @@ static int ipv4_rx ( struct io_buffer *iobuf,
        if ( hdrlen > iob_len ( iobuf ) ) {
                DBGC ( iphdr->src, "IPv4 header too long at %zd bytes "
                       "(packet is %zd bytes)\n", hdrlen, iob_len ( iobuf ) );
+               DBGC_HDA ( iphdr->src, 0, iobuf->data, 128 );
+               DBGC_HDA ( iphdr->src, 0, iobuf->data + 1424, 64 );
                goto err_header;
        }
        if ( ( csum = tcpip_chksum ( iphdr, hdrlen ) ) != 0 ) {
```
That should show the contents of the `IPv4 length too long` packet, which will hopefully give us some idea of where it came from.

konstantin-kelemen, 2021-03-16:
I've tried the patch but got no additional output in the command line during tests. Am I doing something wrong?

[image]

mcb30, 2021-03-16:
My fault, sorry: the patch added the lines to the wrong `IPv4 <thing> too long` error case.  Please try this one instead:

```diff
diff --git a/src/net/ipv4.c b/src/net/ipv4.c
index b9ce5e7f7..4f95a4ee4 100644
--- a/src/net/ipv4.c
+++ b/src/net/ipv4.c
@@ -537,6 +537,8 @@ static int ipv4_rx ( struct io_buffer *iobuf,
        if ( len > iob_len ( iobuf ) ) {
                DBGC ( iphdr->src, "IPv4 length too long at %zd bytes "
                       "(packet is %zd bytes)\n", len, iob_len ( iobuf ) );
+               DBGC_HDA ( iphdr->src, 0, iobuf->data, 128 );
+               DBGC_HDA ( iphdr->src, 0, iobuf->data + 1424, 64 );
                ipv4_stats.in_truncated_pkts++;
                goto err_other;
        }
```

konstantin-kelemen, 2021-03-16:
Thanks, it worked this time and this is what I've got:

[image]

Unfortunately, this doesn't tell me anything.

mcb30, 2021-03-16:
Great.  The payload there is definitely a ping reply: the IPv4 header shows source address `c0 a8 08 26 == 192.168.8.38`, and the ICMPv4 header shows type `0 == ICMP_ECHO_REPLY`.

Curiously, the ICMPv4 sequence field shows `00 0c == seq=12`, so this is a copy of the response from several seconds ago that was already successfully received.  This suggests that either the packet has somehow been duplicated on the network (which is very unlikely), or that the Broadcom PXE ROM has screwed up and used the wrong buffer.

Since we already know that the Broadcom PXE ROM stops returning any further packets after this happens, this looks like fairly conclusive evidence of a bug in the Broadcom PXE ROM.

Slightly complicating the issue is the fact that the pings are using IPv4 fragmentation.  Could you leave the patch in place and reproduce the original `IPv4 length too long` failure seen when attempting to download your kernel?

Thanks.

lukasstockner, 2021-03-16:
FYI, in my case I could get it working by using `undionly.kkpxe`.

konstantin-kelemen, 2021-03-16:
> Could you leave the patch in place and reproduce the original `IPv4 length too long` failure seen when attempting to download your kernel?

Here is what I'm getting when the boot hangs:

[image]

mcb30, 2021-03-16:
Thank you.  That definitely shows a TCP packet from the web server (source address is `c0 a8 08 26 == 192.168.0.38` and source port is `00 50 == port 80`.  There is no IPv4 fragmentation (the fragments field has value `40 00` with the "more fragments" bit clear and the "do not fragment" bit set).  Overall, it's consistent with what we've already seen: the Broadcom PXE ROM seems to be just returning incorrect values and then hanging.

Could you try building with `DEBUG=tcp,ipv4,undinet` to see if the UNDI driver reports any odd behaviour?

konstantin-kelemen, 2021-03-16:
> Could you try building with `DEBUG=tcp,ipv4,undinet` to see if the UNDI driver reports any odd behaviour?

Sure, here is the result:

[image]

Don't see anything suspicious here.

konstantin-kelemen, 2021-03-18:
I've tried using `undionly.kkpxe` as @lukasstockner suggested and found it to be working just fine.

Should I use it instead of `kpxe` version then? I'm not sure since I've never dealt with it before and I'm afraid that it would break PXE boot on some other hardware. Can you provide any recommendations?

Are there any other suggestions of fixes/tests I can run to make `kpxe` version work stable in this case?

konstantin-kelemen, 2021-05-11:
Since there was no new replies and suggestions for a long time I'm going to close this issue. I'd like to thank everyone providing help and assistance here!
In the end we've switched to `kkpxe` version which works fine.

closed 2021-05-11 by konstantin-kelemen

usvi, 2022-10-20:
I'm seeing something very similiar happen in VMware Workstation 15 Player forced 32bit guest mode for ipxe.pxe , but undionly.kpxe seems to be just fine. Curiously, boot stage succeeds on an old real 32bit hardware.