ipxe/ipxe #291: iPXE hang while 'disconnecting existing drivers'

codwazny, 2021-03-11:
This is a continuation of the discussion from https://github.com/ipxe/ipxe/discussions/287

I have an issue on some of our servers where we have iPXE hangs during initialising devices phase. After a few minutes of hanging, they will reboot. The issue does not happen 100% of the time which makes it even harder to figure out. Occasionally it will go through without any issues but often times it will get stuck and hang.

To try and give the best logging back I recompiled iPXE with:
```
DEBUG="device,runtime,pci,init,nii,snp,snpnet,snponly,efi_watchdog,efi_image,image_cmd,image,efi_autoboot,efi_timer,efi_driver:3,efi_debug,efi_init,efi_pci,efi_wrap,intel,intelxl"
```

What else can I try to do to help debugging this? Is this a UEFI issue with the bios?

Extra details:
Built directly from github rev 1192edf394aaebde666c6aafda1c8b782b08a845

Full iPXE log of hang: https://gist.github.com/codwazny/345b95d6d858f547748f050c9871341c
Full iPXE log of working boot: https://gist.github.com/codwazny/1e1f7ce5280b98c08cb2a61063389391

The servers:
Dell R340 w/ X710 card (I often see this same issue on other Dell models as well). The server is running latest BIOS (2.4.1) and latest NIC firmware (19.5.12).

mcb30, 2021-03-11:
Thank you for the updated logs!

There are now no non-trivial differences between the "hang" and "working" logs up to the point of `disconnecting drivers`.

Could you try PXE-booting into the UEFI shell binary (which you can get from https://boot.ipxe.org/Shell.efi) instead of iPXE?  Ideally, overwrite the iPXE binary with `Shell.efi` so that everything else in the boot process remains unaltered.  If that is not possible, then reconfigure the DHCP server to hand out Shell.efi via the same IPv6+TFTP boot process.  Definitely do not boot the shell via some other means (such as from a USB stick), since that could change the outcome of the following test.

Once inside the UEFI shell, use the `devices`, `devtree`, and/or `dh` commands to identify the handle with the device path `PciRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(F8F21E7A0800,0x1)`.  Having identified the handle, try using the `disconnect` command to disconnect all existing drivers from the handle.  This closely mimics what iPXE will do at the point of `disconnecting controllers`.

If this test displays the same pattern of sometimes working with no problems and sometimes hanging, then we will at least have firmly identified it as a bug within the Dell UEFI BIOS.

iPXE has a fairly robust driver veto mechanism in https://github.com/ipxe/ipxe/blob/master/src/interface/efi/efi_veto.c that can be used to forcibly unload (or otherwise destroy) any drivers that are known to fail to disconnect cleanly.  The end fix may well be the addition of more Dell drivers to this list, alongside the existing veto for the Dell IP4ConfigDxe driver.

Since you are performing an IPv6 network boot, and since the Dell IP4ConfigDxe driver is known to have exhibited similar failures in the past (see commit https://github.com/ipxe/ipxe/commit/64b4452bc), my first candidate for vetoing would probably be the Dell IP6ConfigDxe driver.

codwazny, 2021-03-12:
So, I was able to do what you asked. I dropped the Shell.efi binary in place of my normal binary and pxebooted that instead. I was trying to follow what you were saying about finding the device but I'm not sure I was doing it right. I used devtree and looked for what appeared to be the root of the device with lots of things loaded under it. I then disconnected that. Ultimately, I ended disconnecting **21E**, the following device:
```
Ctrl[21E] PciRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)
```
Full output: https://gist.github.com/codwazny/e6e7d391e51503edf93d6fe979e375a9

I was not able to reproduce a full hang. However, I was able to reproduce a mini hang. Sometimes when it would boot up it would disconnect pretty much immediately. Other times, it would hang for between 70-90 seconds and then report a success. In the cases that it took 70-90 seconds, I was able to confirm that afterwards that it actually had unloaded all the children from devtree under that device so it seems to have been working properly.

It seemed that the hangs were less frequent than when using the ipxe binary itself but I don't have proof of it other than a seat of the pants feeling. I tried to see if it was timing related so I let the shell boot up for a while and then disconnected and also tried to rush disconnecting right as it booted. I was not able to find any valuable correlation there.

I can't tell if this is the same issue or not but it seems suspiciously similar. I wonder if the call in ipxe is timing out because it's taking so long to disconnect the drivers? I have watched the ipxe binary boot up tons of times with debugging on and during a successful boot the disconnect logging lines went by instantly. I never once saw a boot that hung for 70-90 seconds at the disconnect line and then continued to work. It either worked immediately or hung until the system rebooted automatically.

mcb30, 2021-03-12:
Most likely the relevant handle is `Ctrl[246] Intel(R) Ethernet Converged Network Adapter X710`.

You can verify this with `dh -v 246` which should include within the output something like:
```
DevicePath(XXXXXXX)
  PciRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/MAC(F8F21E7A0800,0x1)
```

So: we have definitely been able to reproduce what seems likely to be the same underlying problem in the absence of any iPXE code.  This leads now onto finding out which Dell driver we need to add to the veto list.

There are two ways you can try this:

- using Shell.efi, try listing drivers with the `drivers` command and unloading likely candidates with the `unload` command until you find a driver that, after being unloaded, no longer causes problems for `disconnect 246`.  Note that unloading drivers may cause the handle enumeration to change, so you should reverify that `dh -v 246` still points to the correct handle before running the `disconnect` command.
- using iPXE, build with `DEBUG=efi_driver:3,efi_veto:3` to see the list of drivers, and edit `interfaces/efi/efi_veto.c` to kill off likely candidates until you find a driver that, after being vetoed, no longer causes problems for iPXE at the `disconnecting controllers` stage.

Both approaches are similar in effect, so it's largely down to which you will find more convenient.  (On some platforms there may be drivers that do not respond to the `unload` command, in which case only the more forceful `efi_veto` mechanism would work.)

codwazny, 2021-03-19:
So I wanted to give a little update here. I was not able to use the veto mechanism to prevent this hang. Here's what I tried so you can point me in the right direction if I've gone wrong.

I switched DEBUG to what you said: DEBUG="efi_driver:3,efi_veto:3"

Looking through the list of drivers that came up, I figured that "Intel(R) 40GbE 3.9.11" is likely the culprit or would at least trigger the case. I tried vetoing it and got mixed results where it wouldn't hang sometimes but it would other times. It also seemed to break ipxe as it could no longer do the "ifconf -c ipv6" call in my ipxe script. But for testing purposes, maybe that was fine for now.

I went ahead and tried to debug a bit further to see what was happening. I figured that maybe the unload call from efi_veto_unload was still hanging since it's pretty similar to the normal call made to disconnect. So I went ahead and commented out https://github.com/ipxe/ipxe/blob/master/src/interface/efi/efi_veto.c#L353-L355. No improvement. Now since we are just calling efi_veto_disconnect instead, I tried to log what handles we were disconnecting to see if I could see the exact handle that was doing that. I added debug logging here: https://github.com/ipxe/ipxe/blob/master/src/interface/efi/efi_veto.c#L108 with the prefix "disconnect:" (see in the attached logs). This lead me to this being the last line logged on times when it was hanging:

```
 disconnecting: PciRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)
 ```
 
 Hang case: https://gist.github.com/codwazny/5700593d756d9bdae6ca5f42ba4b71ed
 Non hang case: https://gist.github.com/codwazny/4c554fa88b932299b9d7aee01f676df3
 
 I think this leads me to believe that vetoing the driver won't help and that the hang may still occur even if we try to forcefully eject. Any other ideas?