ipxe/ipxe #299: slow loading initrd and linux kernel using i210 / stmmac interfaces

ionutnechita, 2021-03-18:
Hi IPXE Team,

I noticed a problem loading the kernel image with ipxe. I have a very long loading time.

iPXE 1.0.0+ -- Open Source Network Boot Firmware -- http://ipxe.org
Features: DNS HTTP iSCSI TFTP SRP AoE EFI Menu

net0:  xx:xx:xx:xx:xx:xx using NII on NII-0000:00:1d.1 (open)
  [Link:down, TX:0 TXE:0 RX:0 RXE:0]
  [Link status: Unknown (http://ipxe.org/1a086194)]
Configuring (net0 xx:xx:xx:xx:xx:xx)... ok

stmmac driver in linux / I210 network card / 1 Gpbs connection

How can I debug this problem and how can it be solved?

Thanks.

NiKiZe, 2021-03-18:
Have you tested with the latest version?

ionutnechita, 2021-03-18:
Yes, i tested with latest version and is reproduced.

NiKiZe, 2021-04-29:
Could you post updated information with latest version and clearer describe the problem, exactly when is the wait in the output?

Also include information such as pciid (lspci -nn) and information about the machine and which iPXE binary you are using. as well as which kernel you are booting.

With only the information you have posted so far which uses the NII driver and only output from the configuration I have to assume that it is the configuration that takes time, and if so it is the firmware driver, or maybe your network configuration that causes this.

stappersg, 2023-03-02:
> Also try using `ipxe.efi` instead of `snponly.efi`.

Preferable with #907 applied.

suryamurugan, 2023-06-05:
Noticed the same with the default firmware on the same Baremetal. So might not be related to ipxe. I suspect that its related to the network card itself. while the interface was able to get the IP.