ipxe/ipxe #306: Booting image via ISO on ISOLINUX 4.04 fails

matya, 2021-03-30:
Hi,

I am on the current master of 1192edf394.

Booting via an iso image (`make bin/ipxe.iso`) fails by giving an error like this:

[image]

The syslinux used is part of SUSE Linux, in version 4.04:
`/usr/share/syslinux/isolinux.bin: isolinux Loader (version 4.04)`

I have tracked down the issue to the introduction of `genfsimg` via 79c0173d6, where the `isolinux.cfg` file that is produced has changed.

Before it was using the file using the name `ipxe.lkr`, even the display was using the `lkrn` extension.

Example:
```
# isoinfo -l -i ipxe.iso.v1.20.1-34-g5b3331b6_1597918131 -x '/ISOLINUX.CFG;1'

Directory listing of /
d---------   0    0    0            2048 Aug 20 2020 [     24 02]  .
d---------   0    0    0            2048 Aug 20 2020 [     24 02]  ..
----------   0    0    0            2048 Aug 20 2020 [     25 00]  BOOT.CAT;1
----------   0    0    0          349378 Aug 20 2020 [     38 00]  IPXE.KRN;1
----------   0    0    0           24576 Aug 20 2020 [     26 00]  ISOLINUX.BIN;1
# These default options can be changed in the geniso script
SAY iPXE ISO boot image
TIMEOUT 30
DEFAULT ipxe.lkrn
LABEL ipxe.lkrn
 KERNEL ipxe.krn          <<<<<< Previous kernel used krn as extension
----------   0    0    0             146 Aug 20 2020 [    209 00]  ISOLINUX.CFG;1
```

After the commit it builds the resulting `isolinux.cfg` by using the full extension, which is when used to boot a vmware image, not found:

```
# isoinfo -l -i ipxe.iso.v1.21.1-150-g9a0d873a_1617110797.iso -x '/ISOLINUX.CFG;1'

Directory listing of /
d---------   0    0    0            2048 Mar 30 2021 [     29 02]  .
d---------   0    0    0            2048 Mar 30 2021 [     29 02]  ..
----------   0    0    0            2048 Mar 30 2021 [     32 00]  BOOT.CAT;1
----------   0    0    0          348884 Mar 30 2021 [     45 00]  IPXE.LKR;1
----------   0    0    0           24576 Mar 30 2021 [     33 00]  ISOLINUX.BIN;1
SAY iPXE boot image
TIMEOUT 30
DEFAULT ipxe.lkrn
LABEL ipxe.lkrn
 KERNEL ipxe.lkrn          <<<<<< New kernel is referenced as lkrn extension
----------   0    0    0              83 Mar 30 2021 [    216 00]  ISOLINUX.CFG;1
```

This produces a booted isolinux where the kernel is not found:

I have managed to confirm this is related to this filename by patching the genfsimg to use the FAT32 name:
```diff
diff --git a/src/util/genfsimg b/src/util/genfsimg
index c3123f3e..7c6eb27d 100755
--- a/src/util/genfsimg
+++ b/src/util/genfsimg
@@ -184,7 +184,7 @@ for FILENAME ; do
     case "${FILENAME}" in
        *.lkrn)
            DESTDIR="${BIOSDIR}"
-           DESTFILE=$(basename "${FILENAME}")
+           DESTFILE=$(basename "${FILENAME}" | sed 's/.lkrn$/.lkr/')
            if [ -z "${LKRN}" ] ; then
                echo "SAY iPXE boot image" > "${SYSLINUXCFG}"
                echo "TIMEOUT 30" >> "${SYSLINUXCFG}"
```

The important parts of when `genfsimg` is executed is as following:
```
+ genisoimage -quiet -volid iPXE -preparer 'iPXE build system' -appid 'iPXE - Open Source Network Boot Firmware' -publisher ipxe.org -sysid iPXE -o bin/ipxe.iso -J -R -no-emul-boot -eltorito-boot isolinux.bin -boot-load-size 4 -boot-info-table bin/ipxe.iso.VMHWfZ/iso
+ isohybrid --version
+ isohybrid bin/ipxe.iso
```

Not sure why this is has changed, or if it is only broken for me or something has a too old version...

matya, 2021-04-20:
Did you ever seen this on your side @mcb30 ? I am not sure if I should contribute this "Fix" with a pull request, or if others have similar issues, or what is going on...

matya, 2021-04-20:
I think I have further traced it down to commit 79c0173d6df8580ea2e35fc173a5281e65d5321f, where the parameters for generating the iso has changed:

was:
```diff
a/src/util/geniso:
-mkisofs+=(-quiet -l -volid "iPXE" -preparer "iPXE build system"
-       -appid "iPXE ${VERSION} - Open Source Network Boot Firmware"
-       -publisher "http://ipxe.org/" -c boot.cat)
```

Please note `-l` here.

it was rewritten to:
```diff
b/src/util/genfsimg:
+if [ -n "${ISOIMG}" ] ; then
+    "${MKISOFS}" -quiet -volid "iPXE" -preparer "iPXE build system" \
+           -appid "iPXE - Open Source Network Boot Firmware" \
+           -publisher "ipxe.org" -sysid "iPXE" -J -R -o "${ISOIMG}" \
+           ${ISOARGS} "${ISODIR}"
```

The option `-l` was dropped, which according to the man page for genisoimage version 1.1.11:
```
  -l, -full-iso9660-filenames Allow full 31 character filenames for ISO9660 names
```

Also the [Syslinux wiki ](https://wiki.syslinux.org/wiki/index.php?title=ISOLINUX#Notes_about_building) states this:

> ISOLINUX:
>  - only uses plain ISO 9660 filenames, i.e. it does not support Rock Ridge [1], or Joliet filenames. It can still be used on a disc that uses Rock Ridge and/or Joliet extensions, of course. Under Linux, you can verify the plain filenames by mounting with the "-o norock,nojoliet" option of the mount command.
>  - supports long (level 2) ISO 9660 plain filenames, so if compatibility with short-names-only operating systems like MS-DOS is not an issue, **_you can use the "-l" or "-iso-level 2" option of mkisofs to generate long (up to 31 characters) plain filenames_**.

matya, 2021-04-20:
I can also confirm that building an ISO file with adding the `-l` option this issue is fixed:
```diff
diff --git a/src/util/genfsimg b/src/util/genfsimg
index c3123f3e..a53362eb 100755
--- a/src/util/genfsimg
+++ b/src/util/genfsimg
@@ -292,7 +292,7 @@ if [ -n "${ISOIMG}" ] ; then
        echo "${0}: cannot find a suitable mkisofs or equivalent" >&2
        exit 1
     fi
-    "${MKISOFS}" -quiet -volid "iPXE" -preparer "iPXE build system" \
+    "${MKISOFS}" -l -quiet -volid "iPXE" -preparer "iPXE build system" \
            -appid "iPXE - Open Source Network Boot Firmware" \
            -publisher "ipxe.org" -sysid "iPXE" -o "${ISOIMG}" \
            ${ISOARGS} "${ISODIR}"
```

mcb30, 2021-04-20:
@matya Good catch!  Could you submit a pull request that adds the `-l` option?  I would suggest adding it alongside `-J -R`, since that would keep all the filename-related options together.  For example:

```diff
diff --git a/src/util/genfsimg b/src/util/genfsimg
index c3123f3ed..1a0517cc9 100755
--- a/src/util/genfsimg
+++ b/src/util/genfsimg
@@ -230,7 +230,7 @@ done
 # since isohybrid will refuse to work without them.
 #
 if [ -n "${ISOIMG}" ] ; then
-    ISOARGS="-J -R"
+    ISOARGS="-J -R -l"
     copy_syslinux_file "isolinux.bin" "${ISODIR}"
     copy_syslinux_file "ldlinux.c32" "${ISODIR}" 2>/dev/null || true
     ISOARGS="${ISOARGS} -no-emul-boot -eltorito-boot isolinux.bin"
```

matya, 2021-04-20:
yeah, messed up the PR, not used to github, more to gitlab...

matya, 2021-04-20:
@mcb30 PR is open.

mcb30, 2021-04-20:
Merged; thanks!

PR #329 (2021-04-20): [build] Fix genfsimg to build ISO with long filenames
Commit 79c0173 introduced the new genfsimg, which lacks the -l option when building ISO Files.
This option is required to build level 2 (long plain) ISO9660 Filenames, which are required when using 
the .lkrn extensions. Fixes #306
closed 2021-04-20 by mcb30