ipxe/ipxe #320: Domain names not work. IP addresses only on UTM (VM)

sebaxakerhtc, 2021-04-13:
Hi!
Today i have some tests on VM's of a [nice project UTM](https://github.com/utmapp/UTM) on my iPad.
Of course, with iPXE! But the problem is all works just with ip addresses, but with domain names - no :(
If i understand correct - UTM uses qemu (if it helps)
On a screenshot i try to boot Ubuntu installer, but after about a 30 second it will show timeout error.
If i change domain name with ip - there are no problems.
Thank you for your work!
[image]

NiKiZe, 2021-04-13:
This is most likely not a big, so should fit better as discussion rather than issue.

The obvious things to do is to first check in iPXE which IP is used for DNS servers, the config command is helpful.

Try to use wireshark or tcpdump to try and figure out if and where any issues with DNS is, and/or compile iPXE with debug of DNS enabled.

Any failure in iPXE will show you an ipxe.org URL which have more detailed information, including debug steps.

sebaxakerhtc, 2021-04-13:
I think you are right.

sebaxakerhtc, 2021-04-13:
The problem was solved by manual adding DNS server address

closed 2021-04-13 by sebaxakerhtc