ipxe/ipxe #349: ESXi installer not working with v.1.21.1

yurinnick, 2021-04-30:
ESXi installer fails to download installation files after chainloading from iPXE v1.21.1 in UEFI mode.
At some point ESXi installer show an error: `Fatal error: 15 (Not found)` and fails to processed further.
This affects both `ipxe.efi` and `snponly.efi` builds.

Hardware: `HP DL360 G10`

iPXE script:
```
kernel http://<server_ipv6>/esxi/7.0/hpe/efi/boot/bootx64.efi -c http://<server_ipv6>/esxi.cfg -H -S 1 ks=http://<server_ipv6>/kickstart.ks text gdbPort=none logPort=none tty2Port=com1
```

ESXi config:
```
bootstate=0
title=ESXi installer for HP
kernel=b.b00
prefix=http://<server_ipv6>/esxi/7.0/hpe
kernelopt=runweasel addvmportgroup=0 bootproto=dhcp
modules=jumpstrt.gz --- useropts.gz --- features.gz --- k.b00 --- uc_intel.b00 --- uc_amd.b00 --- uc_hygon.b00 --- procfs.b00 --- vmx.v00 --- vim.v00 --- tpm.v00 --- sb.v00 --- s.v00 --- bnxtnet.v00 --- bnxtroce.v00 --- amscli.v00 --- amsd.v00 --- amshelpr.v00 --- amshelpr.v01 --- stage.v00 --- stage.v01 --- cru.v00 --- stage.v02 --- payload.v00 --- ssacli.v00 --- stage.v03 --- ilo.v00 --- ilorest.v00 --- stage.v04 --- stage.v05 --- sut.v00 --- sutesxcl.v00 --- stage.v06 --- stage.v07 --- i40en.v00 --- ixgben.v00 --- nmlx5cor.v00 --- nmlx5rdm.v00 --- nmst.v00 --- qcnic.v00 --- qedentv.v00 --- qedf.v00 --- qedi.v00 --- qedrntv.v00 --- qfle3.v00 --- qfle3f.v00 --- qfle3i.v00 --- atlantic.v00 --- brcmfcoe.v00 --- elxiscsi.v00 --- elxnet.v00 --- iavmd.v00 --- icen.v00 --- igbn.v00 --- irdman.v00 --- iser.v00 --- lpfc.v00 --- lpnic.v00 --- lsi_mr3.v00 --- lsi_msgp.v00 --- lsi_msgp.v01 --- lsi_msgp.v02 --- mtip32xx.v00 --- ne1000.v00 --- nenic.v00 --- nfnic.v00 --- nhpsa.v00 --- nmlx4_co.v00 --- nmlx4_en.v00 --- nmlx4_rd.v00 --- ntg3.v00 --- nvme_pci.v00 --- nvmerdma.v00 --- nvmxnet3.v00 --- nvmxnet3.v01 --- pvscsi.v00 --- qflge.v00 --- rste.v00 --- sfvmk.v00 --- smartpqi.v00 --- vmkata.v00 --- vmkfcoe.v00 --- vmkusb.v00 --- vmw_ahci.v00 --- clusters.v00 --- crx.v00 --- elx_esx_.v00 --- btldr.v00 --- esx_dvfi.v00 --- esx_ui.v00 --- esxupdt.v00 --- tpmesxup.v00 --- weaselin.v00 --- loadesx.v00 --- lsuv2_hp.v00 --- lsuv2_in.v00 --- lsuv2_ls.v00 --- lsuv2_nv.v00 --- lsuv2_oe.v00 --- lsuv2_oe.v01 --- lsuv2_oe.v02 --- lsuv2_sm.v00 --- native_m.v00 --- qlnative.v00 --- vdfs.v00 --- vmware_e.v00 --- vsan.v00 --- vsanheal.v00 --- vsanmgmt.v00 --- tools.t00 --- xorg.v00 --- gc.v00 --- imgdb.tgz --- basemisc.tgz --- resvibs.tgz --- imgpayld.tgz
build=
updated=0
```

Using iPXE v1.20.1 solves all the issues and installation works perfectly fine.

NiKiZe, 2021-05-01:
Not sure if is needed, but could you do a git bisect to know where this issue where introduced? https://ipxe.org/howto/bisect

yurinnick, 2021-05-21:
I didn't have time to pinpoint breaking change with bisect yet. 

I used official ESXi 6.7 U3 for HP Proliant image from [here](https://www.hpe.com/us/en/servers/hpe-esxi.html)

berlin4apk, 2021-05-23:
> I used official ESXi 6.7 U3 for HP Proliant image from [here](https://www.hpe.com/us/en/servers/hpe-esxi.html)

But in your first post it's looks like ESXi 7.0 not 6.7u3t

> ```
> kernel http://<server_ipv6>/esxi/7.0/hpe/efi/boot/bootx64.efi -c http://<server_ipv6>/esxi.cfg -H -S 1 ks=http://<server_ipv6>/kickstart.ks text gdbPort=none logPort=none tty2Port=com1
> ```
>
> ESXi [...]

yurinnick, 2021-05-28:
I tested it separately with ESXi 6.7u3 and ESXi 7.0 and both shows the same behaviour.

NiKiZe, 2021-07-09:
If this is an issue with iPXE, and it has worked before, have you, or could you do a git bisect on iPXE to try and figure out what possibly broke? https://ipxe.org/howto/bisect

yurinnick, 2021-07-19:
Bisect result is:
```
1295b4acff1f2014261c40d9f9d2107ffd668d92 is the first bad commit
commit 1295b4acff1f2014261c40d9f9d2107ffd668d92
Author: Michael Brown <mcb30@ipxe.org>
Date:   Mon Nov 23 15:34:13 2020 +0000

    [efi] Allow initialisation via SNP interface even while claimed
```

omniproc, 2021-11-11:
can confirm this issue. we're seeing exactly the same. error 15 right after chainloading the ESXi bootloader. Tried multiple ESXi versions that worked before. We only see this issue on DL380 servers from HP.
There's not much else log output I could provide but it looks like a issue with the network adapter that works when iPXE boots and loads the ESXi image over the network. But as soon as the ESXi bootloader starts it seems unable to be unable to use that NIC to get the next file over the network, thus the error 15. We have debug mode enabled and we also see "issue while parsing URL http://path-to-file-esxi-wants-to-load". The file exists of course and as mentioned we only see this with some servers so I guess it's hardware and thus driver related.

omniproc, 2021-11-16:
To add: we're building `bin-x86_64-efi/ipxe.efi`

dhekimian, 2022-04-01:
Spent half a day chaising this issue down today... v1.21.1 doesn't work with VMware ESXi 6.7->7.0u3d (released March 29, 2022)

Downloaded v1.20.1, recompiled and volia, it worked.

```
wget https://github.com/ipxe/ipxe/archive/refs/tags/v1.20.1.tar.gz
tar xzvf v1.20.1.tar.gz
cd ipxe-1.20.1/src
make bin-x86_64-efi/ipxe.efi
cp bin-x86_64-efi/ipxe.efi /<tftp root>

stappersg, 2022-04-01:
2021-11-12 got this issue this information:

>  ... and as mentioned we only see this with some servers
> so I guess it's hardware and thus driver related.

On Thu, 2022-03-31 at 05:33:26PM -0700, David Hekimian wrote:
>   ...

@dhekimian:  Which hardware is used by "your VMware"?

Groeten
Geert Stappers
-- 
Silence is hard to parse

stappersg, 2022-04-01:
2021-11-12 got this issue this information:

>  ... and as mentioned we only see this with some servers
> so I guess it's hardware and thus driver related.

On Thu, 2022-03-31 at 05:33:26PM -0700, David Hekimian wrote:
>   ...

@dhekimian:  Which hardware is used by "your VMware"?

dhekimian, 2022-04-01:
My Servers are a mix of Dell PowerEdge. The Network card is a [Mellanox ConnextX-5 EN - MCX516A-CCAT (100GbE Dual Port)](https://www.nvidia.com/en-us/networking/ethernet/connectx-5/). Their documentation says their PXE is based on iPXE https://docs.nvidia.com/networking/display/PreBootDriversv13

koeniglorenz, 2022-05-13:
We have also been experiencing problems with booting the ESXi-Installer with images from Fujitsu (7.0U2 and U3) and Lenovo (7.0U3) on Fujitsu (RX1330) and Lenovo (SR250) hardware.
We are chainloading iPXE and using a config similar to what @yurinnick showed in his first post ( https://github.com/ipxe/ipxe/issues/349#issue-873469718)
When booting these images the ESXi-installer can indeed download all the files, but then hangs at the following screen before the install-process starts:
```
Shutting down firmware services...
Using `simple offset` UEFI RTS mapping policy
```
This curiously doesn't happen all the time but in 9/10 times it hangs at this step.
As this seems to be related to https://github.com/ipxe/ipxe/pull/113 / https://github.com/ipxe/ipxe/pull/120 I first thought that this would be fixed as I complied iPXE from the current master branch, but that did not change the behaviour.
However, applying the change https://github.com/ipxe/ipxe/pull/113/commits/cbff453fd00835fd3735fca28990a83630d66f70 fixes the issue and we can successfully boot the installer.

I don't mean to hijack this issue as at first glance it seems to be some other problem, but since there already was an issue regarding booting ESXi I figured I'd post it here first.
I'd be happy to provide some more information or open another issue if that makes more sense.

stappersg, 2022-05-14:
Acknowledge on _this change works for me_.  Following the provided link:

[image]

(Note the _This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository._)

So now there is the challenge to convert the change into a _merge request_.

koeniglorenz, 2022-05-14:
@stappersg This commit was part of the pull request https://github.com/ipxe/ipxe/pull/113. That pull request was never merged, because, as far as i understand, the root issue seemed to have been fixed with pull request https://github.com/ipxe/ipxe/pull/120 which was merged.
The problem we are seeing however is still exactly the one that was discussed in https://github.com/ipxe/ipxe/pull/113 and  https://github.com/ipxe/ipxe/pull/120 did not fix the problem entirely.

stappersg, 2022-05-14:
Thank you @koeniglorenz  for telling where to find the patch.

The one that I found did not apply cleanly first, but did after some labor.

Because I don't known how to merge request with credits to original author, did I send the patch to the iPXE mailinglist.
It is archived as https://lists.ipxe.org/pipermail/ipxe-devel/2022-May/007557.html

dhekimian, 2022-08-18:
Any chance this will get merged at some point?

stappersg, 2022-08-19:
That 
> Any chance this will get merged at some point?

is probably

> What to do to get the missing part merged?

Options I see:
* https://ipxe.org/contact
* Have it merged into your `git clone` of the _iPXE git repository_.

Vitalium, 2023-06-19:
> Spent half a day chaising this issue down today... v1.21.1 doesn't work with VMware ESXi 6.7->7.0u3d (released March 29, 2022)
>
> Downloaded v1.20.1, recompiled and volia, it worked.
>
> ```
> wget [...]

Hello!

I tried to compile 1.20.1, but got error with `gcc version 10.2.1 20210110 (Debian 10.2.1-6)`:
```ld: bin-x86_64-efi/blib.a(dmfe.o):(.bss+0x0): multiple definition of `_shared_bss'; bin-x86_64-efi/blib.a(davicom.o):(.bss+0x0): first defined here
ld: bin-x86_64-efi/blib.a(epic100.o):(.bss+0x0): multiple definition of `_shared_bss'; bin-x86_64-efi/blib.a(davicom.o):(.bss+0x0): first defined here
...
and so on```

Does anyone have prebuild 1.20.1 efi binary?

NiKiZe, 2023-06-19:
> Hello!
>
> I tried to compile 1.20.1, but got error with `gcc version 10.2.1 20210110 (Debian 10.2.1-6)`: `` ld: bin-x86_64-efi/blib.a(dmfe.o):(.bss+0x0): multiple definition of `_shared_bss'; [...]

As indicated by users above, a git bisect gives 1295b4acff1f2014261c40d9f9d2107ffd668d92, try to revert that commit might cause conflicts, but could be worth a try, going the other way around might also be possible and cherry pick any GCC fixes on top of the last working commit.

Vitalium, 2023-06-20:
Thank you for the hint! I tried several commits before Nov 24, 2020 and have success to load ESXi installer, but encountered problems with loading another kernels :(

mcb30, 2023-06-22:
@Vitalium I am unable to reproduce the problem with the current master branch (which includes the fix from #120).

You may want to retest with the current master branch.  There was a recent commit https://github.com/ipxe/ipxe/commit/25a3d3aca to work around a bug in the VMware UefiPxeBcDxe driver provided by recent versions of ESX.

Vitalium, 2023-06-22:
Thank you for the answer! I just tried a have no luck :(
Still got `Fatal error: 15 (Not found)` 

For testing I use ESXi 8 VM and boot there ESXi 8 latest installer.

However, I compiled 1.20.1 using old gcc and have no problem here! But with that version, SystemRescue hangs upon boot.
So I think the best workaround will be chainload another iPXE from itself to load specifically ESXi.

anthonyclouvider, 2023-10-25:
Issue still exists with the current master branch and latest 7.x version of esxi 7.0.3-0.95.21930508 Update 3n | JUL 06 2023

anthonyclouvider, 2023-10-25:
> Issue still exists with the current master branch and latest 7.x version of esxi 7.0.3-0.95.21930508 Update 3n | JUL 06 2023

I thought I would share my solution for others in the same situation that are also hit by the gcc issues.

```
cd /tmp
wget https://github.com/ipxe/ipxe/archive/refs/tags/v1.20.1.tar.gz
tar xzvf v1.20.1.tar.gz
mv ipxe-1.20.1 ipxe
cd ipxe/src
```

Then in src/ edit the Makefile.housekeeping file.

Change lines 164 to 171 

Old:
```
ifeq ($(CCTYPE),gcc)
CFI_TEST = $(CC) -fno-dwarf2-cfi-asm -fno-exceptions -fno-unwind-tables \
                 -fno-asynchronous-unwind-tables -x c -c /dev/null \
                 -o /dev/null >/dev/null 2>&1
CFI_FLAGS := $(shell $(CFI_TEST) && \
               $(ECHO) '-fno-dwarf2-cfi-asm -fno-exceptions ' \
                       '-fno-unwind-tables -fno-asynchronous-unwind-tables')
WORKAROUND_CFLAGS += $(CFI_FLAGS)
endif
```

NEW:
```
ifeq ($(CCTYPE),gcc)
CFI_TEST = $(CC) -fno-dwarf2-cfi-asm -fno-exceptions -fno-unwind-tables \
                 -fno-asynchronous-unwind-tables -x c -c /dev/null \
                 -o /dev/null >/dev/null 2>&1
CFI_FLAGS := $(shell $(CFI_TEST) && \
               $(ECHO) '-fno-dwarf2-cfi-asm -fno-exceptions ' \
                       '-fno-unwind-tables -fno-asynchronous-unwind-tables')
WORKAROUND_CFLAGS += $(CFI_FLAGS)

FCOMMON_TEST = $(CC) -fcommon -x c -c /dev/null \
		   -o /dev/null >/dev/null 2>&1
FCOMMON_FLAGS := $(shell $(FCOMMON_TEST) && $(ECHO) '-fcommon')
WORKAROUND_CFLAGS += $(FCOMMON_FLAGS)
endif
```

Now modify line 368 to be:
```
status = nodnic_port_create_qp(&port->port_priv, (nodnic_queue_pair_type) qp->type,
```

and lines 408 to 411 to:

```
nodnic_port_destroy_qp(&port->port_priv, (nodnic_queue_pair_type) qp->type,
                        flexboot_nodnic_qp->nodnic_queue_pair);

free(flexboot_nodnic_qp);
```

Now compile.

```
make bin-x86_64-efi/ipxe.efi
```

This now lets esxi work as of the latest 7.x version at the time of this comment.